zephyrproject-rtos/zephyr #37790: Bluetooth: host: Confusion about periodic advertising interval

joerchan, 2021-08-19:
**Is your enhancement proposal related to a problem? Please describe.**
The periodic advertising parameter defines re-uses advertising interval define for the periodic advertising interval.
This causes unexpected periodic advertising interval because these are in different time units. Advertising interval is in 0.625 s units and periodic advertising interval is in 1.25 s units.

**Describe the solution you'd like**
Periodic advertising defines should use periodic advertising interval defines.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

Thalley, 2021-08-19:
@joerchan Shouldn't this have been created as a bug? If the interval set is twice what is requested, that sounds like a bug to me :)

joerchan, 2021-08-19:
I consider it a documentation issue. The periodic advertising is specifying it's using "SLOW" interval. But you can't correctly read what the slow interval is from the documentation.