zephyrproject-rtos/zephyr #37821: pm: `pm_device_request` incorrectly returns errors

JordanYates, 2021-08-20:
**Describe the bug**

When the same device is taken multiple times, the releases of that device incorrectly report failure (`-EIO`).
This error is not valid, as nothing has gone wrong beyond the device still being used by another piece of code.

**To Reproduce**

Apply the following diff:
```
diff --git a/samples/subsys/pm/device_pm/src/dummy_driver.c b/samples/subsys/pm/device_pm/src/dummy_driver.c
index b70b75a923..0350b9dacd 100644
--- a/samples/subsys/pm/device_pm/src/dummy_driver.c
+++ b/samples/subsys/pm/device_pm/src/dummy_driver.c
@@ -23,6 +23,10 @@ static int dummy_open(const struct device *dev)
 	if (ret < 0) {
 		return ret;
 	}
+	ret = pm_device_get(parent);
+	if (ret < 0) {
+		return ret;
+	}
 
 	ret = pm_device_get_async(dev);
 	if (ret < 0) {
@@ -80,7 +84,16 @@ static int dummy_close(const struct device *dev)
 
 	/* Parent can be suspended */
 	if (parent) {
-		pm_device_put(parent);
+		ret = pm_device_put(parent);
+		if (ret) {
+			printk("FAILED 1\n");
+			return ret;
+		}
+		ret = pm_device_put(parent);
+		if (ret) {
+			printk("FAILED 2\n");
+			return ret;
+		}
 	}
 
 	return ret;
```
Build and run:
```
*** Booting Zephyr OS build zephyr-v2.6.0-2297-gfbd5e1f31687  ***
Device PM sample app start
open()
parent resuming..
child resuming..
Async wakeup request queued
Dummy device resumed
write()
transfer()
read()
transfer()
close()
child suspending..
FAILED 1
Device PM sample app complete
```

**Expected behavior**
Errors should only be returned on actual error conditions.

**Environment (please complete the following information):**
 - v2.6 main

**Additional context**

The problem is in these lines. When releasing a device that has been claimed multiple times, it is not expected that the current state is equal to the requested state:

https://github.com/zephyrproject-rtos/zephyr/blob/fbd5e1f31687711beef069412d5b5e9a52c1d8e9/subsys/pm/device_runtime.c#L141-L146

JordanYates, 2021-08-20:
@gmarull @ceolin

ceolin, 2021-08-24:
@JordanYates -EIO in this case is returned because the device reference count is not 0. I don't think return 0 is right here, because the device ultimately didn't change its state to the one requested. -EIO is also not accurate IMHO. I think what best describe this situation is EBUSY, take a look in the linked pr.

JordanYates, 2021-08-25:
I disagree with returning `-EBUSY`, the call is not failing because it is already in operation, it has succeeded.

The 4 functions that call `pm_device_request` (`pm_device_{get/sync}{_async}`), are all documented in terms of the usage count.
The first requesting call will turn the device on, the last release will turn it off.
Any additional requests+releases after the first request should not result in error codes as the device and API are operating per the documentation.