zephyrproject-rtos/zephyr #37827: stm32h747i_disco M4 will not working, if k_malloc() are used

cameled, 2021-08-20:
**Describe the bug**
Hi, I meet a strange problem this week, stm32h747i_disco M4 will not working, if k_malloc() are used.
I have test it on fresh build zephyr blinky sample. with below patch
```
diff --git a/samples/basic/blinky/prj.conf b/samples/basic/blinky/prj.conf
index 91c3c15b37..e8fe3b581f 100644
--- a/samples/basic/blinky/prj.conf
+++ b/samples/basic/blinky/prj.conf
@@ -1 +1,2 @@
 CONFIG_GPIO=y
+CONFIG_HEAP_MEM_POOL_SIZE=4096
diff --git a/samples/basic/blinky/src/main.c b/samples/basic/blinky/src/main.c
index e2fbfd3dc5..1e5c24277a 100644
--- a/samples/basic/blinky/src/main.c
+++ b/samples/basic/blinky/src/main.c
@@ -43,6 +43,14 @@ void main(void)
                return;
        }
 
+       gpio_pin_set(dev, PIN, (int)0);
+       k_msleep(SLEEP_TIME_MS);
+       gpio_pin_set(dev, PIN, (int)1);
+       k_msleep(SLEEP_TIME_MS);
+
+       int *p1 = (int *)k_malloc(4);
+       *p1 = INT_MAX;
+
        while (1) {
                gpio_pin_set(dev, PIN, (int)led_is_on);
                led_is_on = !led_is_on;
```

**To Reproduce**
Steps to reproduce the behavior:
1. cd zephyr
2. west build -p -b stm32h747i_disco_m4 samples/basic/blinky
3. west flash

**Expected behavior**
blue led for M4 is blink.

**Impact**
open-amp or other module/app using k_malloc() can not work on stm32h747i_disco M4 core.

**Logs and console output**
Actually, that's where the strange thing happened. 
If I remove __WFE() on soc_m4.c(not let M4 to Cstop mode,) blue led for M4 are blinking.
If M4 enter CStop mode, event though M7 detecdt M4 is wake up, but M4 code are not working.(no blink)

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: zephyr-sdk-0.13.0
 - main branch.

**Additional context**
Add any other context about the problem here.

cameled, 2021-08-20:
Looks it's a link issue, If reduce the CONFIG_HEAP_MEM_POOL_SIZE to 1024, blinky sample start to work.

cameled, 2021-09-06:
After erase by STM32Programer, this issue disappeared.

erwango, 2021-09-06:
@cameled Strange. Anyway, thanks for feedback.

cameled, 2021-09-09:
Looks that's a random success, STM32Programer is not the key.

cameled, 2021-09-11:
As said in  #38322, disable BCM4 config can fix this issue.

closed 2021-09-11 by cameled