zephyrproject-rtos/zephyr #37830: intel_adsp_cavs15: run queue testcases failed on ADSP

KangJianX, 2021-08-20:
**Describe the bug**
run tests/kernel/queue failed on ADSP

**To Reproduce**
Steps to reproduce the behavior:
1. twister -p intel_adsp_cavs15 --device-testing -T tests/kernel/queue --west-flash="/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/flash.sh,/home/ztest/work/zephyrproject/modules/audio/sof/keys/otc_private_key.pem,/home/ztest/work/rimage/config,/home/ztest/work/rimage/build/rimage" --device-serial-pty="/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/adsplog.py,--no-history"

**Logs and console output**

> DEBUG   - DEVICE: START - test_multiple_queues
DEBUG   - DEVICE: PASS - test_multiple_queues in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_queue_multithread_competition
DEBUG   - DEVICE:
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/queue/src/test_queue_contexts.c:452: high_prio_t1_wait_for_queue: (*ret == 0xaaa is false)
DEBUG   - DEVICE: The highest priority and waited longest get the queue data failed firstly
DEBUG   - DEVICE: FAIL - test_queue_multithread_competition in 0.61 seconds

**Environment (please complete the following information):**
 - OS: Fedora28
 - Toolchain: Zephyr-sdk-0.13.0
 - Commit ID: 7c2dfc1558e5

andyross, 2021-09-08:
This works for me in current main.  Might potentially have been fixed by the multiprocessing PR that just went in?  Can you retest?

KangJianX, 2021-09-14:
> This works for me in current main. Might potentially have been fixed by the multiprocessing PR that just went in? Can you retest?

Blocked by #38446 . I will retest this one when #38446 is fixed.

closed 2021-10-06 by nashif