zephyrproject-rtos/zephyr #37838: cmake 3.20 not supportet (yet) by recent Ubuntu

caco3, 2021-08-20:
You updated the requirement for cmake to `3.20` in https://github.com/zephyrproject-rtos/zephyr/commit/1cccc8a8feda1b1cbff4a81fbb552b75f0f1ce28

How ever not even Ubuntu 21.04 provides this version, see https://packages.ubuntu.com/search?keywords=cmake

It seems not wise to set the requirement highter than currently provided by recent distros!

caco3, 2021-08-20:
@carlescufi FYI

de-nordic, 2021-08-20:
You should be able to update cmake via pip3 to the 3.20 version.

caco3, 2021-08-20:
3.20 only got released in March 2021, so its still pritty new! 

https://blog.kitware.com/cmake-3-20-0-available-for-download/

de-nordic, 2021-08-20:
I got mine updated, via `pip3  install -U cmake` to
```
$$ cmake --version
cmake version 3.21.1

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```

caco3, 2021-08-20:
> 
> 
> You should be able to update cmake via pip3 to the 3.20 version.

Thanks, I was not aware that cmake is also a Python based tool.

closed 2021-08-20 by caco3