zephyrproject-rtos/zephyr #37839: SX1272 LoRa driver is broken and fails to build

thedjnK, 2021-08-20:
**Describe the bug**
We are adding our SX1272-based development board and trying to compile the lora sample application under samples/subsys/lorawan/class_a, it fails to build with the following output:
```
In file included from C:/zeph/zephyr/drivers/lora/sx12xx_common.c:11:
C:/zeph/zephyr/include/sys/atomic_builtin.h:37:20: error: redefinition of 'atomic_cas'
   37 | static inline bool atomic_cas(atomic_t *target, atomic_val_t old_value,
      |                    ^~~~~~~~~~
In file included from C:/zeph/zephyr/include/sys/atomic_c.h:75,
                 from C:/zeph/zephyr/include/sys/atomic.h:30,
                 from C:/zeph/zephyr/include/kernel_includes.h:21,
                 from C:/zeph/zephyr/include/kernel.h:17,
                 from C:/zeph/zephyr/include/init.h:11,
                 from C:/zeph/zephyr/include/device.h:29,
                 from C:/zeph/zephyr/include/drivers/gpio.h:23,
                 from C:/zeph/zephyr/drivers/lora/sx12xx_common.c:8:
C:/zeph/zephyr/samples/subsys/lorawan/class_a/build2/zephyr/include/generated/syscalls/atomic_c.h:28:20: note: previous definition of 'atomic_cas' was here
   28 | static inline bool atomic_cas(atomic_t * target, atomic_val_t old_value, atomic_val_t new_value)
      |                    ^~~~~~~~~~
In file included from C:/zeph/zephyr/drivers/lora/sx12xx_common.c:11:
C:/zeph/zephyr/include/sys/atomic_builtin.h:58:20: error: redefinition of 'atomic_ptr_cas'
   58 | static inline bool atomic_ptr_cas(atomic_ptr_t *target, atomic_ptr_val_t old_value,
      |                    ^~~~~~~~~~~~~~
In file included from C:/zeph/zephyr/include/sys/atomic_c.h:75,
                 from C:/zeph/zephyr/include/sys/atomic.h:30,
                 from C:/zeph/zephyr/include/kernel_includes.h:21,
                 from C:/zeph/zephyr/include/kernel.h:17,
                 from C:/zeph/zephyr/include/init.h:11,
                 from C:/zeph/zephyr/include/device.h:29,
                 from C:/zeph/zephyr/include/drivers/gpio.h:23,
                 from C:/zeph/zephyr/drivers/lora/sx12xx_common.c:8:
C:/zeph/zephyr/samples/subsys/lorawan/class_a/build2/zephyr/include/generated/syscalls/atomic_c.h:42:20: note: previous definition of 'atomic_ptr_cas' was here
   42 | static inline bool atomic_ptr_cas(atomic_ptr_t * target, atomic_ptr_val_t old_value, atomic_ptr_val_t new_value)
      |                    ^~~~~~~~~~~~~~
In file included from C:/zeph/zephyr/drivers/lora/sx12xx_common.c:11:
C:/zeph/zephyr/include/sys/atomic_builtin.h:77:28: error: redefinition of 'atomic_add'
   77 | static inline atomic_val_t atomic_add(atomic_t *target, atomic_val_t value)
      |                            ^~~~~~~~~~
In file included from C:/zeph/zephyr/include/sys/atomic_c.h:75,
                 from C:/zeph/zephyr/include/sys/atomic.h:30,
                 from C:/zeph/zephyr/include/kernel_includes.h:21,
                 from C:/zeph/zephyr/include/kernel.h:17,
                 from C:/zeph/zephyr/include/init.h:11,
                 from C:/zeph/zephyr/include/device.h:29,
                 from C:/zeph/zephyr/include/drivers/gpio.h:23,
                 from C:/zeph/zephyr/drivers/lora/sx12xx_common.c:8:
C:/zeph/zephyr/samples/subsys/lorawan/class_a/build2/zephyr/include/generated/syscalls/atomic_c.h:56:28: note: previous definition of 'atomic_add' was here
   56 | static inline atomic_val_t atomic_add(atomic_t * target, atomic_val_t value)
      |                            ^~~~~~~~~~
In file included from C:/zeph/zephyr/drivers/lora/sx12xx_common.c:11:
C:/zeph/zephyr/include/sys/atomic_builtin.h:93:28: error: redefinition of 'atomic_sub'
   93 | static inline atomic_val_t atomic_sub(atomic_t *target, atomic_val_t value)
      |                            ^~~~~~~~~~
In file included from C:/zeph/zephyr/include/sys/atomic_c.h:75,
                 from C:/zeph/zephyr/include/sys/atomic.h:30,
                 from C:/zeph/zephyr/include/kernel_includes.h:21,
                 from C:/zeph/zephyr/include/kernel.h:17,
                 from C:/zeph/zephyr/include/init.h:11,
                 from C:/zeph/zephyr/include/device.h:29,
                 from C:/zeph/zephyr/include/drivers/gpio.h:23,
                 from C:/zeph/zephyr/drivers/lora/sx12xx_common.c:8:
C:/zeph/zephyr/samples/subsys/lorawan/class_a/build2/zephyr/include/generated/syscalls/atomic_c.h:70:28: note: previous definition of 'atomic_sub' was here
   70 | static inline atomic_val_t atomic_sub(atomic_t * target, atomic_val_t value)
      |                            ^~~~~~~~~~
In file included from C:/zeph/zephyr/drivers/lora/sx12xx_common.c:11:
C:/zeph/zephyr/include/sys/atomic_builtin.h:108:28: error: redefinition of 'atomic_inc'
  108 | static inline atomic_val_t atomic_inc(atomic_t *target)
      |                            ^~~~~~~~~~
In file included from C:/zeph/zephyr/include/sys/atomic.h:30,
                 from C:/zeph/zephyr/include/kernel_includes.h:21,
                 from C:/zeph/zephyr/include/kernel.h:17,
[... 159 lines cut ...]
                 from C:/zeph/zephyr/include/kernel.h:17,
                 from C:/zeph/zephyr/include/init.h:11,
                 from C:/zeph/zephyr/include/device.h:29,
                 from C:/zeph/zephyr/include/drivers/gpio.h:23,
                 from C:/zeph/zephyr/drivers/lora/sx12xx_common.c:8:
C:/zeph/zephyr/samples/subsys/lorawan/class_a/build2/zephyr/include/generated/syscalls/atomic_c.h:154:28: note: previous definition of 'atomic_nand' was here
  154 | static inline atomic_val_t atomic_nand(atomic_t * target, atomic_val_t value)
      |                            ^~~~~~~~~~~
[49/161] Building C object zephyr/subsys/lorawan/CMakeFile...dir/C_/zeph/modules/lib/loramac-node/src/mac/LoRaMac.c.obj
ninja: build stopped: subcommand failed.
```

**To Reproduce**
Steps to reproduce the behavior: Build an SX1272-based board with the lorawan sample app

**Expected behavior**
For the compilation to complete

**Impact**
Showstopper

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain: GNU Arm Embedded Toolchain 10.3-2021.07
 - Commit SHA or Version used: b6a3dc8d715d0cdb6edeb9e144f2687fd569cb16

mniestroj, 2021-08-20:
@thedjnK Please share the command you use for building. Or at least the platform you are building for.

thedjnK, 2021-08-20:
```cmake -GNinja -DBOARD=rm1xx_dvk ..
ninja```

Board contains an nRF51822 and SX1272

thedjnK, 2021-08-20:
```
cmake -GNinja -DBOARD=rm1xx_dvk ..
ninja
```
to build

mniestroj, 2021-08-20:
@thedjnK Please test #37840 and let us know if that fixes your build error (I tried with nrf51dk and it seems to build properly).

PR #37840 (2021-08-20): drivers: lora: sx12xx: fix atomic include
Include sys/atomic.h instead of sys/atomic_builtin.h, so that build for
platforms with non-default atomic implementation will still succeed.

Fixes: #37839
closed 2021-08-20 by nashif