zephyrproject-rtos/zephyr #37855: STM32 - kconfigs to determine if peripheral is available

stoberblog, 2021-08-22:
**Is your enhancement proposal related to a problem? Please describe.**
While looking into implementing a counter driver for the STM32 hardware timers (see issue https://github.com/zephyrproject-rtos/zephyr/issues/35816), I was looking at the Nordic NRF driver and found it has kconfigs that can be checked such as:
```
config HAS_HW_NRF_TIMER0
	bool

config HAS_HW_NRF_TIMER1
	bool

config HAS_HW_NRF_TIMER2
	bool

config HAS_HW_NRF_TIMER3
	bool

config HAS_HW_NRF_TIMER4
	bool
```
This would be great for other platforms, in this case STM32. Each SOC series and variant could then enable the correct set of peripherals/timers. Currently, I don't think there is an easy way for to determine which timers are valid per soc in use. For my prototype implementation, I'll just hard code which timers are available to make sure the code works.

**Describe the solution you'd like**
a kconfig setup similar to this https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/arm/nordic_nrf/Kconfig.peripherals

**Describe alternatives you've considered**
Hard coding each driver module to determine using SOC name but this would be hard to update for new series and variants

stoberblog, 2021-08-22:
I forgot to mention, I tried looking for something in soc/arm/st_stm32 but I couldn't see anything there, hence why I assume there isn't a comparative method

erwango, 2021-08-23:
@stoberblog We typically use dts (dts/arm/st/xx/*.dtsi) to describe which peripherals are available on which SoCs.
Can you explain why this is not sufficient ?

closed 2023-01-26 by erwango