zephyrproject-rtos/zephyr #37856: tests: arm: uninitialized FPSCR

JordanYates, 2021-08-22:
**Describe the bug**

The `arm_thread_swap` test fails on main for the FPU subtests on `mps3_an547`.

**To Reproduce**

```
west build -b mps3_an547 zephyr/tests/arch/arm/arm_thread_swap -- -DCONFIG_FPU=y -DCONFIG_FPU_SHARING=y
west build -t run
```

**Expected behavior**

Test should pass.

**Impact**

Causing CI failures in #37279.
https://buildkite.com/zephyr/zephyr/builds/38666

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v2.6.0-2360-gc94e53c9f5b6  ***
Running test suite arm_thread_swap
===================================================================
START - test_arm_thread_swap

    Assertion failed at WEST_TOPDIR/zephyr/tests/arch/arm/arm_thread_swap/src/arm_thread_arch.c:464: test_arm_thread_swap: (__get_FPSCR() == 0 is false)
FPSCR is not cleared at initialization: 0x40000

 FAIL - test_arm_thread_swap in 0.3 seconds
===================================================================
START - test_arm_syscalls
Available IRQ line: 127
USR Thread: IRQ Line: 127
 PASS - test_arm_syscalls in 0.3 seconds
===================================================================
START - test_syscall_cpu_scrubs_regs
Writing 0xDEADBEEF values into registers
Exit from system call
 PASS - test_syscall_cpu_scrubs_regs in 0.3 seconds
===================================================================
Test suite arm_thread_swap failed.
===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - Zephyr SDK 13.0.0
 - c94e53c9f5b6be29ce42b3cc8b50c6a27a487246

stephanosio, 2021-08-24:
cc @galak @microbuilder @ioannisg

stephanosio, 2021-08-24:
Downgrading it to low priority since it is not an arch implementation bug, but rather a bug with this test.

The Armv8.1-M architecture, which the Cortex-M55 of `mps3_an547` implements, added new fields to the FPSCR register -- in particular, the `FPSCR[18:16] LTPSIZE` field reads the value of `0b010` if no FP context is active, hence why we see `0x40000` above.

From Armv8-M ARM, p.1517:

> The loop hardware behaves as if this field had the value 4 (indicating no low-overhead-loop predication) if no
FP context is active. This field reads as 4 and ignores writes if MVE is not implemented.

We need to update the ARM tests so that it ignores the value of `LTPSIZE`. All the other fields are RAZ/WI, so we need not do anything for them.