zephyrproject-rtos/zephyr #37861: tests/lib/ringbuffer failed on ARC hsdk board

IRISZZW, 2021-08-23:
**Describe the bug**
tests/lib/ringbuffer failed on ARC hsdk board

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b hsdk -d build tests/lib/ringbuffer
2. west -v flash  -r mdb-hw  --cores=4
3. See error

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v2.6.0-2360-gc94e53c9f5b6  ***
Running test suite test_ringbuffer_api
===================================================================
START - test_ringbuffer_init
 PASS - test_ringbuffer_init in 0.1 seconds
===================================================================
START - test_ringbuffer_declare_pow2
 PASS - test_ringbuffer_declare_pow2 in 0.1 seconds
===================================================================
START - test_ringbuffer_declare_size
 PASS - test_ringbuffer_declare_size in 0.1 seconds
===================================================================
START - test_ringbuffer_put_get_thread
 PASS - test_ringbuffer_put_get_thread in 0.4 seconds
===================================================================
START - test_ringbuffer_put_get_isr
 PASS - test_ringbuffer_put_get_isr in 0.1 seconds
===================================================================
START - test_ringbuffer_put_get_thread_isr
 PASS - test_ringbuffer_put_get_thread_isr in 0.1 seconds
===================================================================
START - test_ringbuffer_put_get_discard
 PASS - test_ringbuffer_put_get_discard in 0.1 seconds
===================================================================
START - test_ringbuffer_pow2_put_get_thread_isr
 PASS - test_ringbuffer_pow2_put_get_thread_isr in 0.1 seconds
===================================================================
START - test_ringbuffer_size_put_get_thread_isr
 PASS - test_ringbuffer_size_put_get_thread_isr in 0.1 seconds
===================================================================
START - test_ringbuffer_array_perf
 PASS - test_ringbuffer_array_perf in 0.1 seconds
===================================================================
START - test_ringbuffer_partial_putting
 PASS - test_ringbuffer_partial_putting in 0.1 seconds
===================================================================
START - test_ringbuffer_partial_getting
 PASS - test_ringbuffer_partial_getting in 0.1 seconds
===================================================================
START - test_ring_buffer_main
 PASS - test_ring_buffer_main in 0.1 seconds
===================================================================
START - test_ringbuffer_raw
 PASS - test_ringbuffer_raw in 0.1 seconds
===================================================================
START - test_ringbuffer_alloc_put
 PASS - test_ringbuffer_alloc_put in 0.1 seconds
===================================================================
START - test_byte_put_free
 PASS - test_byte_put_free in 0.1 seconds
===================================================================
START - test_ringbuffer_equal_bufs
 PASS - test_ringbuffer_equal_bufs in 0.1 seconds
===================================================================
START - test_capacity
 PASS - test_capacity in 0.1 seconds
===================================================================
START - test_size
 PASS - test_size in 0.1 seconds
===================================================================
START - test_peek
 PASS - test_peek in 0.1 seconds
===================================================================
START - test_reset
 PASS - test_reset in 0.1 seconds
===================================================================
START - test_ringbuffer_performance
1 byte put-get, avg cycles: 264
4 byte put-get, avg cycles: 310
1 byte put claim-finish, avg cycles: 217
5 byte put claim-finish, avg cycles: 261
5 byte get claim-finish, avg cycles: 258
 PASS - test_ringbuffer_performance in 0.19 seconds
===================================================================
START - test_ringbuffer_concurrent

    Assertion failed at WEST_TOPDIR/zephyr/tests/lib/ringbuffer/src/concurrent.c:72: data_read: memcmp(outpu0

 FAIL - test_ringbuffer_concurrent in 0.14 seconds
===================================================================
START - test_ringbuffer_shpsc
preempted: 1152

    Assertion failed at WEST_TOPDIR/zephyr/tests/lib/ringbuffer/src/concurrent.c:373: thread_entry_spsc: (pr)
If thread operation was not preempted multiple times then we cannot have confidance that it validated the moe
 FAIL - test_ringbuffer_shpsc in 5.27 seconds
===================================================================
START - test_ringbuffer_spshc
preempted: 1734
 PASS - test_ringbuffer_spshc in 5.2 seconds
===================================================================
START - test_ringbuffer_cpy_shpsc
preempted: 2552
 PASS - test_ringbuffer_cpy_shpsc in 5.2 seconds
===================================================================
START - test_ringbuffer_cpy_spshc
preempted: 3076
 PASS - test_ringbuffer_cpy_spshc in 5.1 seconds
===================================================================
START - test_ringbuffer_item_shpsc
preempted: 3595
 PASS - test_ringbuffer_item_shpsc in 5.1 seconds
===================================================================
START - test_ringbuffer_item_spshc
preempted: 3948
 PASS - test_ringbuffer_item_spshc in 5.1 seconds
===================================================================
Test suite test_ringbuffer_api failed.
===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: (Linux)
 - Toolchain (Zephyr SDK)
 - Commit SHA (c94e53c9f5b6be29ce42b3cc8b50c6a27a487246)

yerabolu, 2021-08-24:
fails on up_square, EHL_CRB and Microchip as well.

enjiamai, 2021-08-25:
fails on ACRN ehl_crb as well.

IRISZZW, 2021-08-30:
relative issues #38093 #37768

evgeniy-paltsev, 2021-09-25:
Fixed by #38817

closed 2021-09-25 by evgeniy-paltsev