zephyrproject-rtos/zephyr #37862: uart ns16550 driver add DesignWare APB Busy Detect interrupt support

IRISZZW, 2021-08-23:
**Is your enhancement proposal related to a problem? Please describe.**

The DesignWare APB UART has a feature which causes a new Busy Detect interrupt to be generated
if it's busy when the LCR is written. 

When Busy Detect interrupt generated, we need to read the UART status register to reset it.

Similar fix in Linux kernel: https://github.com/torvalds/linux/commit/beab697ab4b2962e3d741b476abe443baad0933d

relative issue; #36254

IRISZZW, 2021-08-23:
@dcpleung @nashif @jenmwms @aasthagr this interrupt not exist in standard ns16550 uart, is there any suggestion how to add the support for it?

dcpleung, 2021-08-24:
TBH, I wonder if we should separate the current 16550 driver into two: one for basic 16550, and one for DesignWare:
* Fewer #ifdef in 16550 driver for DW features.
* No need for all those macros for I/O port access in DW driver.
* Easier to extend DW driver, without the need to re-verify basic 16550 functionality.

abrodkin, 2021-09-07:
AFAIK an issue in question was only seen on ARC HSDK (v1 or original) board when Zephyr application that uses UART IRQs was started after auto-started (by boot-ROM) U-Boot. In other words if U-Boot auto-start is disabled by a DIP-switch (or equally U-Boot image was never programmed in the on-board SPI flash) this problem won't be reproduced.