zephyrproject-rtos/zephyr #37865: nRF Battery measurement issue

gabbas1, 2021-08-23:
Hello guys @galak @pabigot @nashif @carlescufi @tbursztyka ,

**Describe the bug**
According to the [nRF-Battery-Voltage-Sample](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/boards/nrf/battery) it is possible to use a voltage divider (vbatt). Compiling this, with the suggested board works fine:
` zephyr/samples/boards/nrf/battery $ west build -p auto -b thingy52_nrf52832 ./ --pristine`

 But if you comment the vbat-node in the devicetree, or just set its status to `status="disabled"` there is an compiler error:

`zephyr/include/generated/devicetree_unfixed.h:1344:43: warning: initialization of 'unsigned char' from 'char *' makes integer from pointer without a cast [-Wint-conversion]
 1344 | #define DT_N_S_soc_S_adc_40007000_P_label "ADC_0" `

(see full log below)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to `zephyr/samples/boards/nrf/battery`
2. west build -p auto -b thingy52_nrf52832
3. it works
4. In `zephyr/boards/arm/thingy52_nrf52832´ add status to thingy52_nrf52832.dts:
```
vbatt {
        status="disabled";
        compatible = "voltage-divider";
	io-channels = <&adc 4>;
	output-ohms = <180000>;
	full-ohms = <(1500000 + 180000)>;
	power-gpios = <&sx1509b 4 0>;
};
``` 
5. Try to build again: `west build -p auto -b thingy52_nrf52832 --pristine`
6. see the error

**Expected behavior**
I want to make a battery-voltage-measurement on the nrf52840DK/dongle and this one does not use/need a voltage divider, that is why there is no "vbatt" in their devicetree. It should work, since it is also an nRF-Chip and I did it before with other RTOS'.
Nevertheless, I think there must be an problem in the battery library, since it is not even working with the suggested board (thingy52_nrf52832).

**Logs and console output**
```
[98/155] Building C object CMakeFiles/app.dir/src/battery.c.obj
FAILED: CMakeFiles/app.dir/src/battery.c.obj 
ccache /home/smiethe/Programme/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-gcc -DBUILD_VERSION=zephyr-v2.6.0-406-g9530d3237661 -DKERNEL -DNRF52832_XXAA -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I/home/smiethe/git/zephyrproject1/zephyr/include -Izephyr/include/generated -I/home/smiethe/git/zephyrproject1/zephyr/soc/arm/nordic_nrf/nrf52 -I/home/smiethe/git/zephyrproject1/modules/hal/cmsis/CMSIS/Core/Include -I/home/smiethe/git/zephyrproject1/modules/hal/nordic/nrfx -I/home/smiethe/git/zephyrproject1/modules/hal/nordic/nrfx/drivers/include -I/home/smiethe/git/zephyrproject1/modules/hal/nordic/nrfx/mdk -I/home/smiethe/git/zephyrproject1/zephyr/modules/hal_nordic/nrfx/. -isystem /home/smiethe/git/zephyrproject1/zephyr/lib/libc/minimal/include -isystem /home/smiethe/Programme/gcc-arm-none-eabi-9-2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/include -isystem /home/smiethe/Programme/gcc-arm-none-eabi-9-2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/include-fixed -Os -imacros /home/smiethe/git/zephyrproject1/zephyr/samples/boards/nrf/battery/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -imacros /home/smiethe/git/zephyrproject1/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-address-of-packed-member -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/smiethe/git/zephyrproject1/zephyr/samples/boards/nrf/battery=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/smiethe/git/zephyrproject1/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/smiethe/git/zephyrproject1=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT CMakeFiles/app.dir/src/battery.c.obj -MF CMakeFiles/app.dir/src/battery.c.obj.d -o CMakeFiles/app.dir/src/battery.c.obj -c ../src/battery.c
In file included from /home/smiethe/git/zephyrproject1/zephyr/include/devicetree.h:19,
                 from /home/smiethe/git/zephyrproject1/zephyr/include/arch/arm/aarch32/arch.h:20,
                 from /home/smiethe/git/zephyrproject1/zephyr/include/arch/cpu.h:19,
                 from /home/smiethe/git/zephyrproject1/zephyr/include/kernel_includes.h:33,
                 from /home/smiethe/git/zephyrproject1/zephyr/include/kernel.h:17,
                 from /home/smiethe/git/zephyrproject1/zephyr/include/zephyr.h:18,
                 from ../src/battery.c:12:
zephyr/include/generated/devicetree_unfixed.h:1344:43: warning: initialization of 'unsigned char' from 'char *' makes integer from pointer without a cast [-Wint-conversion]
 1344 | #define DT_N_S_soc_S_adc_40007000_P_label "ADC_0"
      |                                           ^~~~~~~
/home/smiethe/git/zephyrproject1/zephyr/include/devicetree.h:2326:24: note: in expansion of macro 'DT_N_S_soc_S_adc_40007000_P_label'
 2326 | #define DT_CAT(a1, a2) a1 ## a2
      |                        ^~
/home/smiethe/git/zephyrproject1/zephyr/include/devicetree.h:527:32: note: in expansion of macro 'DT_CAT'
  527 | #define DT_PROP(node_id, prop) DT_CAT(node_id, _P_##prop)
      |                                ^~~~~~
/home/smiethe/git/zephyrproject1/zephyr/include/devicetree.h:648:27: note: in expansion of macro 'DT_PROP'
  648 | #define DT_LABEL(node_id) DT_PROP(node_id, label)
      |                           ^~~~~~~
../src/battery.c:74:3: note: in expansion of macro 'DT_LABEL'
   74 |   DT_LABEL(DT_NODELABEL(adc)),
      |   ^~~~~~~~
/home/smiethe/git/zephyrproject1/zephyr/include/devicetree.h:2326:24: note: in expansion of macro 'DT_N_NODELABEL_adc'
 2326 | #define DT_CAT(a1, a2) a1 ## a2
      |                        ^~
/home/smiethe/git/zephyrproject1/zephyr/include/devicetree.h:173:29: note: in expansion of macro 'DT_CAT'
  173 | #define DT_NODELABEL(label) DT_CAT(DT_N_NODELABEL_, label)
      |                             ^~~~~~
../src/battery.c:74:12: note: in expansion of macro 'DT_NODELABEL'
[... 62 lines cut ...]
../src/battery.c:74:3: note: in expansion of macro 'DT_LABEL'
   74 |   DT_LABEL(DT_NODELABEL(adc)),
      |   ^~~~~~~~
/home/smiethe/git/zephyrproject1/zephyr/include/devicetree.h:2326:24: note: in expansion of macro 'DT_N_NODELABEL_adc'
 2326 | #define DT_CAT(a1, a2) a1 ## a2
      |                        ^~
/home/smiethe/git/zephyrproject1/zephyr/include/devicetree.h:173:29: note: in expansion of macro 'DT_CAT'
  173 | #define DT_NODELABEL(label) DT_CAT(DT_N_NODELABEL_, label)
      |                             ^~~~~~
../src/battery.c:74:12: note: in expansion of macro 'DT_NODELABEL'
   74 |   DT_LABEL(DT_NODELABEL(adc)),
      |            ^~~~~~~~~~~~
[107/155] Building C object zephyr/drivers/regulator/CMakeFiles/drivers__regulator.dir/regulator_fixed.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/smiethe/git/zephyrproject1/zephyr/samples/boards/nrf/battery/build

```

**Environment**
 - OS: Linux/Gentoo
 - Toolchain: gcc, West, Zephyr-v2.6.0-406-g9530d3237661

Thanks for your help.
Best regards,
Sebastian

carlescufi, 2021-09-15:
@jfischer-no to investigate in the 2.7.0 timeframe.

ck-telecom, 2021-09-22:
Is it reasonable to rewrite a sensor driver implemented SENSOR_CHAN_VOLTAGE?

gmarull, 2021-09-23:
After some initial investigation, I can certify that the sample works fine as it is when using a `thingy52_nrf52832` board. Steps I've followed:

1. Connect the board (battery, serial adapter to P0.02/0.03, programmer)
2. Build the sample and flash it: `west build -b thingy52_nrf52832 samples/boards/nrf/battery`, `west flash`
3. Observe serial output, e.g. `[0:09:51.117]: 4190 mV; 10000 pptt`
4. Measure battery voltage with a multimeter, to check if the printed value is in line with the measurement:
[image]

The problem seems to be that the sample claims it can work without the `vbatt` node in Devicetree (i.e. no voltage divider), but it doesn't seem to be the case. I'll see how this path can be fixed.

PR #38771 (2021-09-23): samples: boards: nrf: battery: fix usage without voltage divider
The nRF battery sample claims that it works without the need of a vbatt
node (voltage divider). However, this was currently not possible because
the sample code did have no means to extract the necessary ADC details
(instance and channel) from Devicetree unless vbatt was present. Similar
to other ADC samples, the zephyr,user node is used to define io-channels
property in this case. The sample README has been updated with an
example of how measurement without a voltage divider can be achieved.

Fixes #37865
closed 2021-09-23 by nashif