zephyrproject-rtos/zephyr #37882: SHUTDOWN mode support for STM32WB5x microcontrollers

r2r0, 2021-08-23:
**Is your feature request related to a problem? Please describe.**
It will be very useful for low power embedded devices based on STM32WB5x microcontrollers to use SHUTDOWN (HW) low power mode.

**Describe the solution you'd like**
My proposal for implementation of this feature in Zephyr is adding switching to one of the most power saving power states i.e. PM_DEVICE_STATE_SUSPENDED in meaning of transition to SHUTDOWN MCU state.

**Describe alternatives you've considered**
I haven't found good alternative till now beside change of device HW design (by adding HW power gating and special wakeup circuit) which seems unreasonable in case when MCU has already well designed SHUTDOWN mode.

**Additional context**
It is possible now to "manually" enter SHUTDOWN mode on this MCU but it requires disabling (at compile time) BLE support in Zephyr which makes a little sense for microcontroller with embedded BLE stack. With BLE enabled (and second core enabled) it is not possible to use standard shutdown procedure from ST low level API.
This feature can be also useful for STM32WL55 family.

FRASTM, 2021-08-24:
In order to disable the BLE support before going to shutdown, could the CPU2 reset help : setting the Low-power mode selection for CPU2 and keeping the C2BOOT bit of the PWR_CR4 register low

r2r0, 2021-09-02:
Unfortunately I cannot find a proper way to implement entering SHUTDOWN mode when BLE is enabled in Zephyr.
I tried LL_C2_PWR_SetPowerMode then LL_PWR_DisableBootC2 and then trying to use shci_send to send reset command to CPU2 but without luck.

r2r0, 2021-09-02:
Regarding wake up - we use WKUP pin event functionality and it works very well for us.

Sure no problem, see below - goto_shutdown is called from pm_power_state_set when PM_STATE_SOFT_OFF is requested.

```C
static void prepare_clock_for_sleep(void)
{
    /* Implementation of STM32 AN5289 algorithm to enter/exit lowpower */
    z_stm32_hsem_lock(CFG_HW_RCC_SEMID, HSEM_LOCK_WAIT_FOREVER);
    if (!LL_HSEM_1StepLock(HSEM, CFG_HW_ENTRY_STOP_MODE_SEMID)) {
        if (LL_PWR_IsActiveFlag_C2DS()) {
            /* Release ENTRY_STOP_MODE semaphore */
            LL_HSEM_ReleaseLock(HSEM, CFG_HW_ENTRY_STOP_MODE_SEMID, 0);

            /* The switch on HSI before entering Stop Mode is required */
            switch_on_hsi();
        }
    } else {
        /* The switch on HSI before entering Stop Mode is required */
        switch_on_hsi();
    }
}

#define ACI_HAL_STACK_RESET 0xFC3B

static void goto_shutdown(void)
{
    LL_C2_PWR_SetPowerMode(LL_PWR_MODE_SHUTDOWN);
    LL_PWR_DisableBootC2();

    TL_EvtPacket_t p_rsp;

    shci_send(ACI_HAL_STACK_RESET, 0, 0, &p_rsp);

#if 0
    if ((SHCI_CmdStatus_t)(((TL_CcEvt_t*)(p_rsp.evtserial.evt.payload))->payload[0]) != SHCI_Success)
    {
        LOG_ERR("Cannot send ACI_HAL_STACK_RESET command, error: %d", err);
        return;
    }
#endif
    
#if 0
    /* Wait till C2DS set */
    while(LL_PWR_IsActiveFlag_C2DS() == 0);

    /* Re-init CPU2 */
    SHCI_CmdStatus_t sret = SHCI_C2_Reinit();
    if (sret != SHCI_Success)
    {
        LOG_ERR("Cannot reinit CPU2, error: %d", sret);
    }

    /* Wait till C2DS set */
    while(LL_PWR_IsActiveFlag_C2DS() == 0);
#endif

    prepare_clock_for_sleep();

    /* Clear all Wake-Up flags */
    LL_PWR_ClearFlag_WU();

    LL_C2_PWR_SetPowerMode(LL_PWR_MODE_SHUTDOWN);
    LL_PWR_SetPowerMode(LL_PWR_MODE_SHUTDOWN);

    z_stm32_hsem_unlock(CFG_HW_RCC_SEMID);
    LL_LPM_EnableDeepSleep();
    k_cpu_idle();
}
```

erwango, 2021-09-02:
Thanks for sharing, I'll have a try on my side.
May I know this is not working as expected (behavior vs expected behavior) ?

r2r0, 2021-09-02:
Present code sample does not switch CPU to SHUTDOWN and CPU continues execution after goto_shutdown.
Expected behavior is to enter SHUTDOWN mode (power consumption at level below 10 uamps) and be able to to wake up (with reset) using WKUP pin event.
Below code we use to arm WKUP pin.
```C
    LL_PWR_DisableWakeUpPin(PWR_CR3_EWUP3);
    LL_PWR_ClearFlag_WU();
    LL_PWR_SetWakeUpPinPolarityLow(PWR_CR3_EWUP3);
    LL_PWR_EnableWakeUpPin(PWR_CR3_EWUP3);
    LL_PWR_ClearFlag_WU();
```

erwango, 2021-09-03:
@r2r0 Please have a look to https://github.com/erwango/zephyr/tree/dev_wb_ble_shutdown.
I think I'm able to properly shutdown BLE stack and both cores with the modification done in power.c file.
I've added a quick sample as demo.

I'm getting following consumption (4 sec advetising + shutdown)
[image]
Board reset button to exit shutdown state.

Can you have a test with this code ?

r2r0, 2021-09-06:
@erwango I confirm that sample in branch you mentioned works well  - after few seconds MCU switches to SHUTDOWN mode.

erwango, 2021-09-06:
> @erwango I confirm that sample in branch you mentioned works well - after few seconds MCU switches to SHUTDOWN mode.

Thanks for confirming. I'll push a PR soon. Though, it won't be merge before few weeks now that merge window is closed.

closed 2021-11-09 by cfriedt

busadhruv, 2024-09-04:
> @r2r0 Please have a look to https://github.com/erwango/zephyr/tree/dev_wb_ble_shutdown. I think I'm able to properly shutdown BLE stack and both cores with the modification done in power.c file. [...]

"https://github.com/erwango/zephyr/tree/dev_wb_ble_shutdown." link shows not found

erwango, 2024-09-04:
Please check https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/boards/stm32/power_mgmt/stm32wb_ble

PR #38980 (2021-11-09): stm32wb: ble core shutdown
Add shutdown procedure to stm32wb power management,
including clean ble stack reset.

Add a sample to demonstrate conjuction of ble and pm operations.

Fixes #37882