zephyrproject-rtos/zephyr #37883: P-NUCLEO-WB55RG Mesh Bluetooth Sample Failed to start advertiser

sundayglee, 2021-08-23:
**Running the samples/bluetooth/mesh in P-NUCLEO-WB55RG fails with `Failed` to start advertiser error`**`
I am trying to run the samples/bluetooth/mesh on a P-NUCLEO-WB55RG using the latest Zephyr 0.13 without modifying any of the configuration option but the P-NUCLEO-WB55RG advertises itself and disconnect disconnect every few seconds with the error `Failed to start advertiser`

**To Reproduce**
Steps to reproduce the behavior:
1. cd zephyr
2. west build -p auto -b nucleo_wb55rg samples/bluetooth/mesh
3. west flash

**Expected behavior**
The P-Nucleo-WB55rg should advertise successfully and get detected by nRF Mesh application

**Impact**
Can't run mesh application on STM32WBCG and STM32WBRG micro-controllers

**Logs and console output**
`*** Booting Zephyr OS build zephyr-v2.6.0-2332-gc910138adec2  ***
Initializing...
Bluetooth initialized
Mesh initialized
[00:00:00.346,000] <inf> fs_nvs: 3 Sectors of 4096 bytes
[00:00:00.346,000] <inf> fs_nvs: alloc wra: 0, fd0
[00:00:00.346,000] <inf> fs_nvs: data wra: 0, 20
[00:00:00.364,000] <inf> bt_hci_core: No ID address. App must call settings_load()
[00:00:00.367,000] <inf> bt_hci_core: Identity: 80:E1:26:1B:69:79 (public)
[00:00:00.367,000] <inf> bt_hci_core: HCI: version 5.0 (0x09) revision 0x0026, manufacturer 0x0030
[00:00:00.367,000] <inf> bt_hci_core: LMP: version 5.0 (0x09) subver 0x2126
[00:00:00.367,000] <inf> bt_mesh_prov_device: Device UUID: 00000000-2a00-2e00-1050-4e5457323220
[00:00:00.378,000] <wrn> bt_hci_core: opcode 0x200a status 0x86
[00:00:00.378,000] <err> bt_adv: Failed to start advertiser
[00:00:00.389,000] <wrn> bt_hci_core: opcode 0x200a status 0x85
[00:00:00.389,000] <err> bt_adv: Failed to start advertiser
[00:00:00.389,000] <err> bt_mesh_adv_legacy: Advertising failed: err -5
[00:00:05.390,000] <wrn> bt_hci_core: opcode 0x200a status 0x85
[00:00:05.390,000] <err> bt_adv: Failed to start advertiser
[00:00:05.390,000] <err> bt_mesh_adv_legacy: Advertising failed: err -5`

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Version: 0.13

**Additional context**
When scanning for the node using nRF Connect application, it shows momentary and disappear when trying to add it as a node. The issue keeps repeating every few seconds.

erwango, 2021-08-25:
@sundayglee Did you had a try with latest M0 BLE controller f/w (1.12.0) which support was added quite recently ?

sundayglee, 2021-08-25:
@erwango Yes, I tested it with M0 BLE f/w version 1.12.1. And did test also with M0 BLE f/w version 1.9.0. Both were stm32wb5x_BLE_Stack_full_fw firmware.

Generally HCI interface is working fine since I tested it with the samples/bluetooth/st_ble_sensor and it works fine.
All the other normal Bluetooth samples are working except the Mesh specific ones.

Also, there are some instances were running a mesh sample, the program terminates with BUS FAULT error. Setting in the project file CONFIG_SETTINGS=n and recompiling the code works and then the error Advertising Failed(specifically  bt_mesh_adv_legacy) as the error shows.
This error keeps on repeating every few milliseconds, meaning there must be some checks that fails after few milliseconds that subsequently disables advertising. The process keeps on repeating over and over.

sundayglee, 2021-08-28:
@erwango Thank you.

I also tested the sample with a custom board containing STM32WB55CG microcontroller, and it gives the same errors.

erwango, 2021-08-31:
@sundayglee I've open an internal issue with the trace you provided.
It may take few time before we get an answer, I'll keep you posted in this thread.

sundayglee, 2021-09-01:
@erwango thank you much.

erwango, 2021-11-03:
@sundayglee Here is experts' feedback:

In fact SCAN Interval and SCAN Windows should not be the same.
Meaning some bandwidth should be given to the advertising.
Thus more than 20msec is requested between these two values
In case SCAN WINDOW is fixed to 0x40
        SCAN INTERVAL should be set to 0x80
or in case SCAN WINDOW is fixed to 0x400
           SCAN INTERVAL should be set to 0x440
For advertising parameters, min could stay to ADV min=20ms, while adv Max should be set to 2 times SCAN INTERVAL.

Example:
If customer request a SCAN WINDOW =1000 then SCAN INTERVAL --> 1040
ADV min = 20
ADV Max --> 2080

Would you be able to try to test this ?

Marco-Peter, 2021-12-12:
I also ran into the same issue.

Using the timings provided by @erwango works in principle.

The problem now is that many - if not all - timings in BT Mesh operation are hardcoded into the Bluetooth code.

Scanning interval and window are taken from [subsys/bluetooth/mesh/adv.h](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/mesh/adv.h), line 16 and 17:
```
#define BT_MESH_SCAN_INTERVAL_MS 30
#define BT_MESH_SCAN_WINDOW_MS 30
```

GATT provisioning bearer timings are taken from [include/bluetooth/gap.h](https://github.com/zephyrproject-rtos/zephyr/blob/main/include/bluetooth/gap.h) lines 82..85.
```
#define BT_GAP_PER_ADV_FAST_INT_MIN_2           0x0050  /* 100 ms   */
#define BT_GAP_PER_ADV_FAST_INT_MAX_2           0x0078  /* 150 ms   */
#define BT_GAP_PER_ADV_SLOW_INT_MIN             0x0320  /* 1 s      */
#define BT_GAP_PER_ADV_SLOW_INT_MAX             0x03C0  /* 1.2 s    */
```
(used in function `bt_mesh_pb_gatt_adv_start()`)

The advertising bearer (PB-ADV) defines its timings in [subsys/bluetooth/mesh/adv_legacy.c](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/mesh/adv_legacy.c) lines 95..96:
```
	param.interval_min = BT_MESH_ADV_SCAN_UNIT(adv_int);
	param.interval_max = param.interval_min;
```

All those timings cause an error, when sent to the stack (status 0x85 or 0x86) during startup of the advertiser.
By manually overriding all those timings, I at least could make it into the configuration stage.

Since I just started using Bluetooth myself, I absolutely have no idea about the best way to proceed from here.

erwango, 2021-12-13:
Thanks @Marco-Peter for the tests.

@jhedberg cf previous comment. How would you suggest to proceed in order to help progressing on this topic. Would it be acceptable for us to propose configurable or alternate timings for BT Mesh ?

jhedberg, 2021-12-13:
@erwango scan interval == scan window is known as "continuous scanning" and IIRC the mesh spec either implies or explicitly recommends it. So far the way radio scheduling has been implemented in other controllers hasn't caused any similar issues. @trond-snekvik @cvinayak any comments here?

trond-snekvik, 2021-12-13:
+1 for continuous scanning, this should be the way all controllers do scanning with mesh.

> Would it be acceptable for us to propose configurable or alternate timings for BT Mesh ?

That makes sense to me

jhedberg, 2021-12-13:
Regarding the question of whether these should be made configurable through Kconfig - I'm not opposed to that.

erwango, 2022-01-24:
Moving this as enhancement regarding the nature of problem identified and change requested.

cvinayak, 2022-01-25:
> Regarding the question of whether these should be made configurable through Kconfig - I'm not opposed to that.

As the issue at hand is Controller specific, should this be handled as a quirk bit in the HCI driver (Zephyr controller specific is handled for example here: https://github.com/zephyrproject-rtos/zephyr/blob/140837ce3ac6ba6ac01d804ebd40c8aaf00a3011/subsys/bluetooth/controller/hci/hci_driver.c#L735

erwango, 2022-07-26:
@sundayglee With the latest STM32WB Cube BLE firmware V1.14.0, MESH is now functional. This should be available with upcoming https://github.com/zephyrproject-rtos/zephyr/pull/48160.
Please use stm32wb5x_BLE_HCILayer_fw.bin.

erwango, 2022-08-24:
Fixed following merge of #48160

closed 2022-08-24 by erwango