zephyrproject-rtos/zephyr #37915: led_pwm driver not generating correct linker symbol

riggs, 2021-08-25:
**Describe the bug**
The `led_pwm` driver isn't generating the matching linker symbol defined by the `DEVICE_DT_GET` and `DT_PATH` macros referencing a node in the devicetree.

**To Reproduce**
Alter `samples/drivers/led_pwm/src/main.c`, line 111:
```
-       led_pwm = device_get_binding(LED_PWM_DEV_NAME);
+       led_pwm = DEVICE_DT_GET(DT_PATH(pwmleds));  // Node path for nrf52840dk_52840
```
Build project.

On the nRF52840DK, the linker errors with `undefined reference to '__device_dts_ord_22'`. 

**Expected behavior**
Correct linking to driver-created structure.

**Impact**
`led_pwm` driver is unusable with unlabeled DT nodes.

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain: NCS, arm-none-eabi-gcc (9-2019-q4-major) 9.2.1 20191025 
 - v2.6.0-ncs1-rc3 (Kconfig errors in upstream w/ the nRF52840DK)

**Additional context**
Work-around is to add a `label = "node_label"` and use `device_get_binding`, but that can't be statically initialized.

mbolivar-nordic, 2021-09-09:
@riggs sorry for the delay and thanks for taking the time to file an issue.

**TL;DR it looks like you are using an old NCS version which can't combine DEVICE_DT_GET with this driver's devices.**

Your options are:

- cherry-pick df01148aebb43da1d5d005150c980a08157e1757 onto your zephyr tree, or
- use mainline zephyr, or
- wait for NCS v1.7.0

Details in a Dickensian "Tale of Too Many Zephyrs":

This does not reproduce for me on today's mainline zephyr master (I tested with f79fbeabd58b713c70af0a657c7a7b1456915cd8).

```
$ git diff
diff --git a/samples/drivers/led_pwm/src/main.c b/samples/drivers/led_pwm/src/main.c
index fbee89c1d1..1d17a4a054 100644
--- a/samples/drivers/led_pwm/src/main.c
+++ b/samples/drivers/led_pwm/src/main.c
@@ -109,7 +109,7 @@ void main(void)
        const struct device *led_pwm;
        uint8_t led;
 
-       led_pwm = device_get_binding(LED_PWM_DEV_NAME);
+       led_pwm = DEVICE_DT_GET(DT_PATH(pwmleds));
        if (led_pwm) {
                LOG_INF("Found device %s", LED_PWM_DEV_NAME);
        } else {

$ west build -b nrf52840dk_nrf52840
[3/9] Linking C executable zephyr/zephyr_prebuilt.elf

[9/9] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       28488 B         1 MB      2.72%
            SRAM:        7840 B       256 KB      2.99%
        IDT_LIST:          0 GB         2 KB      0.00%
```

> v2.6.0-ncs1-rc3 (Kconfig errors in upstream w/ the nRF52840DK)

That zephyr version looks like the one from NCS v1.6.0-rc3: https://github.com/nrfconnect/sdk-nrf/commit/c815d9404ec61536543fa08a6163c0bfd893eb12.

Note that this zephyr tag is not an upstream one, and NCS v1.6.0 final was also already released, so

- this downstream codebase should not be used to file upstream Zephyr bug reports (i.e. here in zephyrproject-rtos/zephyr, "there is no NCS"), and 
- an rc3 tree shouldn't be used to file downstream bug reports either, since the final release tree is out

This issue does reproduce on NCS v1.6.0 final.... but: I took a look at the driver for that version of Zephyr, and found this: https://github.com/nrfconnect/sdk-zephyr/blob/a62ea8fa297a12c2d17332218ff45c1e54c55a8e/drivers/led/led_pwm.c#L235

That is old code which predates zephyr commit df01148aebb43da1d5d005150c980a08157e1757, which changed the DEFINE_DEFINE to the DT-aware DEVICE_DT_INST_DEFINE macro, making the device accessible with DEVICE_DT_GET. You can cherry-pick that commit onto your local tree and retest; it resolves the issue.

I also went ahead and tested on NCS v1.7.0-rc2, which I cut yesterday (https://github.com/nrfconnect/sdk-nrf/releases/tag/v1.7.0-rc2). DEVICE_DT_GET works as expected out of the box; no link error. So you should be good to go with NCS once v1.7.0 is out as well.

closed 2021-09-09 by mbolivar-nordic

riggs, 2021-09-09:
Thanks for the thorough reply! I'm planning to settle on the LTS release, when it's available, so I'm happy to be able to migrate to the updated NCS.