zephyrproject-rtos/zephyr #37927: tests-ci: net-lib: test/net/lib : build missing drivers__net and application can not run

hakehuang, 2021-08-25:
**Describe the bug**
build warns 
```
CMake Warning at /build/src/workspace/mimxrt1060_evk_samples_master/zephyr/CMakeLists.txt:718 (message):
  No SOURCES given to Zephyr library: drivers__interrupt_controller
  Excluding target from build.
CMake Warning at /build/src/workspace/mimxrt1060_evk_samples_master/zephyr/CMakeLists.txt:718 (message):
  No SOURCES given to Zephyr library: drivers__net
  Excluding target from build.
```

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=mimxrt1050_evk
3. make
4. See error

**Expected behavior**
test PASS

**Impact**
all net work related cases

**Logs and console output**

```
CMake Warning at /build/src/workspace/mimxrt1060_evk_samples_master/zephyr/CMakeLists.txt:718 (message):

  No SOURCES given to Zephyr library: drivers__interrupt_controller

  Excluding target from build.

CMake Warning at /build/src/workspace/mimxrt1060_evk_samples_master/zephyr/CMakeLists.txt:718 (message):

  No SOURCES given to Zephyr library: drivers__net

  Excluding target from build.
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux,)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: zephyr-v2.6.0-2462-g93e674e37382

hakehuang, 2021-08-25:
all RT1xxx board have same problem

hakehuang, 2021-08-25:
@MaureenHelm @dleach02 @mmahadevan108

ibirnbaum, 2021-09-07:
I'm getting the drivers__net related warning with a QEMU target I'm currently working on.

It seems like this is caused by the fact that CONFIG_NETWORKING as the top-level switch for all networking features leads to the inclusion of drivers/net/CMakeLists.txt via drivers/CMakeLists.txt.

The CMakeLists.txt in drivers/net contains three conditional source file inclusions: loopback.c for CONFIG_NET_LOOPBACK, slip.c for CONFIG_SLIP and ppp.c for CONFIG_PPP. If neither of these features is used (in my case, I don't need the loopback interface and I'm using QEMU native networking instead of PPP or SLIP), the warning "No SOURCES given to Zephyr library: drivers__net" appears when building an image. If I enable the loopback interface, the warning disappears.

Maybe the top-level networking feature switch should not be associated with conditional source file inclusions relating to three features that will likely in many cases not be enabled.

dleach02, 2021-09-13:
@hakehuang can you retest after #38473 was merged?

hakehuang, 2021-09-14:
@dleach02  issue fixed in v2.7.0-rc1-95-gdefda27969

closed 2021-09-14 by hakehuang