zephyrproject-rtos/zephyr #38090: LPS22HH: int32_t overflow in pressure calculations

sanchox, 2021-08-29:
Conversion in `lps22hh_press_convert` produce int32_t overflow and negative value in `sensor_value.val2` 
```C
static inline void lps22hh_press_convert(struct sensor_value *val,
					 int32_t raw_val)
{
	int32_t press_tmp = raw_val >> 8; /* raw value is left aligned (24 msb) */

	/* Pressure sensitivity is 4096 LSB/hPa */
	/* Also convert hPa into kPa */

	val->val1 = press_tmp / 40960;
	val->val2 = (press_tmp % 40960) * 1000000 / 40960;
}
```
Overflow happens when `(press_tmp % 40960) * 1000000` calculated.

Example of calculations where overflow occurs:
```
raw = 1065675008 (0x3f84e900)
press_tmp = 4162793 (0x3f84e9)
val1 = 4162793 / 40960 = 101
val2 = (4162793 % 40960) * 1000000 / 40960 = 
     = (25833)           * 1000000 / 40960 =
     = 25833000000                 / 40960 = -417887
         ^ this value is greater than int32_t max positive value
```

avisconti, 2021-09-03:
@sanchox can you possibly take a look to this fix? Thanks!

sanchox, 2021-09-03:
Looks OK for me =)

PR #38304 (2021-09-06): drivers/sensor: lps22hh: Fix int32 overflow in the val2 part
The val2 calculation was done using (1000000 / 40960) as
multiplying factor, which was sometimes leading to a
int32 overflow. So, let's use the equivalent (but smaller)
(3125 / 128).

Fixes #38090
commit c3050a5a (2021-09-06): drivers/sensor: lps22hh: Fix int32 overflow in the val2 part

The val2 calculation was done using (1000000 / 40960) as
multiplying factor, which was sometimes leading to a
int32 overflow. So, let's use the equivalent (but smaller)
(3125 / 128).

Fix #38090
closed 2021-09-06 by cfriedt