zephyrproject-rtos/zephyr #38104: kheap buffer own section attribute causing memory overflow in ESP32

sylvioalves, 2021-08-30:
**Describe the bug**
After this d92d1f162af3ba24963f1026fc0a304f1a44d1f3, ESP32 can't build a few samples as dram0_1 memory overflows.

it might be related to `__FILE__` but it needs some check:
```
+#define __in_section_unique_named(seg, name) \
+	___in_section(seg, __FILE__, name)
```

**To Reproduce**
Steps to reproduce the behavior:
1. rm -rf build; 
2. west build -p -b esp32 samples/boards/esp32/wifi_station
3. See error

```
/home/sylvio/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.bfd: zephyr/zephyr_prebuilt.elf section `noinit' will not fit in region `dram0_1_seg'
/home/sylvio/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.bfd: region `dram0_1_seg' overflowed by 71136 bytes
```

**Expected behavior**
Build complete.

uLipe, 2021-08-30:
@sylvioalves , @dcpleung  it seems the problem will occur because now the `K_HEAP_DEFINE` will place data on the `noinit `section, and esp32 linker uses a smaller `DRAM` area to gather all things that goes to there, adding the heaps, causes this `DRAM` segment explosion.

dcpleung, 2021-08-30:
The kheap buffers are named like this `.noinit."<source file name>".kheap_buf_<variable name>`. Maybe try specifying them in linker script under BSS section? Would that work for you?

sylvioalves, 2021-08-30:
Sure, it would work. It also came to me that default system heap set as default is too big. I will change that.

closed 2021-08-30 by sylvioalves