zephyrproject-rtos/zephyr #38108: nxp: usb driver build failure due to d92d1f162af3ba24963f1026fc0a304f1a44d1f3

hakehuang, 2021-08-30:
**Describe the bug**
```
error: section of 'kheap_ep_buf_pool' conflicts with previous declaration
5042 |       kheap_##name[MAX(bytes, Z_HEAP_MIN_SIZE)];  \
      |       ^~~~~~
/home/shared/disk/zephyr_project/zephyr_test/zephyr/drivers/usb/device/usb_dc_mcux.c:71:11: note: in expansion of macro 'K_HEAP_DEFINE'
   71 | __nocache K_HEAP_DEFINE(ep_buf_pool, 1024 * EP_BUF_NUMOF_BLOCKS);
      |           ^~~~~~~~~~~~~
[78/107] Building C object zephyr/kernel/CMakeFiles/kernel.dir/init.c.obj
```

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=mimxrt1060_evk ..
3. make
4. See error

**Expected behavior**
build pass

**Impact**
use functions

**Logs and console output**

```

error: section of 'kheap_ep_buf_pool' conflicts with previous declaration
5042 |       kheap_##name[MAX(bytes, Z_HEAP_MIN_SIZE)];  \
      |       ^~~~~~
/home/shared/disk/zephyr_project/zephyr_test/zephyr/drivers/usb/device/usb_dc_mcux.c:71:11: note: in expansion of macro 'K_HEAP_DEFINE'
   71 | __nocache K_HEAP_DEFINE(ep_buf_pool, 1024 * EP_BUF_NUMOF_BLOCKS);
      |           ^~~~~~~~~~~~~
[78/107] Building C object zephyr/kernel/CMakeFiles/kernel.dir/init.c.obj
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used:  zephyr-v2.6.0-2723-ge7436ea862

**Additional context**
bisect found the issue is intruduced by below commit
```
Author: Daniel Leung <daniel.leung@intel.com>
Date:   Mon Jul 19 14:09:25 2021 -0700

    kernel: each kheap buffer has its own section attribute

    This attaches a unique section attribute for each kheap
    buffer defined with K_HEAP_DEFINE(). This allows them
    to be placed as needed via linker scripts. This is useful
    for demand paging as developers can choose which can be
    pinned in memory.

    Signed-off-by: Daniel Leung <daniel.leung@intel.com>

diff --git a/include/kernel.h b/include/kernel.h
index 7183feef10..57e885503d 100644
--- a/include/kernel.h
+++ b/include/kernel.h
@@ -5037,7 +5037,8 @@ void k_heap_free(struct k_heap *h, void *mem);
  * @param bytes Size of memory region, in bytes
  */
 #define K_HEAP_DEFINE(name, bytes)                             \
-       char __aligned(8) /* CHUNK_UNIT */                      \
+       char __noinit_named(kheap_buf_##name)                   \
+            __aligned(8) /* CHUNK_UNIT */                      \
             kheap_##name[MAX(bytes, Z_HEAP_MIN_SIZE)];         \
        STRUCT_SECTION_ITERABLE(k_heap, name) = {               \
                .heap = {                                       
```

mmahadevan108, 2021-08-30:
@dcpleung any suggestions?

dcpleung, 2021-08-30:
Could you try this PR https://github.com/zephyrproject-rtos/zephyr/pull/38147? It builds on my machine, but I don't have the board so I can't test whether that works correctly.

hakehuang, 2021-08-31:
@dcpleung it does not works on NXP paltform, you do not need a board to test your fix, you just need build with below command
```
cd tests/subsys/usb/device
west build -b mimxrt1060_evk
```
below is my failure log
```
In file included from /home/shared/disk/zephyr_project/zephyr_dev/zephyr/include/init.h:11,
                 from /home/shared/disk/zephyr_project/zephyr_dev/zephyr/include/device.h:29,
                 from /home/shared/disk/zephyr_project/zephyr_dev/zephyr/include/drivers/usb/usb_dc.h:20,
                 from /home/shared/disk/zephyr_project/zephyr_dev/zephyr/drivers/usb/device/usb_dc_mcux.c:12:
/home/shared/disk/zephyr_project/zephyr_dev/zephyr/include/kernel.h:5043:7: error: section of 'kheap_ep_buf_pool' conflicts with previous declaration
 5043 |       kheap_##name[MAX(bytes, Z_HEAP_MIN_SIZE)];  \
      |       ^~~~~~
/home/shared/disk/zephyr_project/zephyr_dev/zephyr/drivers/usb/device/usb_dc_mcux.c:71:11: note: in expansion of macro 'K_HEAP_DEFINE'
   71 | __nocache K_HEAP_DEFINE(ep_buf_pool, 1024 * EP_BUF_NUMOF_BLOCKS);
      |           ^~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /home/ubuntu/.local/bin/cmake --build /home/shared/disk/zephyr_project/zephyr_test/zephyr/tests/subsys/usb/device/build

```

commit 89ddd0bf (2021-08-31): usb: usb_dc_mcux: use K_HEAP_DEFINE_NOCACHE()

This changes "__nocache K_HEAP_DEFINE()" to use the new
K_HEAP_DEFINE_NOCACHE() macro. This fixes a build error
as K_HEAP_DEFINE() is specifying its own linker section
so that it is no longer possible to specify another
linker section.

Fixes #38108
closed 2021-08-31 by nashif