zephyrproject-rtos/zephyr #38112: [Coverity CID: 239612] Out-of-bounds read in /zephyr/include/generated/syscalls/log_ctrl.h (Generated Code)

zephyrbot, 2021-08-30:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/fe2014ae1f4a7861771f86d500ac516fb5f59776//zephyr/include/generated/syscalls/log_ctrl.h#L86

Category: Memory - illegal accesses
Function: `log_filter_set`
Component: Other
CID: [239612](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=239612)

Details:

```
80     __pinned_func
81     static inline uint32_t log_filter_set(struct log_backend const *const backend, uint32_t domain_id, int16_t source_id, uint32_t level)
82     {
83     #ifdef CONFIG_USERSPACE
84         if (z_syscall_trap()) {
85             /* coverity[OVERRUN] */
>>>     CID 239612:    (OVERRUN)
>>>     Overrunning array of 4 bytes at byte offset 7 by dereferencing pointer "(uintptr_t *)&level".
86             return (uint32_t) arch_syscall_invoke4(*(uintptr_t *)&backend, *(uintptr_t *)&domain_id, *(uintptr_t *)&source_id, *(uintptr_t *)&level, K_SYSCALL_LOG_FILTER_SET);
87         }
88     #endif
89         compiler_barrier();
90         return z_impl_log_filter_set(backend, domain_id, source_id, level);
91     }
80     __pinned_func
81     static inline uint32_t log_filter_set(struct log_backend const *const backend, uint32_t domain_id, int16_t source_id, uint32_t level)
82     {
83     #ifdef CONFIG_USERSPACE
84         if (z_syscall_trap()) {
85             /* coverity[OVERRUN] */
>>>     CID 239612:    (OVERRUN)
>>>     Overrunning array of 4 bytes at byte offset 7 by dereferencing pointer "(uintptr_t *)&level".
86             return (uint32_t) arch_syscall_invoke4(*(uintptr_t *)&backend, *(uintptr_t *)&domain_id, *(uintptr_t *)&source_id, *(uintptr_t *)&level, K_SYSCALL_LOG_FILTER_SET);
87         }
88     #endif
89         compiler_barrier();
90         return z_impl_log_filter_set(backend, domain_id, source_id, level);
91     }
```

For more information about the violation, check the [Coverity Reference](https://scan9.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERRUN). ([CWE-125](http://cwe.mitre.org/data/definitions/125.html))

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v29271/p12996

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

cfriedt, 2021-09-23:
@nashif - is this a false positive?

nordic-krch, 2021-09-23:
Same as #38139. False positive. Will close this one and keep second open as there is an open question why suppressing comment is not working.

closed 2021-09-23 by nordic-krch