zephyrproject-rtos/zephyr #38115: [Coverity CID: 239605] Logically dead code in drivers/flash/nrf_qspi_nor.c

zephyrbot, 2021-08-30:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/fe2014ae1f4a7861771f86d500ac516fb5f59776/drivers/flash/nrf_qspi_nor.c#L929

Category: Control flow issues
Function: `qspi_nor_read`
Component: Drivers
CID: [239605](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=239605)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/fe2014ae1f4a7861771f86d500ac516fb5f59776/drivers/flash/nrf_qspi_nor.c#L929

```
923             return -EINVAL;
924         }
925    
926         int rc = ANOMALY_122_INIT(dev);
927    
928         if (rc != 0) {
>>>     CID 239605:  Control flow issues  (DEADCODE)
>>>     Execution cannot reach this statement: "goto out;".
929             goto out;
930         }
931    
932         qspi_lock(dev);
933    
934         nrfx_err_t res = read_non_aligned(dev, addr, dest, size);
```

For more information about the violation, check the [Coverity Reference](https://scan9.coverity.com/doc/en/cov_checker_ref.html#static_checker_DEADCODE). ([CWE-561](http://cwe.mitre.org/data/definitions/561.html))

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v29271/p12996

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

cfriedt, 2021-09-23:
@nvlsianpu  - can you please have a look at this?

nvlsianpu, 2021-09-23:
this is false postive

nashif, 2021-09-23:
> this is false postive

@nvlsianpu  mark it as such in coverity and close here please.

nvlsianpu, 2021-09-23:
@nashif I can't log in - I used to log in using Github help. Doesn't it works anymore

nashif, 2021-09-24:
someone already did that.

closed 2021-09-24 by nashif