zephyrproject-rtos/zephyr #38128: [Coverity CID: 239574] Out-of-bounds access in subsys/storage/flash_map/flash_map.c

zephyrbot, 2021-08-30:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/fe2014ae1f4a7861771f86d500ac516fb5f59776/subsys/storage/flash_map/flash_map.c

Category: Memory - corruptions
Function: `flash_area_check_int_sha256`
Component: Other
CID: [239574](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=239574)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/fe2014ae1f4a7861771f86d500ac516fb5f59776/subsys/storage/flash_map/flash_map.c

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERRUN). ([CWE-119](http://cwe.mitre.org/data/definitions/119.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

cfriedt, 2021-09-23:
@nvlsianpu  - can you please have a look at this?

de-nordic, 2021-10-06:
This will require closer look as the message from the coverity is very enigmatic and it is not even pointing where the overrun access happens.

carlescufi, 2021-10-19:
We think we have found out why this is a false positive.
Coverity says:

[image]

But in fact what is happening is that the structure in TinyCrypt looks like this:

```
#define TC_SHA256_BLOCK_SIZE (64)
#define TC_SHA256_DIGEST_SIZE (32)
#define TC_SHA256_STATE_BLOCKS (TC_SHA256_DIGEST_SIZE/4)

struct tc_sha256_state_struct {
	unsigned int iv[TC_SHA256_STATE_BLOCKS];
	uint64_t bits_hashed;
	uint8_t leftover[TC_SHA256_BLOCK_SIZE];
	size_t leftover_offset;
};

typedef struct tc_sha256_state_struct *TCSha256State_t;
```

This structure is 108 bytes long in native_posix, because building with `-m32` seems to ignore the rule that the size of the structure should be a multiple of the size of its biggest member. In Cortex-M, however, the struct is 112 bytes long because the compiler adds 4 bytes of padding at the end of the struct so that the size of the struct is a multiple of 8.

Here is the actual data:

native_posix:
```
*** Booting Zephyr OS build v2.7.99-574-g79f8af047c6e  ***
Hello World! native_posix
Size: 108
int: 4
uint64_t: 8
size_t: 4
offs iv: 0
offs bits_hashed: 32
offs leftover: 40
offs leftover_offset: 104
```

Cortex-M:

```
*** Booting Zephyr OS build v2.7.99-574-g79f8af047c6e  ***
Hello World! nrf52840dk_nrf52840
Size: 112
int: 4
uint64_t: 8
size_t: 4
offs iv: 0
offs bits_hashed: 32
offs leftover: 40
offs leftover_offset: 104
```

x86-64:

```
$ ./main
Hello World! x86-64
Size: 112
int: 4
uint64_t: 8
size_t: 8
offs iv: 0
offs bits_hashed: 32
offs leftover: 40
offs leftover_offset: 104
```

closed 2021-10-19 by carlescufi