zephyrproject-rtos/zephyr #38134: [Coverity CID: 239563] Logically dead code in subsys/bluetooth/host/id.c

zephyrbot, 2021-08-30:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/fe2014ae1f4a7861771f86d500ac516fb5f59776/subsys/bluetooth/host/id.c

Category: Control flow issues
Function: `bt_le_oob_get_local`
Component: Bluetooth
CID: [239563](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=239563)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/fe2014ae1f4a7861771f86d500ac516fb5f59776/subsys/bluetooth/host/id.c#L1615

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.coverity.com/doc/en/cov_checker_ref.html#static_checker_DEADCODE). ([CWE-561](http://cwe.mitre.org/data/definitions/561.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

cfriedt, 2021-09-23:
@joerchan , @jhedberg , @Vudentz  - can you please have a look at this?

joerchan, 2021-09-23:
Closing as false positive: Code is not dead if CONFIG_BT_ID_MAX > 1.

joerchan, 2021-09-24:
Re-opening to submit fix, easy to add the ifdef here, and it would both prevent future coverity reports, as well as establishing a pattern to compile out code only needed when multiple identities is supported.

PR #38813 (2021-09-28): Bluetooth: host: compile out check for multiple identities
Compile out check that is only needed when multiple identities have
been enabled.

Fixes: #38134
commit 28423833 (2021-09-28): Bluetooth: host: compile out check for multiple identities

Add check that can be removed by the compiler since the rest is only
needed when multiple identities have been enabled.

Fixes: #38134
commit 8691e3e0 (2021-09-29): Bluetooth: host: compile out check for multiple identities

Add check that can be removed by the compiler since the rest is only
needed when multiple identities have been enabled.

Fixes: #38134
closed 2021-09-28 by cfriedt