zephyrproject-rtos/zephyr #38138: [Coverity CID: 239554] Out-of-bounds read in /zephyr/include/generated/syscalls/log_msg2.h (Generated Code)

zephyrbot, 2021-08-30:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/fe2014ae1f4a7861771f86d500ac516fb5f59776//zephyr/include/generated/syscalls/log_msg2.h

Category: Memory - illegal accesses
Function: `z_log_msg2_static_create`
Component: Other
CID: [239554](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=239554)

Details:

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERRUN). ([CWE-125](http://cwe.mitre.org/data/definitions/125.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

cfriedt, 2021-09-23:
@nordic-krch - can you please have a look at this?

nordic-krch, 2021-09-23:
Same as #38139. Closing this one.

closed 2021-09-23 by nordic-krch