zephyrproject-rtos/zephyr #38144: [backport v2.6-branch] backport of #37787 failed

zephyrbot, 2021-08-30:
This issue was created automatically because the backport of #37787 to `v2.6-branch` failed.

The backport to `v2.6-branch` failed:
```
The process '/usr/bin/git' failed with exit code 1
```
To backport manually, run these commands in your terminal:
```bash
# Fetch latest updates from GitHub
git fetch
# Create a new working tree
git worktree add .worktrees/backport-v2.6-branch v2.6-branch
# Navigate to the new working tree
cd .worktrees/backport-v2.6-branch
# Create a new branch
git switch --create backport-37787-to-v2.6-branch
# Cherry-pick the merged commits of this pull request and resolve the conflicts
git cherry-pick 482198bf79874ebea1730cd16f8e1b263e207d28
# Push it to GitHub
git push --set-upstream origin backport-37787-to-v2.6-branch
# Go back to the original working tree
cd ../..
# Delete the working tree
git worktree remove .worktrees/backport-v2.6-branch
```
Then, create a pull request where the `base` branch is `v2.6-branch` and the `compare`/`head` branch is `backport-37787-to-v2.6-branch`.

cfriedt, 2021-09-07:
@ABOSTM - are you able to look into this?

erwango, 2021-09-08:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/38163

closed 2021-09-08 by erwango