zephyrproject-rtos/zephyr #38153: Zephyr Native POSIX select() implementation too frequent wakeup

bob2oneil, 2021-08-30:
**Describe the bug**
The implementation of the POSIX select() statement operates a bit inefficiently, by waking up prematurely at intervals much more frequently than the timeout specified to the statement.  This results in some thread inefficiencies. What have you tried to diagnose or workaround this issue?

To ensure that select() was not being tripped up with any of the selectors, to isolate the return to just the timeout mechanism, I removed them from the statement similar to the following test condition:

`        int count = zsock_select(maxFD + 1, 0, 0, 0, (struct zsock_timeval *)&tvTimeout);`

**To Reproduce**
Steps to reproduce the behavior:
1. west build for Native-POSIX target
2. west run
3. make
4. See error

**Expected behavior**
I expect the select() statement update to be somewhat consistent with the timeout value provided as an argument.  Perhaps I misunderstand the specifics of how it is implement for the POSIX environment.   For example, when I set a timeout value of approximately 10 ms, I expect a wake up in that interval, i.e. I expect the thread to pend for a much longer duration.  Instead, the statement is not pending for anywhere near that interval, most likely it pends for hundreds of microsecond ranges.

**Impact**
Not a show stopper, just trying to understand the behavior and seeing if anything can be done to make the implementation more efficient.

**Logs and console output**
The following log shows the timing for condition IN to the select() and OUT of the select. Here the timeout values are in the 10 ms range.  The OUT log shows the elapsed time at exit in the hundred of microsecond range.  Epochs are listed numerically based on a 10 ms interval.  The underlying time base used is the POSIX monatomic clock at **clock_gettime()** as shown below.  Emulation is being performed on Native-POSIX without implementing any delays in Zephyr (e.g. the kernel configuration setting **CONFIG_NATIVE_POSIX_SLOWDOWN_TO_REAL_TIME=n** is present).

**int rc = clock_gettime(getTimeOfDay ? CLOCK_REALTIME : CLOCK_MONOTONIC_RAW, &ts);**

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used latest of this date as: 
 -  *** Booting Zephyr OS build zephyr-v2.6.0-2852-g4b5bab2bc1d2  ***

**Additional context**
Program logs as follows:
```

4:42.510,000] <dbg> app.exec: select() OUT[epoch 16]: now (21192/169501), timeout (0/834), tvNextEpoch (21192/170000), elapsed (0/335), count 0
[00:34:42.510,000] <dbg> app.exec: select() IN [epoch 16]: now (21192/169807), timeout (0/193), m_tvNextEpoch (21192/170000)
[00:34:42.530,000] <dbg> app.exec: select() OUT[epoch 16]: now (21192/170133), timeout (0/193), tvNextEpoch (21192/170000), elapsed (0/326), count 0
[00:34:42.530,000] <dbg> app.exec: Pre-select: past next epoch: TDMA epoch 17 at now (21192/170431), timeout (0/193), next epoch (21192/170000)
[00:34:42.530,000] <dbg> app.setTvNextEpoch: Set m_tvNextEpoch (21192/180000)
[00:34:42.530,000] <dbg> app.onEpochTimeout: Epoch 17 at (21192/170884)
[00:34:42.530,000] <dbg> app.exec: select() IN [epoch 17]: now (21192/171033), timeout (0/8967), m_tvNextEpoch (21192/180000)
[00:34:42.550,000] <dbg> app.exec: select() OUT[epoch 17]: now (21192/171337), timeout (0/8967), tvNextEpoch (21192/180000), elapsed (0/304), count 0
[00:34:42.550,000] <dbg> app.exec: select() IN [epoch 17]: now (21192/171652), timeout (0/8348), m_tvNextEpoch (21192/180000)
[00:34:42.570,000] <dbg> app.exec: select() OUT[epoch 17]: now (21192/171963), timeout (0/8348), tvNextEpoch (21192/180000), elapsed (0/311), count 0
[00:34:42.570,000] <dbg> app.exec: select() IN [epoch 17]: now (21192/172255), timeout (0/7745), m_tvNextEpoch (21192/180000)
[00:34:42.590,000] <dbg> app.exec: select() OUT[epoch 17]: now (21192/172534), timeout (0/7745), tvNextEpoch (21192/180000), elapsed (0/279), count 0
[00:34:42.590,000] <dbg> app.exec: select() IN [epoch 17]: now (21192/172836), timeout (0/7164), m_tvNextEpoch (21192/180000)
[00:34:42.610,000] <dbg> app.exec: select() OUT[epoch 17]: now (21192/173175), timeout (0/7164), tvNextEpoch (21192/180000), elapsed (0/339), count 0
[00:34:42.610,000] <dbg> app.exec: select() IN [epoch 17]: now (21192/173512), timeout (0/6488), m_tvNextEpoch (21192/180000)
[00:34:42.630,000] <dbg> app.exec: select() OUT[epoch 17]: now (21192/173840), timeout (0/6488), tvNextEpoch (21192/180000), elapsed (0/328), count 0
[00:34:42.630,000] <dbg> app.exec: select() IN [epoch 17]: now (21192/174160), timeout (0/5840), m_tvNextEpoch (21192/180000)
[00:34:42.650,000] <dbg> app.exec: select() OUT[epoch 17]: now (21192/174463), timeout (0/5840), tvNextEpoch (21192/180000), elapsed (0/303), count 0
[00:34:42.650,000] <dbg> app.exec: select() IN [epoch 17]: now (21192/174787), timeout (0/5213), m_tvNextEpoch (21192/180000)
[00:34:42.670,000] <dbg> app.exec: select() OUT[epoch 17]: now (21192/175122), timeout (0/5213), tvNextEpoch (21192/180000), elapsed (0/335), count 0
[00:34:42.670,000] <dbg> app.exec: select() IN [epoch 17]: now (21192/175450), timeout (0/4550), m_tvNextEpoch (21192/180000)
[00:34:42.690,000] <dbg> app.exec: select() OUT[epoch 17]: now (21192/175796), timeout (0/4550), tvNextEpoch (21192/180000), elapsed (0/346), count 0
[00:34:42.690,000] <dbg> app.exec: select() IN [epoch 17]: now (21192/176025), timeout (0/3975), m_tvNextEpoch (21192/180000)
[00:34:42.710,000] <dbg> app.exec: select() OUT[epoch 17]: now (21192/176277), timeout (0/3975), tvNextEpoch (21192/180000), elapsed (0/252), count 0
[00:34:42.710,000] <dbg> app.exec: select() IN [epoch 17]: now (21192/176494), timeout (0/3506), m_tvNextEpoch (21192/180000)
[00:34:42.730,000] <dbg> app.exec: select() OUT[epoch 17]: now (21192/176695), timeout (0/3506), tvNextEpoch (21192/180000), elapsed (0/201), count 0
[00:34:42.730,000] <dbg> app.exec: select() IN [epoch 17]: now (21192/176911), timeout (0/3089), m_tvNextEpoch (21192/180000)
[00:34:42.750,000] <dbg> app.exec: select() OUT[epoch 17]: now (21192/177160), timeout (0/3089), tvNextEpoch (21192/180000), elapsed (0/249), count 0
[00:34:42.750,000] <dbg> app.exec: select() IN [epoch 17]: now (21192/177371), timeout (0/2629), m_tvNextEpoch (21192/180000)
[00:34:42.770,000] <dbg> app.exec: select() OUT[epoch 17]: now (21192/177616), timeout (0/2629), tvNextEpoch (21192/180000), elapsed (0/245), count 0
[00:34:42.770,000] <dbg> app.exec: select() IN [epoch 17]: now (21192/177820), timeout (0/2180), m_tvNextEpoch (21192/180000)
[00:34:42.790,000] <dbg> app.exec: select() OUT[epoch 17]: now (21192/178075), timeout (0/2180), tvNextEpoch (21192/180000), elapsed (0/255), count 0
[00:34:42.790,000] <dbg> app.exec: select() IN [epoch 17]: now (21192/178288), timeout (0/1712), m_tvNextEpoch (21192/180000)
[00:34:42.810,000] <dbg> app.exec: select() OUT[epoch 17]: now (21192/178535), timeout (0/1712), tvNextEpoch (21192/180000), elapsed (0/247), count 0
[00:34:42.810,000] <dbg> app.exec: select() IN [epoch 17]: now (21192/178752), timeout (0/1248), m_tvNextEpoch (21192/180000)
[... 4 lines cut ...]
[00:34:42.870,000] <dbg> app.exec: select() OUT[epoch 17]: now (21192/179822), timeout (0/369), tvNextEpoch (21192/180000), elapsed (0/191), count 0
[00:34:42.870,000] <dbg> app.exec: Pre-select: past next epoch: TDMA epoch 18 at now (21192/180029), timeout (0/369), next epoch (21192/180000)
[00:34:42.870,000] <dbg> app.setTvNextEpoch: Set m_tvNextEpoch (21192/190000)
[00:34:42.870,000] <dbg> app.onEpochTimeout: Epoch 18 at (21192/180338)
[00:34:42.870,000] <dbg> app.exec: select() IN [epoch 18]: now (21192/180438), timeout (0/9562), m_tvNextEpoch (21192/190000)
[00:34:42.890,000] <dbg> app.exec: select() OUT[epoch 18]: now (21192/180677), timeout (0/9562), tvNextEpoch (21192/190000), elapsed (0/239), count 0
[00:34:42.890,000] <dbg> app.exec: select() IN [epoch 18]: now (21192/180885), timeout (0/9115), m_tvNextEpoch (21192/190000)
[00:34:42.910,000] <dbg> app.exec: select() OUT[epoch 18]: now (21192/181127), timeout (0/9115), tvNextEpoch (21192/190000), elapsed (0/242), count 0
[00:34:42.910,000] <dbg> app.exec: select() IN [epoch 18]: now (21192/181338), timeout (0/8662), m_tvNextEpoch (21192/190000)
[00:34:42.930,000] <dbg> app.exec: select() OUT[epoch 18]: now (21192/181584), timeout (0/8662), tvNextEpoch (21192/190000), elapsed (0/246), count 0
[00:34:42.930,000] <dbg> app.exec: select() IN [epoch 18]: now (21192/181799), timeout (0/8201), m_tvNextEpoch (21192/190000)
[... cut ...]
[00:34:43.030,000] <dbg> app.exec: select() OUT[epoch 18]: now (21192/184098), timeout (0/6325), tvNextEpoch (21192/190000), elapsed (0/423), count 0
[00:34:43.030,000] <dbg> app.exec: select() IN [epoch 18]: now (21192/184438), timeout (0/5562), m_tvNextEpoch (21192/190000)
[00:34:43.050,000] <dbg> app.exec: select() OUT[epoch 18]: now (21192/184696), timeout (0/5562), tvNextEpoch (21192/190000), elapsed (0/258), count 0
[00:34:43.050,000] <dbg> app.exec: select() IN [epoch 18]: now (21192/184915), timeout (0/5085), m_tvNextEpoch (21192/190000)

```

aescolar, 2021-09-01:
Hi @bob2oneil . I'm not completely sure I understand what you describe. But it would seem the issue is about mixing simulated time and real time.
I'm assuming the following from your description:
* When you say `select` you always refer to `zsock_select()` (Zephyr's API, and never to the host kernel/Linux); 
* You have a small loop, calling  `clock_gettime()` (a call to the Linux kernel), using the Zephyr logger to print that time stamp out,  , calling `zsock_select()`, and `clock_gettime()` and printing with the logger again.

What you can see is that while the Zephyr kernel advances 1 tick (10ms) of simulated time (the only time the Zephyr kernel knows about), much less real time passes.
That is just due to running native_posix in non-real time. Please see:
https://docs.zephyrproject.org/latest/boards/posix/native_posix/doc/index.html#about-time-in-native-posix

The logger timestamp (between `[]`) is the Zephyr time. And we can see that advances in jumps of 10ms between the logger calls (I presume you have the default 10ms system tick resolution).

If this answers your question, feel free to close the issue

bob2oneil, 2021-09-01:
Thanks for the quick response.  You have characterized this issue correctly in terms of the implementation.  Note that I am running this application with **CONFIG_NATIVE_POSIX_SLOWDOWN_TO_REAL_TIME=n**, so presumably Native-POSIX is running as quickly as possible.  The is broadly is that entry into **zsock_select** specifies a timeout value of close to 10 ms (my epoch interval), but zsock_select(), which should pend for that duration since none of the file handles would trigger an exit, returns very quickly pending for one a few hundred microseconds.

Consider the following log entry, here the timeout is set to 9562 microseconds, but zsock_select() appears to return while waiting only 239 microseconds.

I do not fully understand the correlation between the logging timestamp and the time values returned from the POSIX clock_gettime() API when running without slowdown

[00:34:42.870,000] <dbg> app.exec: select() **IN** [epoch 18]: now (21192/180438), timeout (0/**9562**), m_tvNextEpoch (21192/190000)
[00:34:42.890,000] <dbg> app.exec: select() **OUT**[epoch 18]: now (21192/180677), timeout (0/9562), tvNextEpoch (21192/190000), elapsed (0/**239**),

aescolar, 2021-09-01:
>I do not fully understand the correlation between the logging timestamp and the time values returned from the POSIX clock_gettime() API when running without slowdown

They are not correlated :). The logging timestamp is in simulated time. The call to the **host** `clock_gettime()` is real/host/wall time. It takes very little real time to advance those 10ms of simulated time.

bob2oneil, 2021-09-01:
OK, note that the elapsed time measured is the wall clock time at entry and exit to **zsock_select().**

bob2oneil, 2021-09-01:
Aside for the timestamp correlation (or lack thereof), the primary point was that zsock_select() does not pend for the duration specified (as a timeval) as measured using wall clock time.  Maybe there is an explanation for that behavior?

aescolar, 2021-09-01:
> Aside for the timestamp correlation (or lack thereof), the primary point was that zsock_select() does not pend for the duration specified (as a timeval) as measured using wall clock time. Maybe there is an explanation for that behavior?

That ` zsock_select()` will wait for the timeval duration in **simulated** time. Nothing in Zephyr cares about wall/host/real time when `CONFIG_NATIVE_POSIX_SLOWDOWN_TO_REAL_TIME=n`

bob2oneil, 2021-09-01:
Would it be at all easy to modify the Zephyr source code so that the logging uses wall clock time when CONFIG_NATIVE_POSIX_SLOWDOWN_TO_REAL_TIME is set to 'n'?

aescolar, 2021-09-02:
> Would it be at all easy to modify the Zephyr source code so that the logging uses wall clock time when CONFIG_NATIVE_POSIX_SLOWDOWN_TO_REAL_TIME is set to 'n'?

That would not be a good idea. The basic principle is that when you run decoupled from the host time, nothing in Zephyr or your app is aware of the host time.
The only reason why you are aware in your test of the host time is because you are calling out into the underlaying Linux kernel.

aescolar, 2021-09-09:
Closing as this question would seem to be answered now. If that is not the case, please reopen the issue.

closed 2021-09-09 by aescolar