zephyrproject-rtos/zephyr #38162: Upgrade to 2.6 GPIO device_get_binding("GPIO_0") now returns null

dlktdr, 2021-08-31:
**Describe the bug**
I have a project that's been working great on 2.5. I went and upgraded to version 2.6 to try it out. There were a few Bluetooth kconfig options that were depreciated I removed and it compiles okay. The issue I'm now seeing is that when 
`device_get_binding("GPIO_0");` is called on initialization it always return NULL. On the old version I get an address of `0x20000aa8 <__device_dts_ord_10>` as expected.

I have used the default device tree file shipped in the arduino_nano_33_ble.dts file. I do have an dt overlay but even removed, 
 and default KConfig options seems to return null too.

**Expected behavior**
Work the same as V2.5.

I'm not sure if there was a change to how the device tree is accessed. I didn't see anything obvious in the release notes.
Thank you for any help.

**Environment (please complete the following information):**
PlatformIO, Windows
`(platform: nordicnrf52@8.2.0; board: arduino_nano_33_ble; framework: zephyr)`

Full Code Here. Main issue in io.cpp, arduino_gpio_init()
https://github.com/dlktdr/HeadTracker/tree/zephyr8.2.0/firmware/src

gmarull, 2021-08-31:
You should probably consider using `DEVICE_DT_GET(DT_NODELABEL(gpio0))` (combined with `device_is_ready`). I'd also suggest watching this great presentation from @mbolivar-nordic https://www.youtube.com/watch?v=sWaxQyIgEBY

dlktdr, 2021-08-31:
Thank you that works! Also thank you for the video that was helpful.

closed 2021-08-31 by dlktdr