zephyrproject-rtos/zephyr #38170: Shell argument in second position containing a question mark is ignored

MarekPikula, 2021-08-31:
**Describe the bug**
I have a shell command declared like this:
```c
SHELL_CMD_REGISTER(c, NULL, "Shell console", shell_console);
```

It is supposed to serve as a UART passthrough for another device from Zephyr shell. I was testing it with LTE module, which accepts AT commands on UART and wanted to execute `c at+cfun?`, but it seems that if there is a question mark in the first argument of shell command (in any position, so doesn't matter if it's in the beginning, middle, or end) it's simply ignored (`argc=1`, `argv={"c", ""}`). If I execute it with three arguments `argc` is still equal to 1, but the second argument is present in memory:
```c
shell_info(shell, "%i: %s %s %s", argc, argv[0], argv[1], argv[2]);
```
for `c test? test2?` it prints `1: c  test2?`.

At the moment I did a workaround, which ignores the second argument if `argv[1]` is `c` and parses next arguments as before (ie. sending them over UART) and it's not that big of a deal, but either way it seems pretty strange overall.

**To Reproduce**

1. ```c
   #include <shell/shell.h>
   static int shell_console(const struct shell* shell, size_t argc, char** argv) {
       shell_info(shell, "%i: %s %s %s", argc, argv[0], argv[1], argv[2]);
       return 0;
   }
   SHELL_CMD_REGISTER(c, NULL, "Shell console", shell_console);
   ```
2. Execute `c test?`
3. Execute `c test? test2?`

**Expected behavior**
Question mark shouldn't remove the argument with index 1 from argument list and discard all following arguments.

**Impact**
Currently I have a workaround, which isn't a show-stopper, but it doesn't look nice overall and doesn't help with documentation clarity.

**Logs and console output**
```
test:~$ c
1: c  
test:~$ c ? test
1: c  test
test:~$ c asdf? test2?
1: c  test2?
test:~$ 
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: gcc-arm-none-eabi-10-2020-q4-major
 - Commit SHA or Version used: a62ea8fa297a12c2d17332218ff45c1e54c55a8e from Nordic v2.6.0-rc1-ncs1, tested also on 79a6c07536bc14583198f8e3555df6134d8822cf v2.6.0
 - Device: nRF52832 on nRF52 DK

**Additional context**
None

jakub-uC, 2021-09-01:
@MarekPikula 
In my opinion, it is not a bug, it is a feature ;-)
Try to compile your project with the following configuration option:
`CONFIG_SHELL_WILDCARD=n`

jakub-uC, 2021-09-07:
@MarekPikula Please let me know if we can close this?

jakub-uC, 2021-09-16:
I am closing because @MarekPikula is not giving any feedback. If this is still an issue we can reopen it in the future.

closed 2021-09-16 by jakub-uC