zephyrproject-rtos/zephyr #38172: modem_context_sprint_ip_addr return pointer to stack array

ssharks, 2021-08-31:
In the function modem_context_sprint_ip_addr a local array is constructed. The pointer to that structure is returned. Since this array is constructed on the stack, outside this function, this array does not exist, resulting in unexpected behavior.

File location:
drivers/modem/modem_context.c
Lines 23 - 40.

I see two options to solve this problem:
1. Either pass the array as an argument to the function, so it declared outside this function
2. Return a structure containing the array

cfriedt, 2021-08-31:
@ssharks - are you able to submit a PR for either of your proposed fixes?

ssharks, 2021-09-03:
Somewhere next week I'll give it a try.

ssharks, 2021-09-10:
Currently I'm trying to create a pull request, the code looks to be ready, but I have no simple way to compile the modified drivers to test if the fix doesn't break compilation. Any suggestions how to tackle this?

rerickson1, 2021-09-10:
> Currently I'm trying to create a pull request, the code looks to be ready, but I have no simple way to compile the modified drivers to test if the fix doesn't break compilation. Any suggestions how to tackle this?

You can run the `drivers/build_all` test on Linux or MacOS using the following command:
`./scripts/twister -v  -T tests/drivers/build_all`

PR #38459 (2021-10-01): Fix reference to a released stack variable in modem_context_sprint_ip…
Suggested fix for issue #38172. How can I test it?

Fixes #38172.
closed 2021-10-01 by cfriedt