zephyrproject-rtos/zephyr #38177: LORA Module crashes I2C

alfonsotames, 2021-08-31:
**Describe the bug**
Adding the LORA module (CONFIG_LORA=y) in prj.conf crashes I2C.

**To Reproduce**
Add the Lora module to any project that uses I2C (tested on a nucleo_wl55jc) using sensor sht3xd:

Add CONFIG_LORA=y to file  zephyr/samples/sensor/sht3xd/prj.conf
 
I2C will not read from the sensor anymore.

**Expected behavior**
LORA should work alongside any module, including I2C.

**Impact**
Projects cannot use LORA + I2C (maybe other modules).

**Logs and console output**

Without LORA module:

```
*** Booting Zephyr OS build zephyr-v2.6.0-1397-g7d07a941faf4  ***
SHT3XD: 23.54 Cel ; 53.69 %RH
SHT3XD: 23.58 Cel ; 53.71 %RH
SHT3XD: 23.57 Cel ; 53.74 %RH
SHT3XD: 23.54 Cel ; 53.71 %RH
SHT3XD: 23.57 Cel ; 53.69 %RH
```

With LORA module:
```

*** Booting Zephyr OS build zephyr-v2.6.0-1397-g7d07a941faf4  ***
SHT3XD:  Cel ;  %RH
SHT3XD:  Cel ;  %RH
SHT3XD:  Cel ;  %RH
SHT3XD:  Cel ;  %RH
SHT3XD:  Cel ;  %RH
SHT3XD:  Cel ;  %RH
```

**Environment (please complete the following information):**
- zephyr-v2.6.0-1397-g7d07a941faf4

cfriedt, 2021-08-31:
@erwango - may be fairly specific to stm

alfonsotames, 2021-08-31:
I tried the same with the BME280 sensor and it doesn't crash. Seems to work fine. Something is going on with the SHT3DX sensor.

Mani-Sadhasivam, 2021-09-02:
@alfonsotames I think this issue is specific to STM32WL platform or the sensor as lora driver should not interfere with other peripherals. @erwango @fabiobaltieri any ideas?

Mani-Sadhasivam, 2021-09-02:
@alfonsotames To help debugging, can you please share the SHT3DX overlay?

fabiobaltieri, 2021-09-02:
Weird, I have a board with an STM32WLE5 (the Seeed Lora-E5 modules), couple SPI devices and an I2C EEPROM... I2C is working fine. Maybe a pinmux misconfiguration that breaks the i2c when LoRa is enabled?

erwango, 2021-09-22:
@alfonsotames Did you make any progress in identifying the root cause of the issue you're seeing ?

alfonsotames, 2021-09-23:
For my project, I had to  stop using CONFIG_NEWLIB_LIBC=y

That did the trick but I don't know how it is related and why it conflicts
with the LoRa module.

erwango, 2021-10-05:
@alfonsotames I'm not sure we have enough info to debug this point, so I propose to close for now, since you've found a workaround.
Don't hesitate to re-open if you find new elements.

closed 2021-10-05 by erwango