zephyrproject-rtos/zephyr #38186: i.MX RT10xx boards fail to initialize when Ethernet is enabled

MaureenHelm, 2021-09-01:
**Describe the bug**
The kernel fails to initialize on i.MX RT10xx boards when Ethernet is enabled because the board-level PHY reset routine `mimxrt1064_evk_phy_reset()` never completes. 

The PHY reset occurs in `PRE_KERNEL_2` and invokes `k_busy_wait()`.

The system clock driver is initialized in `PRE_KERNEL_2`.

Commit f0b3146ff5538e5381318a309fec292d3fbb1f3f affected the relative order of these two operations, causing the PHY reset to occur before the system clock driver is initialized.

Found while experimenting with #37863

**To Reproduce**
```
$ west build -p auto -b mimxrt1064_evk samples/net/dhcpv4_client
$ west flash
```

No output on the console.

**Expected behavior**
Boot banner and assigned IP address to print on the console.

**Impact**
Affects all i.MX RT boards that currently support Ethernet: `mimxrt1020_evk`, `mimxrt1050_evk`, `mimxrt1060_evk`, `mimxrt1064_evk`

**Logs and console output**
There is no console output.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain zephyr-sdk-0.13.0
 - Commit fd864c61bbd616eeb4ff9e5a3400520228dd48e6

**Additional context**
The following is a cheap fix, but there may be a more robust way to do it.
```diff
$ git diff
diff --git a/boards/arm/mimxrt1064_evk/pinmux.c b/boards/arm/mimxrt1064_evk/pinmux.c
index 484a9a9ca6..ce31f5c2d9 100644
--- a/boards/arm/mimxrt1064_evk/pinmux.c
+++ b/boards/arm/mimxrt1064_evk/pinmux.c
@@ -351,5 +351,5 @@ static int mimxrt1064_evk_phy_reset(const struct device *dev)
 
 SYS_INIT(mimxrt1064_evk_init, PRE_KERNEL_1, 0);
 #if DT_NODE_HAS_STATUS(DT_NODELABEL(enet), okay) && CONFIG_NET_L2_ETHERNET
-SYS_INIT(mimxrt1064_evk_phy_reset, PRE_KERNEL_2, 0);
+SYS_INIT(mimxrt1064_evk_phy_reset, PRE_KERNEL_2, 1);
 #endif
```

cc: @dleach02 @mmahadevan108 @hakehuang @dcpleung @aunsbjerg

hakehuang, 2021-09-01:
in this weekly testing all RT serieses board fails. and this fix work for mimxrt1060_evk.

aunsbjerg, 2021-09-01:
Do we want to wait for a robust fix or go with the quick fix for now? I can update #37863 with the quick fix if we want to get that merged sooner rather than later.

dcpleung, 2021-09-01:
The posted fix is probably the best we can do right now. The ordering of two `SYS_INIT()` and device init having the same init level and priority depends on the how files are linked. So a strict ordering needs to be done manually, until we have device dependency baked in the init code.

dleach02, 2021-09-02:
@MaureenHelm, since there is a dependency, maybe we can be a bit more explicit by doing something like the following:

```
/* 
 * PHY reset has a dependency on SYSTEM Clock being initialized so ensure that the 
 * reset function is called after the system clock is initialized.
 */
SYS_INIT(mimxrt1064_evk_phy_reset, PRE_KERNEL_2, CONFIG_SYSTEM_CLOCK_INIT_PRIORITY+1);
```

hakehuang, 2021-09-08:
this issue is still in 2.7.0-RC1

PR #38473 (2021-09-13): boards: arm: Reset Ethernet PHY after system timer on mimxrt1xxx_evk's
Moves the Ethernet PHY reset routine on mimxrt1xxx_evk boards to execute
in the POST_KERNEL initialization level rather than the PRE_KERNEL_2
level because it relies on a kernel service, k_busy_wait(). Leverages
the existing CONFIG_PHY_INIT_PRIORITY to allow users to configure when
within the POST_KERNEL level this routine occurs, which by default is
set correctly to the following order:

```
Init Function			Init Level	Priority
--------------------------------------------------------------------------------------
sys_clock_ [... cut]
closed 2021-09-13 by galak