zephyrproject-rtos/zephyr #38196: net nbr command might crash

eyuval, 2021-09-01:
I managed to crash the net nbr command by doing the following:
1. create a vlan interface, configure an ip and bring it up
2. run ping from the host to the interface
3. issue the net nbr command

The problem in my case was the following code in the nbr_cb function (file: subsys/net/ip/net_shell.c):

	   nbr->idx == NET_NBR_LLADDR_UNKNOWN ? "?" :
	   net_sprint_ll_addr(
		   net_nbr_get_lladdr(nbr->idx)->addr,
		   net_nbr_get_lladdr(nbr->idx)->len),
	   net_nbr_get_lladdr(nbr->idx)->len == 8U ? "" : padding,
	   net_sprint_ipv6_addr(&net_ipv6_nbr_data(nbr)->addr));

The problem seems to to be, that in the last call to net_nbr_get_lladdr, the nbr->idx == NET_NBR_LLADDR_UNKNOWN  check is not performed, resulting on occasion (i.e. when the idx is 0xff) in the following assert in function net_nbr_get_lladdr:
ASSERTION FAIL [idx < 254] @ ZEPHYR_BASE/subsys/net/ip/nbr.c:182
        idx 255 >= max 254

Adding a check for nbr->idx == NET_NBR_LLADDR_UNKNOWN fixes the issue for me

carlescufi, 2021-09-01:
@eyuval would you mind sending a Pull Request with the fix?

cfriedt, 2021-09-07:
@rlubos - would you mind crafting a PR out of the proposed fix?

rlubos, 2021-09-08:
@cfriedt Yes, I have the issue on my list to look into.

PR #38389 (2021-09-10): net: shell: Fix assertion in net nbr command
Listing a neighbour table with "net nbr" command, when a  neighbour w/o
assigned link address was present, resulted in an assert condition. Add
additional check to prevent this.

Fixes #38196
closed 2021-09-10 by cfriedt