zephyrproject-rtos/zephyr #38212: pcie: emulated endpoint device

cfriedt, 2021-09-01:
**Is your enhancement proposal related to a problem? Please describe.**
Similarly to how Zephyr supports emulated i2c, spi, gpio, and other peripheral devices, it would be advantageous to support an emulated pcie endpoint. 

This would provide a basis for API-level automated testing for application code or middleware that interfaces with the PCIe endpoint, but also facilitate other types of end-to-end testing with virtual machines, virtio, and so on.

**Describe the solution you'd like**
Any options for the emulated device should be made configurable via device tree bindings. It should also be possible to support more than one instance as well.

This will likely require a fairly robust test suite that exercises mostly complete coverage of options. Will likely need to have test plan reviews as well as code reviews from a few experts in the Zephyr community.

The API in `include/drivers/pcie/endpoint/pcie_ep.h` seems to be generic enough to use as a starting point.

**Describe alternatives you've considered**

**Additional context**

cfriedt, 2021-09-01:
Tagging @tbursztyka and @superna9999 for possible interest / pointers.

superna9999, 2021-09-02:
Do you mean a virtual pcie endpoint with a virtual pcie bus containing this endpoint ?

cfriedt, 2021-09-02:
> Do you mean a virtual pcie endpoint with a virtual pcie bus containing this endpoint ?

Yes

superna9999, 2021-09-02:
It must be doable with the generic pcie infrastructure I'm pushing at https://github.com/zephyrproject-rtos/zephyr/pull/37668

For MSI/MSI-X virtual interrupts handling, I'll also push some very generic changes that could also match this usecase.

cfriedt, 2024-02-13:
This feature would still be quite useful for end-to-end host / device communication over PCIe.

CC @PiotrZierhoffer, @kgugala, @mgielda