zephyrproject-rtos/zephyr #38214: xtensa builds fail in CI due to running out of ram to link

galak, 2021-09-01:
The following commit causes the amount of memory the linker needs to grow drastically such that linking on xtensa builds fails:

9505ee89a3401af93ffddc68e7b04fc948e2f89a

See logs here:

https://buildkite.com/zephyr/zephyr-daily/builds/443#73c88eb1-df25-4260-b43a-c3e84d94f87a

galak, 2021-09-01:
@dcpleung don't know if you have any ideas on how we might adjust IDT_BASE

dcpleung, 2021-09-01:
IIRC, on all the modified SoCs, there is nothing after the end of RAM. Maybe we can simply revert that commit?

JordanYates, 2021-09-02:
> IIRC, on all the modified SoCs, there is nothing after the end of RAM. Maybe we can simply revert that commit?

That's not a robust solution. The previous behavior is equivalent to placing `IDT_LIST` after the `sram` node.
You then can't do this for example:
```
soc {
    sram@0 {
        reg = <0 0x80000>;
    }
    my-sram@80000 {
        reg = <0x80000 0x100>;
    }
}
```
That said, I'm not sure how else to work around this compiler bug.

cfriedt, 2021-09-07:
@JordanYates, @galak - if there is no clear solution right now, we will have to revert the commit.

galak, 2021-09-08:
> @JordanYates, @galak - if there is no clear solution right now, we will have to revert the commit.

I think we should revert it for now.

andyross, 2021-09-09:
I'm just noticing this now.  FWIW: the correct solution here (to having a section that links far from another) is to emit it to a separate ELF phdr, which avoids the need to pad.  But that requires changing a bunch of linker scripts.

PR #38399 (2021-09-09): Revert "linker: xtensa: move IDT_LIST region"
This reverts commit 9505ee89a3401af93ffddc68e7b04fc948e2f89a.

Fixes #38214
commit 6e26512b (2021-09-09): Revert "linker: xtensa: move IDT_LIST region"

This reverts commit 9505ee89a3401af93ffddc68e7b04fc948e2f89a.

Fixes #38214
closed 2021-09-09 by galak