zephyrproject-rtos/zephyr #38216: nxp_adsp_imx8 fails to build a number of tests

galak, 2021-09-01:
`./scripts/twister -p nxp_adsp_imx8`

Gets:

```
ERROR   - nxp_adsp_imx8             samples/kernel/condition_variables/condvar/sample.kernel.cond_var FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nxp_adsp_imx8/samples/kernel/condition_variables/condvar/sample.kernel.cond_var/build.log
INFO    - Total complete:   11/  88  12%  skipped: 1316, failed:    1
ERROR   - nxp_adsp_imx8             samples/kernel/condition_variables/simple/sample.kernel.cond_var.simple FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nxp_adsp_imx8/samples/kernel/condition_variables/simple/sample.kernel.cond_var.simple/build.log
INFO    - Total complete:   12/  88  13%  skipped: 1316, failed:    2
ERROR   - nxp_adsp_imx8             tests/kernel/fatal/message_capture/kernel.logging.message_capture FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nxp_adsp_imx8/tests/kernel/fatal/message_capture/kernel.logging.message_capture/build.log
INFO    - Total complete:   13/  88  14%  skipped: 1316, failed:    3
ERROR   - nxp_adsp_imx8             tests/kernel/sleep/kernel.common.timing            FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nxp_adsp_imx8/tests/kernel/sleep/kernel.common.timing/build.log
INFO    - Total complete:   14/  88  15%  skipped: 1316, failed:    4
ERROR   - nxp_adsp_imx8             tests/kernel/condvar/condvar_api/kernel.condvar    FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nxp_adsp_imx8/tests/kernel/condvar/condvar_api/kernel.condvar/build.log
INFO    - Total complete:   15/  88  17%  skipped: 1316, failed:    5
ERROR   - nxp_adsp_imx8             tests/kernel/timer/starve/kernel.timer.starve      FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nxp_adsp_imx8/tests/kernel/timer/starve/kernel.timer.starve/build.log
INFO    - Total complete:   16/  88  18%  skipped: 1316, failed:    6
ERROR   - nxp_adsp_imx8             tests/kernel/fatal/exception/kernel.common.stack_sentinel FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nxp_adsp_imx8/tests/kernel/fatal/exception/kernel.common.stack_sentinel/build.log
INFO    - Total complete:   17/  88  19%  skipped: 1316, failed:    7
ERROR   - nxp_adsp_imx8             tests/kernel/tickless/tickless_concept/kernel.tickless.concept FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nxp_adsp_imx8/tests/kernel/tickless/tickless_concept/kernel.tickless.concept/build.log
INFO    - Total complete:   18/  88  20%  skipped: 1316, failed:    8
ERROR   - nxp_adsp_imx8             tests/kernel/sched/metairq/kernel.scheduler.metairq FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nxp_adsp_imx8/tests/kernel/sched/metairq/kernel.scheduler.metairq/build.log
INFO    - Total complete:   19/  88  21%  skipped: 1316, failed:    9
ERROR   - nxp_adsp_imx8             tests/kernel/semaphore/semaphore/kernel.semaphore  FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nxp_adsp_imx8/tests/kernel/semaphore/semaphore/kernel.semaphore/build.log
INFO    - Total complete:   20/  88  22%  skipped: 1316, failed:   10
ERROR   - nxp_adsp_imx8             tests/kernel/workq/critical/kernel.workqueue       FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nxp_adsp_imx8/tests/kernel/workq/critical/kernel.workqueue/build.log
INFO    - Total complete:   21/  88  23%  skipped: 1316, failed:   11
ERROR   - nxp_adsp_imx8             tests/kernel/timer/timer_monotonic/kernel.timer.monotonic FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nxp_adsp_imx8/tests/kernel/timer/timer_monotonic/kernel.timer.monotonic/build.log
INFO    - Total complete:   22/  88  25%  skipped: 1316, failed:   12
ERROR   - nxp_adsp_imx8             tests/kernel/timer/timer_api/kernel.timer          FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nxp_adsp_imx8/tests/kernel/timer/timer_api/kernel.timer/build.log
INFO    - Total complete:   23/  88  26%  skipped: 1316, failed:   13
ERROR   - nxp_adsp_imx8             tests/kernel/workq/work/kernel.work.api            FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nxp_adsp_imx8/tests/kernel/workq/work/kernel.work.api/build.log
INFO    - Total complete:   24/  88  27%  skipped: 1316, failed:   14
ERROR   - nxp_adsp_imx8             tests/kernel/workq/work_queue/kernel.workqueue     FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nxp_adsp_imx8/tests/kernel/workq/work_queue/kernel.workqueue/build.log
INFO    - Total complete:   25/  88  28%  skipped: 1316, failed:   15
ERROR   - nxp_adsp_imx8             tests/kernel/timer/timer_error_case/kernel.timer   FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nxp_adsp_imx8/tests/kernel/timer/timer_error_case/kernel.timer/build.log
INFO    - Total complete:   26/  88  29%  skipped: 1316, failed:   16
ERROR   - nxp_adsp_imx8             tests/kernel/stack/stack/kernel.stack.usage        FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nxp_adsp_imx8/tests/kernel/stack/stack/kernel.stack.usage/build.log
INFO    - Total complete:   27/  88  30%  skipped: 1316, failed:   17
ERROR   - nxp_adsp_imx8             tests/kernel/sched/schedule_api/kernel.scheduler.multiq_no_timeslicing FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nxp_adsp_imx8/tests/kernel/sched/schedule_api/kernel.scheduler.multiq_no_timeslicing/build.log
INFO    - Total complete:   28/  88  31%  skipped: 1316, failed:   18
ERROR   - nxp_adsp_imx8             tests/kernel/sched/schedule_api/kernel.scheduler.dumb_no_timeslicing FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nxp_adsp_imx8/tests/kernel/sched/schedule_api/kernel.scheduler.dumb_no_timeslicing/build.log
INFO    - Total complete:   29/  88  32%  skipped: 1316, failed:   19
ERROR   - nxp_adsp_imx8             tests/kernel/timer/timer_api/kernel.timer.tickless FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nxp_adsp_imx8/tests/kernel/timer/timer_api/kernel.timer.tickless/build.log
INFO    - Total complete:   30/  88  34%  skipped: 1316, failed:   20
ERROR   - nxp_adsp_imx8             tests/kernel/sched/schedule_api/kernel.scheduler.multiq FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nxp_adsp_imx8/tests/kernel/sched/schedule_api/kernel.scheduler.multiq/build.log
INFO    - Total complete:   33/  88  37%  skipped: 1318, failed:   21
ERROR   - nxp_adsp_imx8             tests/kernel/queue/kernel.queue                    FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nxp_adsp_imx8/tests/kernel/queue/kernel.queue/build.log
[... 87 lines cut ...]
INFO    - Total complete:   80/  88  90%  skipped: 1334, failed:   51
ERROR   - nxp_adsp_imx8             tests/kernel/context/kernel.common                 FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nxp_adsp_imx8/tests/kernel/context/kernel.common/build.log
INFO    - Total complete:   81/  88  92%  skipped: 1334, failed:   52
ERROR   - nxp_adsp_imx8             tests/kernel/fifo/fifo_timeout/kernel.fifo.timeout FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nxp_adsp_imx8/tests/kernel/fifo/fifo_timeout/kernel.fifo.timeout/build.log
INFO    - Total complete:   82/  88  93%  skipped: 1334, failed:   53
ERROR   - nxp_adsp_imx8             tests/kernel/lifo/lifo_api/kernel.lifo             FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nxp_adsp_imx8/tests/kernel/lifo/lifo_api/kernel.lifo/build.log
INFO    - Total complete:   83/  88  94%  skipped: 1334, failed:   54
ERROR   - nxp_adsp_imx8             tests/kernel/threads/thread_stack/kernel.threads.thread_stack FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nxp_adsp_imx8/tests/kernel/threads/thread_stack/kernel.threads.thread_stack/build.log
INFO    - Total complete:   84/  88  95%  skipped: 1334, failed:   55
ERROR   - nxp_adsp_imx8             tests/kernel/fifo/fifo_api/kernel.fifo             FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nxp_adsp_imx8/tests/kernel/fifo/fifo_api/kernel.fifo/build.log
INFO    - Total complete:   85/  88  96%  skipped: 1334, failed:   56
[... cut ...]
ERROR   - nxp_adsp_imx8             tests/kernel/interrupt/arch.interrupt              FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nxp_adsp_imx8/tests/kernel/interrupt/arch.interrupt/build.log
INFO    - Total complete:   87/  88  98%  skipped: 1334, failed:   58
ERROR   - nxp_adsp_imx8             tests/kernel/device/kernel.device.pm               FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nxp_adsp_imx8/tests/kernel/device/kernel.device.pm/build.log
```

galak, 2021-09-01:
@MaureenHelm FYI.

galak, 2021-09-01:
Most of the failures look of the form:

```
[ 93%] Building C object zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj
[ 94%] Linking C executable zephyr_prebuilt.elf
/opt/zephyr-sdk-0.13.0/xtensa/nxp_imx_adsp/xtensa-zephyr-elf/bin/../lib/gcc/xtensa-zephyr-elf/10.3.0/../../../../xtensa-zephyr-elf/bin/ld.bfd: zephyr_prebuilt.elf section `.Level2InterruptVector.text' 
will not fit in region `vector_int2_text'
/opt/zephyr-sdk-0.13.0/xtensa/nxp_imx_adsp/xtensa-zephyr-elf/bin/../lib/gcc/xtensa-zephyr-elf/10.3.0/../../../../xtensa-zephyr-elf/bin/ld.bfd: zephyr_prebuilt.elf section `.Level3InterruptVector.text' 
will not fit in region `vector_int3_text'
/opt/zephyr-sdk-0.13.0/xtensa/nxp_imx_adsp/xtensa-zephyr-elf/bin/../lib/gcc/xtensa-zephyr-elf/10.3.0/../../../../xtensa-zephyr-elf/bin/ld.bfd: zephyr_prebuilt.elf section `.DebugExceptionVector.text' will not fit in region `vector_int4_text'
/opt/zephyr-sdk-0.13.0/xtensa/nxp_imx_adsp/xtensa-zephyr-elf/bin/../lib/gcc/xtensa-zephyr-elf/10.3.0/../../../../xtensa-zephyr-elf/bin/ld.bfd: zephyr_prebuilt.elf section `.NMIExceptionVector.text' will not fit in region `vector_int5_text'
/opt/zephyr-sdk-0.13.0/xtensa/nxp_imx_adsp/xtensa-zephyr-elf/bin/../lib/gcc/xtensa-zephyr-elf/10.3.0/../../../../xtensa-zephyr-elf/bin/ld.bfd: section .Level3InterruptVector.text LMA [00000000596f859c,00000000596f85c4] overlaps section .Level2InterruptVector.text LMA [00000000596f857c,00000000596f85a4]
/opt/zephyr-sdk-0.13.0/xtensa/nxp_imx_adsp/xtensa-zephyr-elf/bin/../lib/gcc/xtensa-zephyr-elf/10.3.0/../../../../xtensa-zephyr-elf/bin/ld.bfd: section .DebugExceptionVector.text LMA [00000000596f85bc,00000000596f85e4] overlaps section .Level3InterruptVector.text LMA [00000000596f859c,00000000596f85c4]
/opt/zephyr-sdk-0.13.0/xtensa/nxp_imx_adsp/xtensa-zephyr-elf/bin/../lib/gcc/xtensa-zephyr-elf/10.3.0/../../../../xtensa-zephyr-elf/bin/ld.bfd: section .NMIExceptionVector.text LMA [00000000596f85dc,00000000596f8604] overlaps section .DebugExceptionVector.text LMA [00000000596f85bc,00000000596f85e4]
/opt/zephyr-sdk-0.13.0/xtensa/nxp_imx_adsp/xtensa-zephyr-elf/bin/../lib/gcc/xtensa-zephyr-elf/10.3.0/../../../../xtensa-zephyr-elf/bin/ld.bfd: section .KernelExceptionVector.text LMA [00000000596f85fc,00000000596f85fe] overlaps section .NMIExceptionVector.text LMA [00000000596f85dc,00000000596f8604]
/opt/zephyr-sdk-0.13.0/xtensa/nxp_imx_adsp/xtensa-zephyr-elf/bin/../lib/gcc/xtensa-zephyr-elf/10.3.0/../../../../xtensa-zephyr-elf/bin/ld.bfd: region `vector_int2_text' overflowed by 13 bytes
/opt/zephyr-sdk-0.13.0/xtensa/nxp_imx_adsp/xtensa-zephyr-elf/bin/../lib/gcc/xtensa-zephyr-elf/10.3.0/../../../../xtensa-zephyr-elf/bin/ld.bfd: region `vector_int3_text' overflowed by 13 bytes
/opt/zephyr-sdk-0.13.0/xtensa/nxp_imx_adsp/xtensa-zephyr-elf/bin/../lib/gcc/xtensa-zephyr-elf/10.3.0/../../../../xtensa-zephyr-elf/bin/ld.bfd: region `vector_int4_text' overflowed by 13 bytes
/opt/zephyr-sdk-0.13.0/xtensa/nxp_imx_adsp/xtensa-zephyr-elf/bin/../lib/gcc/xtensa-zephyr-elf/10.3.0/../../../../xtensa-zephyr-elf/bin/ld.bfd: region `vector_int5_text' overflowed by 13 bytes
collect2: error: ld returned 1 exit status
```

cfriedt, 2021-09-07:
@iuliana-prodan, @dbaluta - are you able to look into this?

dbaluta, 2021-09-08:
@cfriedt will have a week in the following days! Thanks for  your patience.

iuliana-prodan, 2021-09-08:
> Most of the failures look of the form:
>
> ```
> [ 93%] Building C object zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj
> [ 94%] Linking C executable zephyr_prebuilt.elf [...]

Hi @galak, 

Can you, please, let me know, how did you get these errors?

I've just built Zephyr for SOF and is successful:

```
nxa06898@lsv15040:/opt/samba/nxa06898/zephyrproject$ west build -d imx -b nxp_adsp_imx8 zephyr/samples/subsys/audio/sof/
-- west build: generating a build system
Including boilerplate (Zephyr base): /opt/samba/nxa06898/zephyrproject/zephyr/cmake/app/boilerplate.cmake
CMake Deprecation Warning at /opt/samba/nxa06898/zephyrproject/zephyr/cmake/app/boilerplate.cmake:37 (cmake_policy):
  The OLD behavior for policy CMP0079 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  /opt/samba/nxa06898/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /opt/samba/nxa06898/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:35 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Application: /opt/samba/nxa06898/zephyrproject/zephyr/samples/subsys/audio/sof
-- Zephyr version: 2.7.0-rc1 (/opt/samba/nxa06898/zephyrproject/zephyr), build: v2.7.0-rc1-45-gfa7959c19786
-- Found Python3: /usr/bin/python3.6 (found suitable exact version "3.6.9") found components: Interpreter 
-- Found west (found suitable version "0.11.1", minimum required is "0.7.1")
-- Board: nxp_adsp_imx8
-- Cache files will be written to: /home/nxa06898/.cache/zephyr
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Using toolchain: zephyr 0.13.0 (/opt/zephyr-sdk-0.13.0)
-- Found dtc: /opt/zephyr-sdk-0.13.0/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: /opt/samba/nxa06898/zephyrproject/zephyr/boards/xtensa/nxp_adsp_imx8/nxp_adsp_imx8.dts
-- Generated zephyr.dts: /opt/samba/nxa06898/zephyrproject/imx/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /opt/samba/nxa06898/zephyrproject/imx/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /opt/samba/nxa06898/zephyrproject/imx/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /opt/samba/nxa06898/zephyrproject/imx/zephyr/dts.cmake
Parsing /opt/samba/nxa06898/zephyrproject/zephyr/Kconfig
Loaded configuration '/opt/samba/nxa06898/zephyrproject/zephyr/boards/xtensa/nxp_adsp_imx8/nxp_adsp_imx8_defconfig'
Merged configuration '/opt/samba/nxa06898/zephyrproject/zephyr/samples/subsys/audio/sof/prj.conf'
Merged configuration '/opt/samba/nxa06898/zephyrproject/zephyr/samples/subsys/audio/sof/boards/nxp_adsp_imx8.conf'
Configuration saved to '/opt/samba/nxa06898/zephyrproject/imx/zephyr/.config'
Kconfig header saved to '/opt/samba/nxa06898/zephyrproject/imx/zephyr/include/generated/autoconf.h'

warning: SMP_BOOT_DELAY (defined at kernel/Kconfig:785) was assigned the value 'y' but got the value
'n'. Check these unsatisfied dependencies: SMP (=n). See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_SMP_BOOT_DELAY.html and/or look up
SMP_BOOT_DELAY in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

-- The C compiler identification is GNU 10.3.0
-- The CXX compiler identification is GNU 10.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/zephyr-sdk-0.13.0/xtensa/nxp_imx_adsp/xtensa-zephyr-elf/bin/xtensa-zephyr-elf-gcc
-- Found Python3: /usr/bin/python3.6 (found version "3.6.9") found components: Interpreter 
-- version.cmake starting SOF build at 2021-09-08T14:52:25Z UTC
-- Building git commit with parent(s):
fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
-- GIT_TAG / GIT_LOG_HASH : v1.5-rc1-2000-g76feb11d1b8f / 76feb11d1
CMake Warning at /opt/samba/nxa06898/zephyrproject/modules/audio/sof/scripts/cmake/version.cmake:116 (message):
  Source content hash not computed without git, using GIT_LOG_HASH instead
Call Stack (most recent call first):
[... 82 lines cut ...]
INFO    - Total complete:    3/  88   3%  skipped: 1316, failed:    3
ERROR   - nxp_adsp_imx8             tests/kernel/early_sleep/kernel.common.sleep       FAILED: Build failure
ERROR   - see: /opt/samba/nxa06898/zephyrproject/zephyr/twister-out/nxp_adsp_imx8/tests/kernel/early_sleep/kernel.common.sleep/build.log
...
```

But how can I reproduce the errors I see above, the ones from here:
```
[ 94%] Linking C executable zephyr_prebuilt.elf
/opt/zephyr-sdk-0.13.0/xtensa/nxp_imx_adsp/xtensa-zephyr-elf/bin/../lib/gcc/xtensa-zephyr-elf/10.3.0/../../../../xtensa-zephyr-elf/bin/ld.bfd: zephyr_prebuilt.elf section `.Level2InterruptVector.text' 
will not fit in region `vector_int2_text'
/opt/zephyr-sdk-0.13.0/xtensa/nxp_imx_adsp/xtensa-zephyr-elf/bin/../lib/gcc/xtensa-zephyr-elf/10.3.0/../../../../xtensa-zephyr-elf/bin/ld.bfd: zephyr_prebuilt.elf section `.Level3InterruptVector.text' 
will not fit in region `vector_int3_text'
[... cut ...]
/opt/zephyr-sdk-0.13.0/xtensa/nxp_imx_adsp/xtensa-zephyr-elf/bin/../lib/gcc/xtensa-zephyr-elf/10.3.0/../../../../xtensa-zephyr-elf/bin/ld.bfd: region `vector_int3_text' overflowed by 13 bytes
/opt/zephyr-sdk-0.13.0/xtensa/nxp_imx_adsp/xtensa-zephyr-elf/bin/../lib/gcc/xtensa-zephyr-elf/10.3.0/../../../../xtensa-zephyr-elf/bin/ld.bfd: region `vector_int4_text' overflowed by 13 bytes
/opt/zephyr-sdk-0.13.0/xtensa/nxp_imx_adsp/xtensa-zephyr-elf/bin/../lib/gcc/xtensa-zephyr-elf/10.3.0/../../../../xtensa-zephyr-elf/bin/ld.bfd: region `vector_int5_text' overflowed by 13 bytes
collect2: error: ld returned 1 exit status
```

dbaluta, 2021-09-08:
@iuliana-prodan have you tried `./scripts/twister -p nxp_adsp_imx8`. I think it is trying to build nxp target with different config option. A thing that definitely SOF hasn't been tested enough.

iuliana-prodan, 2021-09-08:
> @iuliana-prodan have you tried `./scripts/twister -p nxp_adsp_imx8`. I think it is trying to build nxp target with different config option. A thing that definitely SOF hasn't been tested enough.

@dbaluta Yes, I've tried and I mentioned that above.

@galak I can see those errors now, I just need to look in each` build.log` from `twister.out` folder.

iuliana-prodan, 2021-09-09:
@galak @cfriedt @dbaluta I've fixed the issue with PR https://github.com/zephyrproject-rtos/zephyr/pull/38431
Here's the log:
```
nxa06898@lsv15040:/opt/samba/nxa06898/zephyrproject/zephyr$ ./scripts/twister -p nxp_adsp_imx8
ZEPHYR_BASE unset, using "/opt/samba/nxa06898/zephyrproject/zephyr"
Renaming output directory to /opt/samba/nxa06898/zephyrproject/zephyr/twister-out.1
INFO    - Zephyr version: v2.7.0-rc1-56-g9efd4e1cc869
INFO    - JOBS: 8
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testcase list...
INFO    - 1464 test scenarios (1404 configurations) selected, 1316 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - Total complete:   88/  88  100%  skipped: 1344, failed:    0
INFO    - 60 of 60 test configurations passed (100.00%), 0 failed, 1344 skipped with 0 warnings in 106.74 seconds
INFO    - In total 626 test cases were executed, 18903 skipped on 1 out of total 394 platforms (0.25%)
INFO    - 0 test configurations executed on platforms, 60 test configurations were only built.
INFO    - Saving reports...
INFO    - Writing xunit report /opt/samba/nxa06898/zephyrproject/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /opt/samba/nxa06898/zephyrproject/zephyr/twister-out/twister_report.xml...
INFO    - Run completed
nxa06898@lsv15040:/opt/samba/nxa06898/zephyrproject/zephyr$ 

```
cc: @dleach02 @MaureenHelm

PR #38431 (2021-09-10): boards: xtensa: adsp: select IMX for imx8 board
For some platforms, like NXP's IMX8 or Mediatek's MT8195,
the size of an interrupt vector table entry is 0x1C bytes,
less than usual (0x30 for Intel's platforms).
So, the interrupt handlers don't fit in the vector table entries.
    
I've added a small indirection to bypass this size
constraint and moved the default handlers to the end
of vector table, renaming them to _Level\LVL\()VectorHelper.
For this, I've added a generic configuration -
XTENSA_SMALL_VECTOR_TABLE_ENTRY and enabled it for
NXP_ADSP_IMX8 board.

Fixes #38216
commit fa20ebe4 (2021-09-10): boards: xtensa: adsp: select XTENSA_SMALL_VECTOR_TABLE_ENTRY for imx8 board

Add CONFIG_XTENSA_SMALL_VECTOR_TABLE_ENTRY
for NXP_ADSP_IMX8 board to bypass the size
constraint of the interrupt vector table entry.

Fixes #38216
closed 2021-09-10 by nashif