zephyrproject-rtos/zephyr #38237: [backport v2.6-branch] backport of #37479 failed

zephyrbot, 2021-09-02:
This issue was created automatically because the backport of #37479 to `v2.6-branch` failed.

The backport to `v2.6-branch` failed:
```
The process '/usr/bin/git' failed with exit code 1
```
To backport manually, run these commands in your terminal:
```bash
# Fetch latest updates from GitHub
git fetch
# Create a new working tree
git worktree add .worktrees/backport-v2.6-branch v2.6-branch
# Navigate to the new working tree
cd .worktrees/backport-v2.6-branch
# Create a new branch
git switch --create backport-37479-to-v2.6-branch
# Cherry-pick the merged commits of this pull request and resolve the conflicts
git cherry-pick 114b84a58b0892146441f0ac82591665197b065c
# Push it to GitHub
git push --set-upstream origin backport-37479-to-v2.6-branch
# Go back to the original working tree
cd ../..
# Delete the working tree
git worktree remove .worktrees/backport-v2.6-branch
```
Then, create a pull request where the `base` branch is `v2.6-branch` and the `compare`/`head` branch is `backport-37479-to-v2.6-branch`.

maksimmasalski, 2021-09-03:
@scottwcpg Scott, could you help to backport it to the 2.6 branch?

maksimmasalski, 2021-09-07:
Created a backport to the 2.6 branch #38358