zephyrproject-rtos/zephyr #38242: drivers: wifi: esp_at: support offloaded WiFi networking using ESP-AT over SPI

mniestroj, 2021-09-02:
**Is your enhancement proposal related to a problem? Please describe.**
So far there is only UART interface supported by [esp_at](https://github.com/zephyrproject-rtos/zephyr/tree/main/drivers/wifi/esp_at) WiFi drivers. Firmware that is used on ESP (e.g. ESP32 or ESP8266) supports SPI (among others listed in [esp-at firmware repo](https://github.com/espressif/esp-at/tree/master/main/interface)) interface as an alternative to UART, still using lots of common features like AT commands as serialization format. There are number of boards on the market that use SPI as interface between host processor and WiFi chip, such as:
- [Arduino Nano 33 IoT](https://docs.zephyrproject.org/latest/boards/arm/arduino_nano_33_iot/doc/index.html) ([schematics](https://docs.arduino.cc/static/a24d516d159aa9990c60fdd270803ee9/schematics.pdf))

**Describe the solution you'd like**
Add SPI as new interface in [modem_context](https://github.com/zephyrproject-rtos/zephyr/tree/main/drivers/modem) framework, which is used by [esp_at](https://github.com/zephyrproject-rtos/zephyr/tree/main/drivers/wifi/esp_at) WiFi drivers. Extend [esp_at](https://github.com/zephyrproject-rtos/zephyr/tree/main/drivers/wifi/esp_at) drivers to support SPI interface, by adding SPI bus bindings and corresponding initialization code.

**Describe alternatives you've considered**
Extend [esp_at](https://github.com/zephyrproject-rtos/zephyr/tree/main/drivers/wifi/esp_at) directly without touching [modem_context](https://github.com/zephyrproject-rtos/zephyr/tree/main/drivers/modem), if there is no way of doing AT commands over SPI in a generic way.

beriberikix, 2021-09-02:
Here's some additional boards/modules that I found that either only support SPI or default to SPI:
* [SiFive HiFive1](https://docs.zephyrproject.org/latest/boards/riscv/hifive1/doc/index.html)/[RevB](https://docs.zephyrproject.org/latest/boards/riscv/hifive1_revb/doc/index.html)
* All adafruit boards that use the ESP32 a modem default to ESP32 (though UART can be used):
  * [Metro M4 Express AirLift](https://learn.adafruit.com/adafruit-metro-m4-express-airlift-wifi/pinouts-2#esp32-wifi-co-processor-3026426-6)
  * [PyPortal](https://learn.adafruit.com/adafruit-pyportal/pinouts#wifi-3020485-7), [PyPortal Pynt](https://learn.adafruit.com/adafruit-pyportal/pinouts#wifi-3020485-7), [PyPortal Titano](https://learn.adafruit.com/adafruit-pyportal-titano/pinouts#wifi-3050111-5)
  * All their coprocessors: [Airlift](https://learn.adafruit.com/adafruit-airlift-breakout/pinouts#spi-and-control-pins-3024866-6), [Airlift shield](https://learn.adafruit.com/adafruit-airlift-shield-esp32-wifi-co-processor/pinouts#spi-interface-pins-3031215-6), [Airlift ItsyBitsy](https://learn.adafruit.com/adafruit-airlift-bitsy-add-on-esp32-wifi-co-processor/pinouts#spi-and-control-pins-3038690-5)
* [Arduino Nano RP2040](https://store-usa.arduino.cc/collections/boards/products/arduino-nano-rp2040-connect)

/cc @tannewt @sylvioalves

beriberikix, 2023-09-13:
Can we add the [Platform: Espressif](https://github.com/zephyrproject-rtos/zephyr/labels/platform%3A%20ESP32) label to this issue to increase visibility? As it currently stands, it's really an Espressif-specific implementation.

mniestroj, 2023-09-13:
> Can we add the [Platform: Espressif](https://github.com/zephyrproject-rtos/zephyr/labels/platform%3A%20ESP32) label to this issue to increase visibility? As it currently stands, it's really an Espressif-specific implementation.

Any host MCU vendor can be used. The only Espressif-specific part is the peripheral device. In that case I think we don't specify tags per vendor. It is also not limited to Espressif hardware, as any other WiFi chip acting as SPI peripheral can be used, assuming that the same AT commands will be implemented.

beriberikix, 2023-09-13:
Ah, I didn't realize that this was abstracted in some way! The `esp_at` gave me the impression that's it was Espressif-specific. Can it be expanded to other AT-based Wi-Fi protocols, ala the new `modem` subsystem? AFAIK while [Espressif's AT protocol](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/index.html) is based on `3GPP TS 27.007`, it's all custom from Espressif. And I believe other Wi-Fi vendors also create their own protocols as well? AFAIK, only Espressif uses this AT command set, but I certainly haven't looked!