zephyrproject-rtos/zephyr #38246: samples: drivers: flash_shell: fails on arduino_due due to compilation issue

cfriedt, 2021-09-02:
**Describe the bug**
`samples/drivers/flash_shell/sample.drivers.flash.shell`  fails to build for `arduino_due` with a compilation error.

`IFLASH_PAGE_SIZE`, `EEFC_FSR_FLERR`, `EEFC_FCR_FCMD_EPA` undeclared.

Possibly some additional DT issues.

**To Reproduce**
Steps to reproduce the behavior:
1.`west build -b arduino_due samples/drivers/flash_shell`
2. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
```
FAILED: zephyr/drivers/flash/CMakeFiles/drivers__flash.dir/flash_sam.c.obj
--
  | ccache /opt/toolchains/zephyr-sdk-0.13.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBUILD_VERSION=zephyr-v2.6.0-2928-g626b1b79df62 -DKERNEL -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR_SUPERVISOR__ -D__ZEPHYR__=1 -I../../../../../../include -Izephyr/include/generated -I../../../../../../soc/arm/atmel_sam/sam3x -I../../../../../../lib/util/fnmatch/. -I../../../../../../soc/arm/atmel_sam/common/. -I/var/lib/buildkite-agent/zephyr-module-cache/modules/hal/cmsis/CMSIS/Core/Include -I/var/lib/buildkite-agent/zephyr-module-cache/modules/hal/atmel/asf/sam/include/sam3x -isystem ../../../../../../lib/libc/minimal/include -isystem /opt/toolchains/zephyr-sdk-0.13.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/include -isystem /opt/toolchains/zephyr-sdk-0.13.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/include-fixed -Os -imacros /workdir/zephyr/twister-out/arduino_due/samples/drivers/flash_shell/sample.drivers.flash.shell/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m3 -mthumb -mabi=aapcs -mfp16-format=ieee -imacros /workdir/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-address-of-packed-member -Wno-unused-but-set-variable -Werror=implicit-int -Werror -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/workdir/zephyr/samples/drivers/flash_shell=CMAKE_SOURCE_DIR -fmacro-prefix-map=/workdir/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/workdir=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT zephyr/drivers/flash/CMakeFiles/drivers__flash.dir/flash_sam.c.obj -MF zephyr/drivers/flash/CMakeFiles/drivers__flash.dir/flash_sam.c.obj.d -o zephyr/drivers/flash/CMakeFiles/drivers__flash.dir/flash_sam.c.obj -c /workdir/zephyr/drivers/flash/flash_sam.c
  | /workdir/zephyr/drivers/flash/flash_sam.c: In function 'flash_sam_get_page':
  | /workdir/zephyr/drivers/flash/flash_sam.c:90:18: error: 'IFLASH_PAGE_SIZE' undeclared (first use in this function); did you mean 'IFLASH1_PAGE_SIZE'?
  | 90 \|  return offset / IFLASH_PAGE_SIZE;
  | \|                  ^~~~~~~~~~~~~~~~
  | \|                  IFLASH1_PAGE_SIZE
  | /workdir/zephyr/drivers/flash/flash_sam.c:90:18: note: each undeclared identifier is reported only once for each function it appears in
  | /workdir/zephyr/drivers/flash/flash_sam.c: In function 'flash_sam_wait_ready':
  | /workdir/zephyr/drivers/flash/flash_sam.c:108:13: error: 'EEFC_FSR_FLERR' undeclared (first use in this function); did you mean 'EEFC_FSR_FRDY'?
  | 108 \|   if (fsr & EEFC_FSR_FLERR) {
  | \|             ^~~~~~~~~~~~~~
  | \|             EEFC_FSR_FRDY
  | /workdir/zephyr/drivers/flash/flash_sam.c: In function 'flash_sam_write':
  | /workdir/zephyr/drivers/flash/flash_sam.c:202:27: error: 'IFLASH_PAGE_SIZE' undeclared (first use in this function); did you mean 'IFLASH1_PAGE_SIZE'?
  | 202 \|    eop_len = -(offset \| ~(IFLASH_PAGE_SIZE - 1));
  | \|                           ^~~~~~~~~~~~~~~~
  | \|                           IFLASH1_PAGE_SIZE
  | /workdir/zephyr/drivers/flash/flash_sam.c: In function 'flash_sam_erase_block':
  | /workdir/zephyr/drivers/flash/flash_sam.c:251:4: error: 'EEFC_FCR_FCMD_EPA' undeclared (first use in this function); did you mean 'EEFC_FCR_FCMD_EA'?
  | 251 \|    EEFC_FCR_FCMD_EPA;
  | \|    ^~~~~~~~~~~~~~~~~
  | \|    EEFC_FCR_FCMD_EA
  | In file included from ../../../../../../include/arch/arm/aarch32/arch.h:20,
  | from ../../../../../../include/arch/cpu.h:19,
  | from ../../../../../../include/kernel_includes.h:33,
  | from ../../../../../../include/kernel.h:17,
  | from ../../../../../../include/init.h:11,
  | from ../../../../../../include/device.h:29,
  | from /workdir/zephyr/drivers/flash/flash_sam.c:13:
  | /workdir/zephyr/drivers/flash/flash_sam.c: In function 'flash_sam_erase':
  | zephyr/include/generated/devicetree_unfixed.h:3467:34: error: 'DT_N_S_soc_S_flash_controller_400e0a00_S_flash_80000_P_erase_block_size' undeclared (first use in this function); did you mean 'DT_N_S_soc_S_flash_controller_400e0a00_S_flash_80000_P_write_block_size'?
  | 3467 \| #define DT_N_INST_0_soc_nv_flash DT_N_S_soc_S_flash_controller_400e0a00_S_flash_80000
  | \|                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  | ../../../../../../include/devicetree.h:2641:24: note: in definition of macro 'DT_CAT'
  | 2641 \| #define DT_CAT(a1, a2) a1 ## a2
  | \|                        ^~
  | /workdir/zephyr/drivers/flash/flash_sam.c:11:28: note: in expansion of macro 'DT_PROP'
  | 11 \| #define FLASH_ERASE_BLK_SZ DT_PROP(SOC_NV_FLASH_NODE, erase_block_size)
  | \|                            ^~~~~~~
  | ../../../../../../include/sys/util_internal.h:85:36: note: in expansion of macro 'DT_N_INST_0_soc_nv_flash'
  | 85 \| #define UTIL_PRIMITIVE_CAT(a, ...) a##__VA_ARGS__
  | \|                                    ^
  | ../../../../../../include/sys/util_internal.h:84:26: note: in expansion of macro 'UTIL_PRIMITIVE_CAT'
  | 84 \| #define UTIL_CAT(a, ...) UTIL_PRIMITIVE_CAT(a, __VA_ARGS__)
  | \|                          ^~~~~~~~~~~~~~~~~~
  | ../../../../../../include/devicetree.h:305:31: note: in expansion of macro 'UTIL_CAT'
  | 305 \| #define DT_INST(inst, compat) UTIL_CAT(DT_N_INST, DT_DASH(inst, compat))
  | \|                               ^~~~~~~~
[... 51 lines cut ...]
  | \|                          ^~~~~~~~~~~~~~~~~~
  | ../../../../../../include/devicetree.h:305:31: note: in expansion of macro 'UTIL_CAT'
  | 305 \| #define DT_INST(inst, compat) UTIL_CAT(DT_N_INST, DT_DASH(inst, compat))
  | \|                               ^~~~~~~~
  | /workdir/zephyr/drivers/flash/flash_sam.c:8:27: note: in expansion of macro 'DT_INST'
  | 8 \| #define SOC_NV_FLASH_NODE DT_INST(0, soc_nv_flash)
  | \|                           ^~~~~~~
  | /workdir/zephyr/drivers/flash/flash_sam.c:11:36: note: in expansion of macro 'SOC_NV_FLASH_NODE'
  | 11 \| #define FLASH_ERASE_BLK_SZ DT_PROP(SOC_NV_FLASH_NODE, erase_block_size)
  | \|                                    ^~~~~~~~~~~~~~~~~
  | /workdir/zephyr/drivers/flash/flash_sam.c:340:50: note: in expansion of macro 'FLASH_ERASE_BLK_SZ'
  | 340 \|  .pages_count = DT_REG_SIZE(SOC_NV_FLASH_NODE) / FLASH_ERASE_BLK_SZ,
  | \|                                                  ^~~~~~~~~~~~~~~~~~
  | /workdir/zephyr/drivers/flash/flash_sam.c: In function 'flash_sam_get_page':
  | /workdir/zephyr/drivers/flash/flash_sam.c:91:1: error: control reaches end of non-void function [-Werror=return-type]
  | 91 \| }
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

galak, 2021-09-02:
This is related to commit 626b1b79df62bcdc4142cb671693ea9f31f10909

nandojve, 2021-09-02:
Hi @cfriedt ,

Is this still valid after #38263?

cfriedt, 2021-09-02:
I think it might be fixed already. Thanks for pointing it out

cfriedt, 2021-09-02:
Can you do a quick check to confirm?

nandojve, 2021-09-03:
I could reproduce problem after #38263

closed 2021-09-03 by nandojve