zephyrproject-rtos/zephyr #38251: cmake: DTC_OVERLAY_FILE flags cancel board <board>.overlay files

marinjurjevic, 2021-09-02:
**Is your enhancement proposal related to a problem? Please describe.**
The problem is that when DTC_OVERLAY_FILE flag is passed to cmake, it will **silently** drop `boards/<BOARD>.overlay` and `<BOARD>.overlay`.  This is due to following lines in `boilerplate.cmake`:

https://github.com/zephyrproject-rtos/zephyr/blob/bdc80ebe16c82fa589ce9c750e66d2f319123b8f/cmake/app/boilerplate.cmake#L499-L512

Documentation was not helpful. [Devicetree HOWTOs](https://docs.zephyrproject.org/latest/guides/dts/howtos.html#set-devicetree-overlays) describe `some ways to set` DTC_OVERLAY_FILE. No where it states they are mutually exclusive (my case covers 5. and 6. bulletpoint). 

**Describe the solution you'd like**
A clear documented way how are these files included and what is mutually exclusive, what is not. 
Good reference point for these kind of things is Nordic's documentation on [multi-image builds](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/1.6.0/nrf/ug_multi_image.html#image-specific-variables). 

**Describe alternatives you've considered**
If solution is not acceptable for whatever reason, at least a warning in logs should bring attention of developer that some files are being dropped.

**Additional context**
Relevant folder structure is below. I am using a build script which wraps west execution by handling different use-cases.  Config folder contains optional overlay/conf files which may or not may be used. Folder `boards/` contains overlay files which are project-specific, but I expect them to always be included whenever project is built for that board. 

This problem does not affect Kconfig overlays, only DTS. 

```bash
├── boards/
│  ├── actinius_icarus_ns.conf
│  ├── actinius_icarus_ns.overlay
├── config/
│  ├── external_flash_w25q128jv_actinius_icarus.overlay
├── src/
```

marinjurjevic, 2021-09-02:
I'm currently using following hotfix for `boards/<BOARD>.overlay` specifically. Basically just appending the overlay with other optional DTS files and it ends up as part of DTC_OVERLAY_FILE.

```python
if APP_DT_OVERLAYS:

    # Check if boards/<board>.overlay exists
    boards_overlay_file = "boards/{}.overlay".format(TARGET_BOARD)
    if os.path.isfile(boards_overlay_file):
        print("Found {}".format(boards_overlay_file))
        APP_DT_OVERLAYS.add(boards_overlay_file)

# application
APP_KCONFIG_OVERLAYS_STR = ';'.join(APP_KCONFIG_OVERLAYS)
APP_DT_OVERLAYS_STR = ';'.join(APP_DT_OVERLAYS)
```

results in 

```bash
west build -b actinius_icarus_ns --  -DDTC_OVERLAY_FILE="boards/actinius_icarus_ns.overlay;config/external_flash_w25q128jv_actinius_icarus.overlay"
```

tejlmand, 2021-10-01:
> Documentation was not helpful. Devicetree HOWTOs describe some ways to set DTC_OVERLAY_FILE. No where it states they are mutually exclusive (my case covers 5. and 6. bulletpoint).

I see that it might be too vaguely described, I will try to come up with a proposal that should make this clearer to the reader.

tejlmand, 2021-10-01:
> Folder boards/ contains overlay files which are project-specific, but I expect them to always be included whenever project is built for that board.

The reason for disabling the automatic inclusion of devicetree overlays when a user is specifying `DTC_OVERLAY_FILE` is because then the user has indirectly requested full control.

If we didn't disable automatic inclusion, then there would be no way for a user to request `no inclusion` in such case.
That would require us to introduce one more build flag, like `-DDTC_OVERLAY_FILE=<file1.overlay> -DDTC_AUTO_OVERLAY=n`.

Why do this, when a user wanting to specify both a custom devicetree overlay, but also wants to use the automatic file can already do: `-DDTC_OVERLAY_FILE="<file1.overlay>;boards/<board>.overlay"`

> If solution is not acceptable for whatever reason, at least a warning in logs should bring attention of developer that some files are being dropped.

The log already informs the user of which overlays are being used, for example if I build for a `nrf52840dk_nrf52840` on a sample that has a `boards/nrf52840dk_nrf52840.overlay` then I will see:
```
-- Found BOARD.dts: /.../zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
-- Found devicetree overlay: /.../boards/nrf52840dk_nrf52840.overlay
-- Generated zephyr.dts: /projects/github/ncs/zephyr/build/zephyr/zephyr.dts
```
but if I use `-DDTC_OVERLAY_FILE=my.overlay` I will see this instead:
```
-- Found BOARD.dts: /.../zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
-- Found devicetree overlay: my.overlay
-- Generated zephyr.dts: /projects/github/ncs/zephyr/build/zephyr/zephyr.dts
```

> This problem does not affect Kconfig overlays, only DTS.

Not sure I understand, if a user specifies `-DCONF_FILE=my.conf` then all automatic inclusion of board specific Kconfig fragments are also disabled. conf files and overlay files are behaving identical in this way.
https://docs.zephyrproject.org/latest/guides/build/kconfig/setting.html#the-initial-configuration

marinjurjevic, 2021-10-01:
I understand the reason now. But I don't care `why` is something working, but `how` is something working. I would like to have `how` explained in detail, now `why`. 

If you look at [Set devicetree overlays](https://docs.zephyrproject.org/latest/guides/dts/howtos.html#set-devicetree-overlays) page, it's not totally clear when and how are files included. I would like to have rules explained in an unambiguous way, that's why I pointed out example of how Nordic does it.

tejlmand, 2021-10-01:
> But I don't care why is something working, but how is something working. 

Did you miss this comment ?
https://github.com/zephyrproject-rtos/zephyr/issues/38251#issuecomment-932117431
> I see that it might be too vaguely described, I will try to come up with a proposal that should make this clearer to the reader.

And that has already been done: #39034

marinjurjevic, 2021-10-01:
> > But I don't care why is something working, but how is something working.
> 
> Did you miss this comment ? [#38251 (comment)](https://github.com/zephyrproject-rtos/zephyr/issues/38251#issuecomment-932117431)
> 
> > I see that it might be too vaguely described, I will try to come up with a proposal that should make this clearer to the reader.
> 
> And that has already been done: #39034

I just saw the PR, I'm looking at it now.

marinjurjevic, 2021-10-01:
> Not sure I understand, if a user specifies `-DCONF_FILE=my.conf` then all automatic inclusion of board specific Kconfig fragments are also disabled. conf files and overlay files are behaving [...]

I believe my source of confusion comes from this page actually -> [Important Build System Variables](https://docs.zephyrproject.org/latest/application/index.html#important-build-system-variables). I've been using `-DOVERLAY_CONFIG=` for adding Kconfig overlays. 
If you build with these OVERLAY_CONFIG variables set you get the behavior I was seeing with DTC.

```bash
west build -b nrf52840dk_nrf52840 -- -DCONF_FILE="my.conf"
...
Parsing /home/marin/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/marin/zephyrproject/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840_defconfig'
Merged configuration 'my.conf'
```

tejlmand, 2021-10-01:
Honestly, I agree with you, having `OVERLAY` in those two names `OVERLAY_CONFIG` and `DTC_OVERLAY_FILE` that have very different behavior can be very confusing.

But I guess this will be hard to change, as it has been there since the KBuild days, and was ported along to CMake here:
https://github.com/zephyrproject-rtos/zephyr/blob/12f8f7616567f224b920512b6f81a65421e30bae/cmake/kconfig.cmake#L60

marinjurjevic, 2021-10-01:
> Honestly, I agree with you, having `OVERLAY` in those two names `OVERLAY_CONFIG` and `DTC_OVERLAY_FILE` that have very different behavior can be very confusing.
>
> But I guess this will be hard to [...]

I understand, and I'm okey with things being as they are. I would just like to have things explained from top-level perspective and in single place (preferably in pages closer to new people in zephyr) that's doesn't require digging deep into documentation (or even worse source code of cmake functions, which can be nightmare for beginners). 

I must admit I am taking cmake/kconfig/dts infrastructure to extreme and applying almost all the rules. Most projects probably won't even trigger rules I run into. But if they do, I'm sure having unified documentation for all of them would be appreciated. Kconfig has been moved to Build Configuration and Systems in v2.7, where as Devicetree is still under User and Developer Guides. It can be really hard to group these informations and organize them in sensible way during learning phase.

nashif, 2022-11-19:
seems to be resolved, no further action required?

closed 2022-11-19 by nashif