zephyrproject-rtos/zephyr #38258: newlib: first malloc call may fail on Xtensa depending on image size

stephanosio, 2021-09-02:
**Describe the bug**
The first `malloc` call may fail on the Xtensa platforms (e.g. `qemu_xtensa`) if the requested block size is such that the `malloc` function invokes the `sbrk` function with the heap expansion size that causes the resulting heap address to fall on the 0x1000 boundary (what newlib thinks is "page boundary").

For example, if the initial heap address is `0x60009fd0` and `malloc` is invoked to allocate a 16-byte block, the `malloc` attempts to increase the heap size by 48 by calling `sbrk`, and this causes `sbrk` to return `0x6000a000`, which is a 0x1000-aligned address.

**To Reproduce**
TODO: Post a testable sample without all the unnecessary noises.

**Expected behavior**
`malloc` always succeeds.

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v2.6.0-2927-ga77e7566a8bb  ***
sbrk: 0x60009fd0, 48
sbrk: 0x6000a000, 0
0, (nil)
sbrk: 0x6000a000, 4096
1, 0x60009fe0
2, 0x6000a000
3, 0x6000a020
...
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Toolchain: Zephyr SDK 0.13.0, [zephyr-newlib 3.3.0](https://github.com/zephyrproject-rtos/newlib-cygwin/tree/zephyr-newlib-3.3.0)
 - Commit SHA: a77e7566a8bb0135b52737e62fd1d6503b7b057a

**Additional context**
This only happens on Xtensa -- I have reproduced a similar case on other archs, but they all work as expected:

ARM (qemu_cortex_m3)
```
*** Booting Zephyr OS build zephyr-v2.6.0-2927-ga77e7566a8bb  ***
sbrk: 0x20001fd0, 48
sbrk: 0x20002000, 0
0, 0x20001fd8
sbrk: 0x20002000, 4096
1, 0x20001ff8
2, 0x20002018
3, 0x20002038
...
```

SPARC (qemu_leon3)
```
*** Booting Zephyr OS build zephyr-v2.6.0-2927-ga77e7566a8bb  ***
sbrk: 0x4000efd0, 48
sbrk: 0x4000f000, 0
0, 0x4000efd8
sbrk: 0x4000f000, 4096
1, 0x4000eff8
2, 0x4000f018
3, 0x4000f038
...
```

This is likely a newlib bug.

What is special about Xtensa, compared to other archs, is that its `MALLOC_ALIGNMENT` is 16 rather than 8, and this is causing the newlib malloc algorithm to produce a mishap not observed on the archs with the `MALLOC_ALIGNMENT` of 8.
https://github.com/zephyrproject-rtos/newlib-cygwin/blob/4f5997d3c0f9011135e9627dad700c9d64be4a4b/newlib/libc/stdlib/mallocr.c#L1399-L1401
https://github.com/zephyrproject-rtos/newlib-cygwin/blob/b1fe4401fdbd0860e0b91227219b15d2e0142b78/newlib/libc/include/sys/config.h#L198

stephanosio, 2021-09-02:
NOTE: This issue exists to keep track of this rather serious bug with the newlib on Xtensa. It needs to be fixed in the Zephyr SDK, and also in the upstream newlib if it has not already been patched in the latest releases.

galak, 2021-09-02:
@sylvioalves do you know if you guys newlib version has this issue or a fix for it?

stephanosio, 2021-09-02:
cc @dcpleung @andyross

stephanosio, 2021-09-02:
https://github.com/zephyrproject-rtos/newlib-cygwin/commit/b67ccc265089ded89e9170fcc5a8e2e43d020393#diff-f056d0212a247fc9bf5137a683fa14d9f1f80f23f76eedb68cf81710207fdb27R198

To all Xtensa maintainers: how important is it that we enforce `MALLOC_ALIGNMENT = 16`? Can't we live with `MALLOC_ALIGNMENT = 8` as do the rest of the archs we have?

My hunch is that this issue will go away if we just remove that line. Of course, the newlib bug persists, but at least Xtensa should no longer be affected by it.

stephanosio, 2021-09-02:
Also cc @jcmvbkbc, if you are around.

jcmvbkbc, 2021-09-02:
> how important is it that we enforce MALLOC_ALIGNMENT = 16?

I believe that 16 is not required for the base xtensa architecture with or without standard FP coprocessors. But 16 or even higher alignment may be required in the presence of TIE extensions, so it should really not be hardcoded but taken from the xtensa core configuration.

stephanosio, 2021-09-03:
> I believe that 16 is not required for the base xtensa architecture with or without standard FP coprocessors. But 16 or even higher alignment may be required in the presence of TIE extensions, so it should really not be hardcoded but taken from the xtensa core configuration.

@jcmvbkbc Thanks for clarifying.

stephanosio, 2021-09-03:
https://github.com/zephyrproject-rtos/newlib-cygwin/commit/b67ccc265089ded89e9170fcc5a8e2e43d020393#diff-f056d0212a247fc9bf5137a683fa14d9f1f80f23f76eedb68cf81710207fdb27R198
I locally built and tested newlib with the above line removed so that `MALLOC_ALIGNMENT` defaults to 8 instead of 16. As I thought, this issue is no longer present and malloc works.

On `qemu_xtensa` with patched newlib:
```
*** Booting Zephyr OS build zephyr-v2.6.0-2928-g296cecd30943  ***
_sbrk: 0x60004fd0, 48
_sbrk: 0x60005000, 0
000: 0x60004fd8
_sbrk: 0x60005000, 4096
001: 0x60004ff8
002: 0x60005018
003: 0x60005038
004: 0x60005058
...
```

I will post a Zephyr SDK patch for this.

stephanosio, 2021-09-06:
https://github.com/zephyrproject-rtos/sdk-ng/pull/394 will fix this issue.

sylvioalves, 2021-09-09:
> @sylvioalves do you know if you guys newlib version has this issue or a fix for it?

I'll check and get back soon.

gerekon, 2021-09-10:
> @sylvioalves do you know if you guys newlib version has this issue or a fix for it?
> I'll check and get back soon.

@sylvioalves @galak 
We have the same problem in our [newlib](https://github.com/espressif/newlib-esp32/blob/esp_based_on_3_0_0/newlib/libc/include/sys/config.h#L194). It seems to come with [one](https://github.com/jcmvbkbc/newlib-xtensa/blob/xtensa/newlib/libc/include/sys/config.h#L171) of @jcmvbkbc patches with Xtensa support.
As far as I know alignment for 16 bytes can be required by TIE for our new chip ESP32-S3 (`XCHAL_DATA_WIDTH=16`). Other have `XCHAL_DATA_WIDTH=4` so they can live with malloc alignment of `8`.
But `MALLOC_ALIGNMENT` does not influence ESP-IDF because it does not use newlib's allocators, so this problem has not been seen before. So we are going to remove that line at all in the next toolchain release.

stephanosio, 2021-09-10:
> As far as I know alignment for 16 bytes can be required by TIE for our new chip ESP32-S3 (`XCHAL_DATA_WIDTH=16`)

I suppose the only downside of having `MALLOC_ALIGNMENT=8` in this case is some performance impact? or does ESP32-S3's "TIE" not support unaligned accesses at all?

stephanosio, 2021-09-11:
https://github.com/zephyrproject-rtos/sdk-ng/pull/394 has been merged.

Waiting for the Zephyr SDK 0.13.1 to be released and mainlined in the CI.

gerekon, 2021-09-13:
@stephanosio 

> or does ESP32-S3's "TIE" not support unaligned accesses at all?

Not sure about this. Need to check.

cfriedt, 2021-09-14:
#38261 is a workaround

cfriedt, 2021-09-14:
@iuliana-prodan @dbaluta

stephanosio, 2021-09-14:
_As per the discussion in the bug triage meeting today:_

We will be keeping this issue open until the new SDK is released so that we can keep track of this problem, and also that we do not forget the remove the workaround added in #38261.

SDK 0.13.1 will not be released for and used by the Zephyr 2.7.0 release because it contains other unrelated fixes that can potentially break other things and we cannot afford that at this stage.

stephanosio, 2021-09-20:
_As per the Toolchain WG meeting today:_

SDK 0.13.1 will be released for and used by the Zephyr 2.7.0 release since it contains a critical ARM-related security fix.

galak, 2021-09-22:
Bumping to high as we released SDK 0.13.1 with fix, and we should make sure the PR to revert workaround gets merged.

cfriedt, 2021-09-23:
This will be merged on Friday AM EST, just so that the message on the devel mailing list has time to be circulated, and so that there are enough cycles for CI prior to cutting rc3.

PR #38733 (2021-09-24): Revert "lib: newlib: Add workaround for #38258"
```
Revert "lib: newlib: Add workaround for #38258"

The commit 9bd1483afeb18f4225ec7b0340b0d4e20efb7d01 was added as a
workaround for the Xtensa initial malloc failure bug.

This bug has been fixed in the Zephyr SDK 0.13.1 release and therefore
this workaround is no longer needed.
```

Fixes #38258 

NOTE: This PR also contains a commit to update the minimum required Zephyr SDK version to 0.13.1, which is required to support this.
closed 2021-09-24 by cfriedt