zephyrproject-rtos/zephyr #38259: subsys/shell: `[JJ` escape codes in logs after disabling colors

ycsin, 2021-09-02:
This serves as a continuation of discussion in #38094, for some reason there are still `[JJ` escape codes in logs after disabling color. What do they come from? Is there a way to disable them?

**Impact**
Annoyance, and difficulty to parse log output using a software.

**Logs and console output**
```
[JJ--- 6 messages dropped ---
[JJ[00:00:48.453,000] <inf> modem_gsm: RSSI: -67
[JJ[JJ[01:01:49.694,000] <inf> modem_gsm: RSSI: -69
```

**Environment (please complete the following information):**
 - v2.6.0-1638-g0296d12675fd

ycsin, 2021-09-02:
@jakub-uC @jfischer-no @nordic-krch

nordic-krch, 2021-09-02:
@jakub-uC isn't it line erasing done before each log message printing?
https://github.com/zephyrproject-rtos/zephyr/blob/2208844f17010c4237c21d7b84f0f5e2f634d24d/subsys/shell/shell.c#L1156

ycsin, 2021-09-03:
@nordic-auko I think you are right, I had something simple to test it:
```c
			if (z_flag_echo_get(shell)) {
				z_shell_cmd_line_erase(shell);
			}
```
and now those `[JJ`s are gone.

Do you have any suggestion on how to disable these features that are designed for a terminal?

nordic-auko, 2021-09-07:
> 
> 
> @nordic-auko I think you are right, I had something simple to test it:
> 
> ```c
> 			if (z_flag_echo_get(shell)) {
> 				z_shell_cmd_line_erase(shell);
> 			}
> ```
> 
> and now those `[JJ`s are gone.
> 
> Do you have any suggestion on how to disable these features that are designed for a terminal? I'm trying to implement a MQTT shell backend that cant use these terminal feature. #38107

@nordic-krch ? :)

jakub-uC, 2021-09-07:
> @nordic-auko I think you are right, I had something simple to test it:
>
> ```c
> 			if (z_flag_echo_get(shell)) {
> 				z_shell_cmd_line_erase(shell);
> 			}
> ```
>
> and now those `[JJ`s are gone.
>
> Do you have [...]

If you wish to disable VT100 characters you can try to use `Kconfig` option:
`CONFIG_SHELL_VT100_COMMANDS=n`

ycsin, 2021-09-07:
@jakub-uC I guess the way to do it per instance is to make it a flag?

cfriedt, 2021-09-07:
@ycsin - it sounds as though you have found a workaround, so I'm going to close this. If you would like to continue discussion here, that's fine.

closed 2021-09-07 by cfriedt

jakub-uC, 2021-09-07:
It is not a workaround because it applies to all backends at the same time. I am working on a fix that will solve this issue.