zephyrproject-rtos/zephyr #38264: Modbus over RS485 on samd21g18a re-gpios turning on 1 byte too early

krobotics, 2021-09-02:
**Describe the bug**
When using Modbus over RS485 the driver enable/receive enable pins(s) are being triggers before the transmission has completed. This is causing the last byte to not be transmitted by the rs485 transceiver. 

What have you tried to diagnose or workaround this issue?
Yes the current code is using the DRE interrupt to determine if the transmission has completed, however it should be using the TXC interrupt that actually indicates that the transmission has completed:

```
**26.6.2.5 Data Transmission**
When the entire data frame including stop bit(s) has been transmitted and no new data was written to
DATA, the Transmit Complete interrupt flag in the Interrupt Flag Status and Clear register (INTFLAG.TXC)
will be set, and the optional interrupt will be generated.
```

**To Reproduce**
Steps to reproduce the behavior:
1. create a simple task as in the code below
```
static int client_iface;

const static struct modbus_iface_param client_param = {
	.mode = MODBUS_MODE_RTU,
	.rx_timeout = 500,
	.serial = {
		.baud = 115200,
		.parity = UART_CFG_PARITY_NONE,
//    .stopbits = 1, // workaround as current modbus code forces 2 stopbits on no parity
	},
};

static int init_modbus_client(void)
{
	const char iface_name[] = {DT_PROP(DT_INST(0, zephyr_modbus_serial), label)};

	client_iface = modbus_iface_get_by_name(iface_name);

	return modbus_init_client(client_iface, client_param);
}

void modbus(void)
{
  uint16_t holding_reg[11] = {0};
  if (init_modbus_client())
  {
    printk("Modbus RTU client initialization failed\n");
    return;
  }

  while(1)
  {
    auto err = modbus_read_holding_regs(client_iface, '`', 0, holding_reg,10);
    k_msleep(5000);
  }
}

K_THREAD_DEFINE(modbus_id, 512, modbus, NULL, NULL, NULL, 7, 0, 0);

```
**Expected behavior**
The expected behavior is to transmit all the bytes in the packet before toggling the re/de pin not bytes-1.

**Impact**
showstopper - cannot use the Modbus library

**Logs and console output**
I have attached a screenshot from my logic probe.

**Environment (please complete the following information):**
 - Toolchain Zephyr SDK
 - Version used - latest from git

**Additional context**
Will attempt to fix and then generate a pull request unless there is already a fix
[image]
.

jfischer-no, 2021-09-03:
@rocky134 FYI

rocky134, 2021-09-03:
> Yes the current code is using the DRE interrupt to determine if the transmission has completed, however it should be using the TXC interrupt that actually indicates that the transmission has completed:

Yes good catch, that looks to be the case that DRE is generated when the DATA register is empty (and data can be then written to it) whereas TXC is generated when the data has actually been shifted out. 

However it's not a simple replace DRE to TXC that fixes the problem, from the quick testing I did, TXC is only generated *after* a byte has been written to DATA and shifted out but the byte cannot be written because at the first time TX is enabled, there is no data written to DATA and thus the TXC ISR has not fired yet.

krobotics, 2021-09-03:
Well, you were correct about a simple fix - it definitely was not.

I did a port Arduino Serial Library to RS485 my FreeRTOS port for SAMD21/51. When I started using Zephyr I immediately saw what the issue was. I have also had the same issue on the PSoC 5, kfm64 and other Cortex based procs (yes I do a lot of RS485) in the past using stock serial libraries, especially if they have FIFO's. 

I fixed the Zephyr system and will do a pull request shortly. It was not just an issue with the TXC but also with the IRQ handler and the way it checked to see if DRE was ready. If TXC is enabled the INTFLG bit for DRE might remain set after DRE is disabled, 
 the loop does not check to see if the interrupt is still enabled when the flag is set. (I added a check to the ..._tx_ready). 

I found something strange going on in the handler while the loop However it's not a simple replace DRE to TXC that fixes the problem, from the quick testing I did, TXC is only generated after a byte has been written to DATA and shifted out but the byte cannot be written because at the first time TX is enabled, there is no data written to DATA and thus the TXC ISR has not fired yet.

I could not get the TXC handler invoked while using the loop - I removed the loop and that resolved the issue (maybe something I overlooked). I moved the re/de to the _txc_off(...) routine and defined out the one in _tx_of(...)
```
#ifndef CONFIG_MODBUS_SERIAL_USE_TXC
	while (uart_irq_update(cfg->dev) && uart_irq_is_pending(cfg->dev)) {
#endif
		if (uart_irq_rx_ready(cfg->dev)) {
			cb_handler_rx(ctx);
		}

		if (uart_irq_tx_ready(cfg->dev)) {
			cb_handler_tx(ctx);
		}

#ifdef CONFIG_MODBUS_SERIAL_USE_TXC
		if (uart_irq_txc_ready(cfg->dev)) {
			cb_handler_txc(ctx);
		}
#else
	}
#endif	
```

I need to separate out my stopbit fix and do that as a separate pull request (current library forces 2 stop bits on no parity, does not allow your to use 1).

As you can see below the issue seems to be resolved.

[image]

rocky134, 2021-09-03:
glad you were able to fix the issue but I don't think the modbus driver should be the thing to change.
The issue is with the sam0 uart driver improper irq handling so any fixes should be contained to that file only.
Since if you weren't using a sam0 platform modbus would work as intended.

krobotics, 2021-09-03:
Yes and no, the de/re pin is set on the DRE interrupt so in order to prevent the pin from toggling at the wrong time so the modbus code has to change. If you are using another processor with a fifo on its uart, DRE will empty way before the pin should be toggled. The only proper way to fix this would be to change the modbus subsystem to monitor the correct completion interrupt. However as this subsystem might already be used by others, a fix cannot be made without impacting existing users. 

Ideally one would hope that a change to the sam0 platform would work but you could not do that without creating other issues in code that uses sam0. The big issue really is using RS485 with de/re toggle (not necessarily modbus), for any full duplex serial implementation there would be no issue. 

There is an ".irq_tx_complete" field in 'struct uart_driver_api' but it is not used by modbus and not implemented in the sam0 code.

So this cannot be a fix in only one area - that is why I decided to use a CONFIG option, that way it would not break anything and if someone else had an issue on another processor they could easily implement the lower layers.

Thoughts ?

rocky134, 2021-09-03:
yes, after looking into this issue a bit more over the past hour it seems like just patching uart_sam0 is not enough.
Like you said, modbus needs to wait until TX is completed.

>There is an ".irq_tx_complete" field in 'struct uart_driver_api' but it is not used by modbus and not implemented in the sam0 code.

That's a good method to use actually, rather than adding a new method `uart_irq_txc_ready` to the uart api it would be better to just implement uart_irq_tx_complete for the sam0 and handle it inside of modbus' cb_handler_tx.

>So this cannot be a fix in only one area - that is why I decided to use a CONFIG option, that way it would not break anything and if someone else had an issue on another processor they could easily implement the lower layers.

I wonder though, if a config option even be needed. There are numerious changes that change functionality and are not backwards compatible. Developers who require the old functionality should stay with that release version or update their code.

Especially since half duplex modbus expects tx completed before de-asserting de it sounds like other platforms who have modbus working were only working because of a hardware specific implementation of uart that would generate an interrupt only after the bits were done clocking out.
Modbus at a higher level should always check if tx is completed regardless of the platform's hardware uart implementation.

Thoughts on cb4ce6219dae3a2b607a65fd4203fcb0fbb1d9e8 and 1de56b898aa5dd4ee60b51fc177c7d0b92f7ba8e and waveform output.
[image]

krobotics, 2021-09-03:
Fair comment - I checked in a change to my fork using the CONFIG option and TXC. 
I think I know why uart_irq_tx_complete was not implemented for sam0. There is no status bit that can be queried to determine if the transmission is complete.

```
/**
 * @brief Start processing interrupts in ISR.
 *
 * This function should be called the first thing in the ISR. Calling
 * uart_irq_rx_ready(), uart_irq_tx_ready(), uart_irq_tx_complete()
 * allowed only after this.
 *
 * The purpose of this function is:
 *
 * * For devices with auto-acknowledge of interrupt status on register
 *   read to cache the value of this register (rx_ready, etc. then use
 *   this case).
 * * For devices with explicit acknowledgement of interrupts, to ack
 *   any pending interrupts and likewise to cache the original value.
 * * For devices with implicit acknowledgement, this function will be
 *   empty. But the ISR must perform the actions needs to ack the
 *   interrupts (usually, call uart_fifo_read() on rx_ready, and
 *   uart_fifo_fill() on tx_ready).
 *
 * @param dev UART device structure.
 *
 * @retval 1 Always.
 */
```

Due to the face that there is no enable/disable for this parameter/function one cannot just add TXC and hope for the best. One solution would then require abstracting the TXC, enabling whenever DRE is enabled. The TXC intflag is cleared every time a data byte is written so it could probably be polled. This state can be returned by uart_irq_tx_complete. Any modbus read command has a timeout for response - this timeout can start after the completion flag is detected. For that matter any modbus command can have the polling added (not very elegant).

Another, even less elegant option, would be to have a configurable timeout for toggling the pins, after DRE completion. This can correspond to the uart buffer size and baud rate (I had to do this on a netburner). This cannot be done in an interrupt handler, so would also be tied to a timeout after sending a modbus command.

Considering all this it seems that my simple CONFIG solution might just have the least impact.

rocky134, 2021-09-03:
> I think I know why uart_irq_tx_complete was not implemented for sam0. There is no status bit that can be queried to determine if the transmission is complete.

The status bit is TXC itself, from the datasheet:
[image]

> Due to the fact that there is no enable/disable for this parameter/function one cannot just add TXC and hope for the best.

It wasn't just added. it is added transparently in a minimal way that does not affect existing higher level drivers utilization of the serial driver. 
Interrupt driven uart requires a call to `uart_irq_is_pending` and `uart_irq_update` before handling any interrupts. Therefore because DRE is always 1 when DATA is empty, calls to `uart_irq_is_pending` will always return true as long as data is being transmitted/has already been transmitted. Even more so since DRE triggers before TXC, when DATA is empty the isr will trigger and higher level drivers will fill DATA with the next byte to transmit as expected.
Adding TXC to that list of enabled TX interrupts does not change existing behavior as a call to `uart_irq_tx_disable` only disables INTENSET, the byte being already in the shift register being shifted out will continue to do so as you have already seen.

The behavior of DRE is also why PR #38091 was created, since modbus would not receive data without it(`uart_irq_tx_ready` always returns true if DATA is empty regardless of INTENSET status). Then because enabling TXC would not impact the expected results of `uart_irq_tx_ready` existing drivers which use sam0 uart would simply continue to operate normally, disabling tx when their buffers are empty after the last call to `uart_fifo_fill`.

>One solution would then require abstracting the TXC, enabling whenever DRE is enabled. The TXC intflag is cleared every time a data byte is written so it could probably be polled every time DRE is triggered, but this can be inefficient as DRE is continuously triggered when empty. Maybe this state can be returned by uart_irq_tx_complete, but its an interrupt only call.

Yes TXC is cleared after every write to DATA but it does not matter because higher level drivers should be checking `uart_irq_tx_ready` before calls to `uart_fifo_fill` per this line
```
/**
 * @brief Check if UART TX buffer can accept a new char
 *
 * @details Check if UART TX buffer can accept at least one character
 * for transmission (i.e. uart_fifo_fill() will succeed and return
 * non-zero). This function must be called in a UART interrupt
 * handler, or its result is undefined. Before calling this function
 * in the interrupt handler, uart_irq_update() must be called once per
 * the handler invocation.
 *
 * @param dev UART device structure.
 *
 * @retval 1 If at least one char can be written to UART.
 * @retval 0 Otherwise.
 */
static inline int uart_irq_tx_ready(const struct device *dev)
```
and because uart_sam0_irq_tx_ready has not been changed, it functions as expected.

Then per the comment above `uart_irq_tx_complete`
```
/**
 * @brief Check if UART TX block finished transmission
 *
 * @details Check if any outgoing data buffered in UART TX block was
 * fully transmitted and TX block is idle. When this condition is
 * true, UART device (or whole system) can be power off. Note that
 * this function is *not* useful to check if UART TX can accept more
 * data, use uart_irq_tx_ready() for that. This function must be called
 * in a UART interrupt handler, or its result is undefined. Before
 * calling this function in the interrupt handler, uart_irq_update()
 * must be called once per the handler invocation.
 *
 * @param dev UART device structure.
 *
 * @retval 1 If nothing remains to be transmitted.
 * @retval 0 Otherwise.
 * @retval -ENOTSUP if this function is not supported
 */
static inline int uart_irq_tx_complete(const struct device *dev)
```
a call to uart_irq_tx_complete and thus uart_sam0_irq_tx_complete will return true when tx is idle because TXC is set, which is only set if only if the last byte written to DATA was fully transmitted and no additional data has been written to DATA.
Writing a byte to DATA before then will clear TXC as expected but because a higher level driver is the one who called `uart_fifo_fill` it will know (or should know at least) that `uart_irq_tx_complete` will return false and continue to do so until the data has been completely transmitted.
At which point `uart_irq_tx_complete` will return true.

krobotics, 2021-09-07:
I was thinking along these lines but overlooked the simplicity. I even pointed out enabling TXC with DRE. I tested your changes and they worked for me I am going to close out my issue as you have a better solution.

closed 2021-09-07 by krobotics