zephyrproject-rtos/zephyr #38273: Support UART4 on STM32F303Xe

holgerschurig, 2021-09-03:
**Is your enhancement proposal related to a problem? Please describe.**
The STM32F303Xe soc has a fourth UART, but the provided device tree doesn't support it.

**Describe the solution you'd like**
Currently, I put the following into my device overlay. This makes this UART work for me (tested it with 115200 baud and an external FTDI 3V3-serial-to-USB dongle):

```
/ {
    model = "REDACTED";
    compatible = "REDACTED";

    chosen {
        // USART2: main CPU
        // USART3: display
        // USART4: Debug
        zephyr,console = &uart4;
        zephyr,shell-uart = &uart4;
        zephyr,sram = &sram0;
        zephyr,flash = &flash0;
        uc2uc = &usart3;
    };

    soc {
        // UART 3: Debug port
        uart4: serial@40004c00 {
            compatible ="st,stm32-uart";
            reg = <0x40004c00 0x400>;
            clocks = <&rcc STM32_CLOCK_BUS_APB1 0x00080000>;
            pinctrl-0 = <&uart4_tx_pc10 &uart4_rx_pc11>;
            current-speed = <115200>;
            interrupts = <52 0>;
            status = "okay";
            label = "UART_4";
        };
    };
};
```

**Additional context**
It would be nice if the /soc/uart4 section could be added upstream into Zephyr., then of course with a default status="disabled" :-)

FRASTM, 2021-09-03:
Note that the DTS for the stm32f303 should not be mixed with the pin assignment.

- `pinctrl-0 = <&uart4_tx_pc10 &uart4_rx_pc11>;`     is valid for the smt32f3_disco board

manoj153, 2021-09-03:
> Note that the DTS for the stm32f303 should not be mixed with the pin assignment.
> 
> * `pinctrl-0 = <&uart4_tx_pc10 &uart4_rx_pc11>;`     is valid for the smt32f3_disco board

Need help to create the PR?