zephyrproject-rtos/zephyr #38277: soc: stm32h7: Fails to boot with LDO power supply, if soc has SMPS support.

alexanderendian, 2021-09-03:
**Describe the bug**
In `soc_m7.c`, SMPS is enabled if the soc has SMPS-support, without regard for the actual board design. For boards designed to use LDO, enabling SMPS causes the microcontroller to reset, presumably due to losing power.

**To Reproduce**
Run Zephyr on a STM32H7 design with LDO supply using a part which supports SMPS.

**Expected behavior**
The choice between SMPS and LDO should be configurable on a per board basis.

**Impact**
No impact for the moment as it can be worked around by commenting out the SMPS path in `soc_m7.c`.

**Related issues**
https://github.com/zephyrproject-rtos/zephyr/issues/34732 - also regarding power config of STM32H7.

erwango, 2021-09-03:
@ABOSTM Feel free to close if duplicate of #34732

ABOSTM, 2021-09-07:
Duplicate of #34732

closed 2021-09-07 by ABOSTM