zephyrproject-rtos/zephyr #38279: bluetooth: controller: assert (LL_ASSERT(lll->handle != 0xFFFF);) hits

jsiverskog, 2021-09-03:
**Describe the bug**
when i send large amounts of ble data from a central to a peripheral an assert hits (`LL_ASSERT(lll->handle != 0xFFFF);` from subsys/bluetooth/controller/ll_sw/nordic/lll/lll_conn.c). both sides are running zephyr's 2.6-branch.

there is a lot of sensitive code involved, so the triggering project will be sent in private message.

**To Reproduce**
Steps to reproduce the behavior:
- use two nrf52840 dks (pca10056)
- all instructions are located in the triggering project zip.

**Expected behavior**
data transferred successfully (or error returned - but no crash).

**Impact**
showstopper.

**Logs and console output**
```
[00:15:47.601,654] <err> os: r0/a1:  0x00000003  r1/a2:  0x20012214  r2/a3:  0x200121cb
[00:15:47.601,684] <err> os: r3/a4:  0x000461eb r12/ip:  0xa0000000 r14/lr:  0x00040aa3
[00:15:47.601,684] <err> os:  xpsr:  0x41000011
[00:15:47.601,684] <err> os: Faulting instruction address (r15/pc): 0x00040aae
[00:15:47.601,684] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[00:15:47.601,684] <err> os: Fault during interrupt handling

[00:15:47.601,715] <err> os: Current thread: 0x20003080 (unknown)
[00:15:47.925,231] <err> os: Halting system
```

[zephyr_ble_assert.zip](https://github.com/zephyrproject-rtos/zephyr/files/7104728/zephyr_ble_assert.zip)

**Environment (please complete the following information):**
 - OS: linux
 - Toolchain: arm-gcc toolchain (gcc-arm-none-eabi-10.3-2021.07)
 - Commit SHA: ea55ebfa74ae507b0d43e5dcd3b9747601bf5ae3

cvinayak, 2021-09-07:
Reproducing the issue and analysis lead to LL_ASSERT(!radio_is_ready()) in lll_conn.c line 324 in the v2.6-branch. 

The causes for these can be higher ISR Latencies, which can be due to use of `irq_lock` outside the Bluetooth Controller Subsystem.

Selected (removed some lines) chat communications:
> 10:03 AM
vich yes, addr2line  mentions discriminator 1,  which would means the address can be reached from one other path
10:03 AM
the other path is LL_ASSERT(!radio_is_ready())
10:08 AM
vich ok... I put an explicit BT_ERR there, and it is confirmed it is the radio_is_ready() assertion

> 10:16 AM
jsiverskog when you mention isr latency i wonder if calls to irq_lock may be the issue
10:18 AM
vich is irq_lock  used by the application?
10:18 AM
jsiverskog yes, in a few places
10:19 AM
vich ok.... if this is necessary then Controller will need to use Zero Latency Interrupts in Zephyr
10:20 AM
jsiverskog is it enough to test with CONFIG_ZERO_LATENCY_IRQS=y to see if that may be the cause?
10:26 AM
jsiverskog will try without / reduced irq_lock:s
10:26 AM
vich Add, CONFIG_BT_CTLR_ADVANCED_FEATURES=y, and CONFIG_BT_CTLR_ZLI=y
10:57 AM
jsiverskog i will try some, will get back when i have some results

jsiverskog, 2021-09-07:
thanks for looking into this.

reducing/removing `irq_lock` seems to solve the issue.

closed 2021-09-07 by jsiverskog

cvinayak, 2022-03-03:
FYI, #43396