zephyrproject-rtos/zephyr #38284: DHCP Server

kdhaka, 2021-09-03:
**Is your feature request related to a problem? Please describe.**
When devices connect to my wifi hotspot (used for configuration purpose only), it should provide address to the computer

**Describe the solution you'd like**
A DHCP server that rely to DHCP client.

**Describe alternatives you've considered**
- BOOTP <https://en.wikipedia.org/wiki/Bootstrap_Protocol>
- Automatic configuration <https://en.wikipedia.org/wiki/Address_autoconfiguration>

**Additional context**
Show up as a Wifi Hotspot and accept configuration.

enthunilu, 2023-02-20:
Hi all, 

Any update or plan on DHCP server support ?

Thanks!!!

rlubos, 2024-01-23:
Implemented in https://github.com/zephyrproject-rtos/zephyr/pull/67713

closed 2024-01-23 by rlubos