zephyrproject-rtos/zephyr #38303: The current BubbleSim tests build system based on bash scripts hides warnings

alxelax, 2021-09-03:
**Describe the bug**
The current BubbleSim tests build system hides compile warnings and ignores them. The PR checks miss them (test step is green even warnings present) and allow to merge PRs with warnings(potential bugs).

It is possible to observe already committed in main warnings just adding syntax errors in some of the tests. Then build system will show all existing warnings together with the syntax error report.

**To Reproduce**
Steps to reproduce the behavior:
1. Add any C syntax issue in BubbleSim test 
2. Buld test using bash compile.sh
3. See existing warnings

**Expected behavior**
The PR checks should catch warnings and prevent merging.

**Impact**
BubbleSim test quality

**Logs and console output**
\`\`\`
Building /home/alkh/bsim/bin/bs_nrf52_bsim_tests_bluetooth_bsim_bt_bsim_test_mesh_prj_conf
[1/226] Preparing syscall dependency handling

[2/226] Generating misc/generated/syscalls_subdirs.trigger
[3/226] Generating misc/generated/syscalls.json, misc/generated/struct_tags.json
[4/226] Generating include/generated/syscall_dispatch.c, include/generated/syscall_list.h
[5/226] Generating include/generated/driver-validation.h
[6/226] Generating include/generated/kobj-types-enum.h, include/generated/otype-to-str.h, include/generated/otype-to-size.h
[7/226] Building C object zephyr/CMakeFiles/offsets.dir/arch/posix/core/offsets/offsets.c.obj
[8/226] Generating include/generated/offsets.h
[9/226] Building C object CMakeFiles/app.dir/src/main.c.obj
[10/226] Building C object CMakeFiles/app.dir/src/mesh_test.c.obj
[11/226] Building C object CMakeFiles/app.dir/src/oob_server.c.obj
[12/226] Building C object CMakeFiles/app.dir/src/test_transport.c.obj
[13/226] Building C object CMakeFiles/app.dir/src/oob_client.c.obj
[14/226] Building C object CMakeFiles/app.dir/src/test_friendship.c.obj
[15/226] Building C object CMakeFiles/app.dir/src/test_provision.c.obj
[16/226] Building C object CMakeFiles/app.dir/src/test_beacon.c.obj
[17/226] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/cbprintf.c.obj
[18/226] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/cbprintf_packaged.c.obj
[19/226] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc32c_sw.c.obj
[20/226] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc32_sw.c.obj
[21/226] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc16_sw.c.obj
[22/226] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc8_sw.c.obj
[23/226] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc7_sw.c.obj
[24/226] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/dec.c.obj
[25/226] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/fdtable.c.obj
[26/226] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/hex.c.obj
[27/226] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/notify.c.obj
[28/226] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/printk.c.obj
[29/226] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/onoff.c.obj
[30/226] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/rb.c.obj
[31/226] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/sem.c.obj
[32/226] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/thread_entry.c.obj
[33/226] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/timeutil.c.obj
[34/226] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/heap.c.obj
[35/226] Building C object CMakeFiles/app.dir/src/test_scanner.c.obj
FAILED: CMakeFiles/app.dir/src/test_scanner.c.obj 
ccache /usr/bin/gcc -DBUILD_VERSION=zephyr-v2.6.0-2888-g12d0e5b73c5a -DKERNEL -DNRF52832_XXAA -D_POSIX_C_SOURCE=200809 -D_XOPEN_SOURCE=600 -D_XOPEN_SOURCE_EXTENDED -D__ZEPHYR__=1 -I/home/alkh/ncs/modules/bsim_hw_models/nrf_hw_models/src/nrfx/mdk_replacements -I/home/alkh/ncs/zephyr/include -I/home/alkh/ncs/zephyr/bsim_bt_out/tests/bluetooth/bsim_bt/bsim_test_mesh/prj.conf/zephyr/include/generated -I/home/alkh/ncs/zephyr/soc/posix/inf_clock -I/home/alkh/ncs/zephyr/boards/posix/nrf52_bsim -I/home/alkh/ncs/zephyr/subsys/bluetooth -I/home/alkh/ncs/modules/hal/nordic/nrfx -I/home/alkh/ncs/modules/hal/nordic/nrfx/drivers/include -I/home/alkh/ncs/modules/hal/nordic/nrfx/mdk -I/home/alkh/ncs/zephyr/modules/hal_nordic/nrfx/. -I/home/alkh/ncs/modules/bsim_hw_models/nrf_hw_models/src/HW_models -I/home/alkh/ncs/modules/bsim_hw_models/nrf_hw_models/src/nrfx/nrfx_replacements -I/home/alkh/ncs/modules/crypto/tinycrypt/lib/include -I/home/alkh/bsim/components/libUtilv1/src -I/home/alkh/bsim/components/libPhyComv1/src -O0 -imacros /home/alkh/ncs/zephyr/bsim_bt_out/tests/bluetooth/bsim_bt/bsim_test_mesh/prj.conf/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -imacros /home/alkh/ncs/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-address-of-packed-member -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/alkh/ncs/zephyr/tests/bluetooth/bsim_bt/bsim_test_mesh=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/alkh/ncs/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/alkh/ncs=WEST_TOPDIR -ffunction-sections -fdata-sections -fprofile-arcs -ftest-coverage -fno-inline -m32 -include /home/alkh/ncs/zephyr/arch/posix/include/posix_cheats.h -fno-freestanding -fshort-enums -std=c11 -MD -MT CMakeFiles/app.dir/src/test_scanner.c.obj -MF CMakeFiles/app.dir/src/test_scanner.c.obj.d -o CMakeFiles/app.dir/src/test_scanner.c.obj -c /home/alkh/ncs/zephyr/tests/bluetooth/bsim_bt/bsim_test_mesh/src/test_scanner.c
/home/alkh/ncs/zephyr/tests/bluetooth/bsim_bt/bsim_test_mesh/src/test_scanner.c:13:3: error: expected ‘;’ before ‘struct’
   13 | gg    **<------  artificial C error syntax  to make warnings visible**
      |   ^
      |   ;
/home/alkh/ncs/zephyr/tests/bluetooth/bsim_bt/bsim_test_mesh/src/test_scanner.c: In function ‘test_tx_invalid_ad_type’:
/home/alkh/ncs/zephyr/tests/bluetooth/bsim_bt/bsim_test_mesh/src/test_scanner.c:126:45: warning: passing argument 2 of ‘test_tx_send_ad_type_msg’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
  126 |  test_tx_send_ad_type_msg(BT_DATA_BIG_INFO, valid_message,
      |                                             ^~~~~~~~~~~~~
/home/alkh/ncs/zephyr/tests/bluetooth/bsim_bt/bsim_test_mesh/src/test_scanner.c:72:61: note: expected ‘uint8_t *’ {aka ‘unsigned char *’} but argument is of type ‘const uint8_t *’ {aka ‘const unsigned char *’}
   72 | static void test_tx_send_ad_type_msg(uint8_t type, uint8_t *data, uint8_t len)
      |                                                    ~~~~~~~~~^~~~
/home/alkh/ncs/zephyr/tests/bluetooth/bsim_bt/bsim_test_mesh/src/test_scanner.c:133:49: warning: passing argument 2 of ‘test_tx_send_ad_type_msg’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
[... 54 lines cut ...]
[21/225] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc16_sw.c.obj
[22/225] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc8_sw.c.obj
[23/225] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc7_sw.c.obj
[24/225] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/dec.c.obj
[25/225] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/fdtable.c.obj
[26/225] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/hex.c.obj
[27/225] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/notify.c.obj
[28/225] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/printk.c.obj
[29/225] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/onoff.c.obj
[30/225] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/rb.c.obj
[31/225] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/sem.c.obj
[32/225] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/thread_entry.c.obj
[33/225] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/timeutil.c.obj
[34/225] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/heap.c.obj
[35/225] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/heap-validate.c.obj
[36/225] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/bitarray.c.obj
[37/225] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/cbprintf_complete.c.obj
[38/225] Building C object CMakeFiles/app.dir/src/test_scanner.c.obj
FAILED: CMakeFiles/app.dir/src/test_scanner.c.obj 
[... cut ...] rc/test_scanner.c.obj.d -o CMakeFiles/app.dir/src/test_scanner.c.obj -c /home/alkh/ncs/zephyr/tests/bluetooth/bsim_bt/bsim_test_mesh/src/test_scanner.c
\`\`\`

**Additional context**
The suggestion how to fix it provided by @aescolar is possible to find here:
The issue has been highlighted here: https://github.com/zephyrproject-rtos/zephyr/pull/38205/files?file-filters%5B%5D=.c&file-filters%5B%5D=.sh&file-filters%5B%5D=.txt#diff-e56b290605f51cc3f6e2c017cd66a56827db908303cdb04b4ee0f426a117b6ecR38

alxelax, 2021-12-22:
The issue came back after merging this PR https://github.com/zephyrproject-rtos/zephyr/pull/40073.

alxelax, 2021-12-22:
It hides warnings in nrf52_bsim platform about potential hardfaults:
```
modules/nrf_hw_models/CMakeFiles/..__modules__bsim_hw_models__nrf_hw_models.dir/src/HW_models/NRF_RADIO.c.obj 
ccache /usr/bin/gcc -DBUILD_VERSION=v2.7.99-2659-ge1b845cc8875 -DKERNEL -DNO_POSIX_CHEATS -DNRF52832_XXAA -D_POSIX_C_SOURCE=200809 -D_XOPEN_SOURCE=600 -D_XOPEN_SOURCE_EXTENDED -D__ZEPHYR__=1 -I/home/alkh/bsim/components/libUtilv1/src -I/home/alkh/bsim/components/libPhyComv1/src -I/home/alkh/bsim/components/ext_2G4_libPhyComv1/src -I/home/alkh/bsim/components/libRandv2/src -I/home/alkh/ncs/modules/bsim_hw_models/nrf_hw_models/src/HW_models -I/home/alkh/ncs/modules/bsim_hw_models/nrf_hw_models/src/nrfx/mdk_replacements -I/home/alkh/ncs/zephyr/include -I/home/alkh/ncs/zephyr/bsim_bt_out/tests/bluetooth/bsim_bt/bsim_test_mesh/prj.conf/zephyr/include/generated -I/home/alkh/ncs/zephyr/soc/posix/inf_clock -I/home/alkh/ncs/zephyr/boards/posix/nrf52_bsim -I/home/alkh/ncs/zephyr/subsys/bluetooth -I/home/alkh/ncs/modules/hal/nordic/nrfx -I/home/alkh/ncs/modules/hal/nordic/nrfx/drivers/include -I/home/alkh/ncs/modules/hal/nordic/nrfx/mdk -I/home/alkh/ncs/zephyr/modules/hal_nordic/nrfx/. -I/home/alkh/ncs/modules/bsim_hw_models/nrf_hw_models/src/nrfx/nrfx_replacements -I/home/alkh/ncs/modules/crypto/tinycrypt/lib/include -Werror -O0 -imacros /home/alkh/ncs/zephyr/bsim_bt_out/tests/bluetooth/bsim_bt/bsim_test_mesh/prj.conf/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/alkh/ncs/zephyr/tests/bluetooth/bsim_bt/bsim_test_mesh=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/alkh/ncs/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/alkh/ncs=WEST_TOPDIR -ffunction-sections -fdata-sections -fprofile-arcs -ftest-coverage -fno-inline -m32 -include /home/alkh/ncs/zephyr/arch/posix/include/posix_cheats.h -fno-freestanding -fshort-enums -std=c11 -MD -MT modules/nrf_hw_models/CMakeFiles/..__modules__bsim_hw_models__nrf_hw_models.dir/src/HW_models/NRF_RADIO.c.obj -MF modules/nrf_hw_models/CMakeFiles/..__modules__bsim_hw_models__nrf_hw_models.dir/src/HW_models/NRF_RADIO.c.obj.d -o modules/nrf_hw_models/CMakeFiles/..__modules__bsim_hw_models__nrf_hw_models.dir/src/HW_models/NRF_RADIO.c.obj -c /home/alkh/ncs/modules/bsim_hw_models/nrf_hw_models/src/HW_models/NRF_RADIO.c
/home/alkh/ncs/modules/bsim_hw_models/nrf_hw_models/src/HW_models/NRF_RADIO.c: In function ‘start_Tx’:
/home/alkh/ncs/modules/bsim_hw_models/nrf_hw_models/src/HW_models/NRF_RADIO.c:856:32: error: taking address of packed member of ‘struct <anonymous>’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
  856 |   p2G4_freq_from_d(freq_off, 1,&ongoing_tx.radio_params.center_freq);
      |                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/alkh/ncs/modules/bsim_hw_models/nrf_hw_models/src/HW_models/NRF_RADIO.c: In function ‘start_Rx’:
/home/alkh/ncs/modules/bsim_hw_models/nrf_hw_models/src/HW_models/NRF_RADIO.c:1011:32: error: taking address of packed member of ‘struct <anonymous>’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
 1011 |   p2G4_freq_from_d(freq_off, 1,&ongoing_rx.radio_params.center_freq);
      |                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
```

just buld bsim tests with disabled warnings on main e1b845cc8875a0f69f2e0a1e7670327a78a1cbd3

carlescufi, 2021-12-22:
These packed member warnings were only recently enabled in main. Maybe that’s why we see them here now.

closed 2022-01-03 by aescolar