zephyrproject-rtos/zephyr #38305: Update to LVGL v8

kisvegabor, 2021-09-03:
**Is your enhancement proposal related to a problem? Please describe.**
Zephyr still uses LVGL v7 but v8 is alreardy out with many new features. 
See the change log: https://docs.lvgl.io/master/CHANGELOG.html
See the examples: https://docs.lvgl.io/master/examples.html

**Describe the solution you'd like**
The Zephyr specific parts of LVGL hasn't changed much so I hope it would be straightforward  to update to the new version. A few things to highlight:
- the [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig) file of LVGL was rewritten (mainly reorganization)  
- the [examples](https://github.com/lvgl/lvgl/tree/master/examples) are now part of LVGL
- v8 is not backward compatible with v7 (mainly the style and event API has changed)

**Describe alternatives you've considered**
I wonder if there is a good way (or do you already have a practice for it) to keep LVGL automatically updated in Zephyr on releases. 

**Additional context**
When LVGL was integrated into Zephyr we were working with @vanwinkeljan so cc him here. 
Of course, I'm very open to help the integration from LVGL's side.

vanwinkeljan, 2021-09-04:
Currently I have very little time to look into this but first thing that needs to be done, in my opinion, is to use the upstream LVGL Kconfig instead of Zephyr version of the LVGL Kconfigs.

There was a RFC PR https://github.com/zephyrproject-rtos/zephyr/pull/30461 that toke in the new, at that time, LVGL config. Not sure if PR can still serve as a base to implement a new PR?

kisvegabor, 2021-09-05:
@vanwinkeljan  Thank you for the link to the PR. Using the upstream Kconfig seems reasonable on the long term. 

I assume changing `CONFIG_LVGL` to `CONFIG_LV` would break the UI part of the existing projects but updating to lvgl v8 would break the lvgl v7 projects anyway. So it'd be great to do the 2 updates in one step to break only once.

skannan89, 2021-11-07:
Hello. Any updates on this PR? Would like to see lvgl 8 in Zephyr.

kisvegabor, 2021-11-08:
When I opened this issue I needed LVGL v8 with Zephyr in an actual project, but we have solved it in a hacky way. Just update `lvgl` added minor refactoring in `lvgl.c` and `lvgl_display.c`. For the sake of simplicity we have used `lv_conf.h` instead of Kconfig. 

The update was smooth except that LVGL v8 requires a custom `realloc` too and I found only `malloc` and `free` in user or kernel space (I don't remember exactly). So finally we have used LVGL's internal memory manager.

Regardless to our specific issue is solved, I think it'd be important to officially update LVGL in Zephyr. I can't comment on Zephyr/Kconfig part I'm definitely open to helping with the changes in LVGL.

brgl, 2021-11-29:
I would like to take this over and work on upgrading lvgl to v8 in zephyr. If possible, I would like to discuss the approach during the dev-review.

carlocaione, 2021-11-29:
assigned this to @brgl and marked ready to be discussed during the next `dev-review`

galak, 2021-12-02:
Dev-review: (Dec 2, 2021):
* General feeling is its fine to upgrade to v8 and drop v7.
* Need discussion in API meeting to discuss how we handle API changes with regards to 3rd party libs (do we use same process as we do for breaking Zephyr API changes)
* suggestion that an email to zephyr-devel & zephyr-user that there is plan to move to v8 to allow anyone to comment if they have major issues with that.

kisvegabor, 2021-12-03:
Glad to hear that the proposal had a warm welcome! :slightly_smiling_face: 

> (do we use same process as we do for breaking Zephyr API changes)

Can you link documentation that shows what is the current method of Zephyr for handling breaking changes?

carlocaione, 2021-12-03:
> Can you link documentation that shows what is the current method of Zephyr for handling breaking changes?

https://docs.zephyrproject.org/latest/reference/api/api_lifecycle.html#introducing-incompatible-changes but @brgl is already taking care of that AFAIK

brgl, 2021-12-20:
PRs just dropped:

https://github.com/zephyrproject-rtos/zephyr/pull/41336
https://github.com/zephyrproject-rtos/lvgl/pull/15

PR #41336 (2022-02-24): lvgl: upgrade to v8.1.0
This PR contains modifications to the LVGL glue code in the kernel needed in order to support the v8 major release of LVGL.

The bulk of the changes are adjustments needed to use the LVGL interfaces that broke compatiblity with v7. There are some minor tweaks and fixes here and there too.

One important change is removing the support for using k_heap by lvgl. This was suggested to me by @npitre as implementing `k_realloc()` would be tricky and not worth the hassle as lvgl has no business using k_heap anyway. We're replacing this by allowing to use a `realloc()` implemen [... cut]
closed 2022-02-24 by carlescufi