zephyrproject-rtos/zephyr #38309: ARC context switch to interrupted thread busted with CONFIG_ARC_FIRQ=y and CONFIG_NUM_IRQ_PRIO_LEVELS=1

nbowlersnps, 2021-09-03:
**Describe the bug**
The ARC can be configured with support for fast IRQs and only a single priority level.  In this configuration, fast IRQs is the only interrupt type supported by the hardware.

However, the `_switch_return_from_firq` implementation is expecting the lowest priority interrupt to perform a "regular" interrupt return and thus restore context from the stack.  This does not happen and the processor state is not restored correctly on such a context switch.  The system immediately goes off the rails as, among other things, neither the stack pointer nor the program counter are restored correctly.

**To Reproduce**
Steps to reproduce the behavior:
1. Configure the ARC interrupt controller with firq_option=true and number_of_levels=1, and configure Zephyr to match.
2. Run any application that performs a context switch to a previously-interrupted task (I called k_msleep twice: the first context switch to the idle thread works but the second crashes after the idle thread is interrupted by the timer).
3. Watch the processor do crazy things.

**Expected behavior**
Processor context should be correctly restored.

**Impact**
Behaviour is essentially random, processor typically faults after a few instructions and the system halts.

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: Zephyr
 - Toolchain Zephyr SDK
 - Version 2.6

nbowlersnps, 2021-09-03:
The attached patch corrects the issue for me.

[fix-arc-context-switch.patch.txt](https://github.com/zephyrproject-rtos/zephyr/files/7107962/fix-arc-context-switch.patch.txt)

ruuddw, 2021-09-08:
Thanks Nick, for not only reporting the issue but also debugging and suggesting a fix.
Indeed the implicit assumption in the switch code is that a fake interrupt return can be done using the RIRQ. But, if the hardware is configured with only 1 priority (which is not very common, usually there are at least 2 levels), the rtie will follow the FIRQ sequence. We'll need to check if there are other corner cases for your configuration (are you using usermode?).
Would you like to submit your patch as a PR yourself, or do you prefer the ARC team to handle it further?

GavinGonsalves, 2021-10-01:
Probably best to have the ARC team handle it. The patch was the bare minimum to unblock our current project.

ruuddw, 2021-10-07:
In discussion with the reporters, it was decided not to support the FIRQ with only single priority level. To avoid others run in the same issue, a constraint on the configurations is added in #39103.