zephyrproject-rtos/zephyr #38314: ci: create release: message body is too long

cfriedt, 2021-09-04:
**Describe the bug**
The CI process "Create Release" has worked well until now. However, with v2.7.0-rc1 we encountered a situation where the release notes were far too large to include in the message.

**To Reproduce**
Tag a release (or rc) with a long list of changes (e.g. `git shortlog -n v2.6.0..v2.7.0-rc1`)

**Expected behavior**
Probably best to direct the shortlog to ChangeLog.txt and to upload that as an additional artifact.
See https://github.com/zephyrproject-rtos/zephyr/releases/tag/v2.7.0-rc1

**Impact**
It just means that there are additional manual steps. However, this might become the typical behaviour as Zephyr's commit rate grows.

**Logs and console output**
https://github.com/zephyrproject-rtos/zephyr/runs/3509386580

**Environment (please complete the following information):**
 - CI

**Additional context**
Found while tagging v2.7.0-rc1

nashif, 2021-09-04:
need to figure out a better changelog format I guess.

nashif, 2021-09-16:
we should not have this issue anymore, as diffs between release candidates are going to be smaller

cfriedt, 2021-09-18:
What about for the final v2.7.0 release? Will that have a very large changelog as well?