zephyrproject-rtos/zephyr #38316: Synchronize multiple DF TX devices in the DF Connectionless RX Example

Faizan-Ellahi-DE, 2021-09-05:
I want to use an existing example to sync and receive constant tone extension of two or more devices.

Is there any simple way to do it? I am not an expert with RTOS, so with any parametric configuration can this will be resolved, or is there any other way around to develop such features.

I have tried to solve this problem by creating a function with the following process 
1) Disable CTE RX
2) Delete and Terminate Sync

and call this function when I received the first CTE packet, but after some time(approx 1 hr) the error shows up.
Error : **Scan disable...ASSERTION FAIL [(ticker_status == 0) || (ticker_status == 2)] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/ull_scan_aux.c:492**

I am stuck kindly guide, please.

cvinayak, 2021-09-07:
Periodic Advertising List is work in progress, please try this PR: https://github.com/zephyrproject-rtos/zephyr/pull/38338

cvinayak, 2021-09-09:
@Thalley Does the host support synchronization from periodic advertising lists?

Shouldn't this check addresses in the list? https://github.com/zephyrproject-rtos/zephyr/blob/f50574a6a0773e74efb468265ac94cdbabeb97c1/subsys/bluetooth/host/scan.c#L696-L702

Thalley, 2021-09-09:
> @Thalley Does the host support synchronization from periodic advertising lists?
> 
> Shouldn't this check addresses in the list?
> 
> https://github.com/zephyrproject-rtos/zephyr/blob/f50574a6a0773e74efb468265ac94cdbabeb97c1/subsys/bluetooth/host/scan.c#L696-L702

Yes, that is actually still missing. Basically we need to implement this paragraph: 
> If the Periodic Advertiser List is not used, the Advertising_SID, Advertiser Address_Type, and Advertiser Address parameters specify the periodic advertising device to listen to; otherwise they shall be ignored.

So if the option to use the list is set, then we shall ignore the address parameters and instead either just accept it, or store the list in the host and validate whether the address of the event is in the list. The former is the simpler solution, and I am not sure if we need to validate the address in the host.

PR #38338 (2021-10-01): Bluetooth: Controller: Implement Periodic Advertiser List
Implement support for Periodic Advertiser List to be used
in LE Periodic Advertising Create Sync command.

Fixes #38316.
closed 2021-10-01 by carlescufi